Clon1998/mobileraker #228: notification from another printer

Krlesul, 2023-08-21:
Hi, can help me somebody please? I installed Mobileraker in to my printer, my friend installed it too in to his printer. I have other Wi-Fi acces point then my friend. But I got notifications of his printer And nothing of my printer when he has switch on his printer. When is switch off his printer then I got notifications of my printer without problem. When I get his notification And click on this in my phone then is open my printer. 
Do you have some idea what must be changed or what I much check? Thank you so much

Clon1998, 2023-08-21:
Hey,
are you guys using the same raspberry pi to run the two printers or is each printer connected to its own host/RaspberryPi?
Also did you add your friend's printer to your app?

Krlesul, 2023-08-21:
Hi,
I have my raspberry pi and my friend has other raspberry pi. I not added
his printer in to my app and he too.
Maybe we used first day the same acces point for printer but now have each
other Wi-Fi acces point for printer.

Krlesul, 2023-08-21:
Or will help it uninstall it all and again install it? If yes, can you send
me short instructions how to uninstall it? We using octoeverywhere too if
is not here problem but each one has own account.

Krlesul, 2023-08-22:
This is my printer on which I get notifications of my friend. I Will send my friend Log too. 

{"result": {"namespace": "mobileraker", "key": "fcm", "value": {"client": {"lastSeen": "2023-08-21T21:21:23.386412", "version": "v0.4.0-21-g7bb9f8f"}, "d9bf71b8-6501-4479-8ce0-53178d1394a1": {"created": "2023-06-15T20:44:49.767769", "lastModified": "2023-06-15T20:44:49.767814", "fcmToken": "dU2xksbapEjpkbDDiarTmo:APA91bHYhk-FtYsJ_T4O8oIb2up0ZK0lFI9SZ0blzvaQSo7qNK-iEFVJZUrXEeOcdrk-p-I0qEtdCPsYvypbZ1oFTK6nIFIDUDy4h5BJ4_Ob-jIJ11Wzsa_hI357xR9_9U5ayYrfgHtA", "machineName": "Ender 3 S1 PRO", "language": "en", "settings": {"created": "2023-06-15T20:44:49.767771", "lastModified": "2023-08-07T06:37:05.423276", "progress": 0.1, "states": ["complete", "error"]}, "snap": {"progress": 0.1, "state": "printing", "m117": ""}}}}}

Krlesul, 2023-08-22:
[💡 An OE config is available. Can do handover in case local client fails! ref:71943513]
[💡 JsonRpcClient (ws://192.168.1.103/websocket , ClientType.local) CREATED!!]
[💡 [786493245]Trying to connect to ws://192.168.1.103/websocket]
[💡 [786493245-ws://192.168.1.103/websocket] ClientState.disconnected ➝ ClientState.connecting]
[⚠️ Returning LocalClient]
[💡 Ender 3 S1 PRO(ws://192.168.1.101/websocket)  Device's FCM token: dU2xksbapEjpkbDDiarTmo:APA91bHYhk-FtYsJ_T4O8oIb2up0ZK0lFI9SZ0blzvaQSo7qNK-iEFVJZUrXEeOcdrk-p-I0qEtdCPsYvypbZ1oFTK6nIFIDUDy4h5BJ4_Ob-jIJ11Wzsa_hI357xR9_9U5ayYrfgHtA]
[💡 An OE config is available. Can do handover in case local client fails! ref:819656062]
[💡 JsonRpcClient (ws://192.168.1.101/websocket , ClientType.local) CREATED!!]
[💡 [690052298]Trying to connect to ws://192.168.1.101/websocket]
[💡 [690052298-ws://192.168.1.101/websocket] ClientState.disconnected ➝ ClientState.connecting]
[⚠️ Returning LocalClient]
[💡 Theme selected: 1, available theme len: 5]
[💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-14T19:23:24Z, originalAppUserId: $RCAnonymousID:245c833dd9f84b25bd551746876c1f96, allExpirationDates: {}, requestDate: 2023-08-22T06:31:34Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)]
[💡 RCat ID: $RCAnonymousID:245c833dd9f84b25bd551746876c1f96]
[💡 Theme selected: 1, available theme len: 4]
[⚠️ Got channel error SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.1.103, port: 80]
[⚠️ Got channel error SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.1.101, port: 80]
[⚠️ Got channel error SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.1.103, port: 80]
[⛔ [ws://192.168.1.103/websocket786493245] WS-Stream error: SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.1.103, port: 80]
[💡 [786493245-ws://192.168.1.103/websocket] ClientState.connecting ➝ ClientState.error]
[⚠️ Got channel error SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.1.101, port: 80]
[⛔ [ws://192.168.1.101/websocket690052298] WS-Stream error: SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.1.101, port: 80]
[💡 [690052298-ws://192.168.1.101/websocket] ClientState.connecting ➝ ClientState.error]
[💡 Local clientState is ClientState.error. Will switch to remoteClient. ref:71943513]
[⚠️ Returned RemoteClient]
[💡 Local clientState is ClientState.error. Will switch to remoteClient. ref:819656062]
[⚠️ Returned RemoteClient]
[💡 JsonRpcClient (wss://lrLg6PuQLR:3kp9UgyCZmcZzJudmp1SjtvwSMMxQo7B@app-cmgncz46ye0vmtn6mq6xcdh17oew7elv.octoeverywhere.com/websocket , ClientType.octo) CREATED!!]
[💡 [324675121]Trying to connect to wss://lrLg6PuQLR:3kp9UgyCZmcZzJudmp1SjtvwSMMxQo7B@app-cmgncz46ye0vmtn6mq6xcdh17oew7elv.octoeverywhere.com/websocket]
[💡 [324675121-wss://lrLg6PuQLR:3kp9UgyCZmcZzJudmp1SjtvwSMMxQo7B@app-cmgncz46ye0vmtn6mq6xcdh17oew7elv.octoeverywhere.com/websocket] ClientState.disconnected ➝ ClientState.connecting]
[💡 JsonRpcClient (wss://QSfgYEXYpv:OCGYbMyli9rgj5h3cbs9RZpwxMrFrm0T@app-sbxs5qw9eyy5w2luth2asy8w90mpbwni.octoeverywhere.com/websocket , ClientType.octo) CREATED!!]
[💡 [43228998]Trying to connect to wss://QSfgYEXYpv:OCGYbMyli9rgj5h3cbs9RZpwxMrFrm0T@app-sbxs5qw9eyy5w2luth2asy8w90mpbwni.octoeverywhere.com/websocket]
[💡 [43228998-wss://QSfgYEXYpv:OCGYbMyli9rgj5h3cbs9RZpwxMrFrm0T@app-sbxs5qw9eyy5w2luth2asy8w90mpbwni.octoeverywhere.com/websocket] ClientState.disconnected ➝ ClientState.connecting]
[💡 JsonRpcClient (ws://192.168.1.103/websocket, ClientType.local) disposed!]
[💡 JsonRpcClient (ws://192.168.1.101/websocket, ClientType.local) disposed!]
[💡 [324675121-wss://lrLg6PuQLR:3kp9UgyCZmcZzJudmp1SjtvwSMMxQo7B@app-cmgncz46ye0vmtn6mq6xcdh17oew7elv.octoeverywhere.com/websocket] ClientState.connecting ➝ ClientState.connected]
[💡 >>>Fetching Server.Info]
[💡 >>>Fetching Printer.Info]
[💡 [43228998-wss://QSfgYEXYpv:OCGYbMyli9rgj5h3cbs9RZpwxMrFrm0T@app-sbxs5qw9eyy5w2luth2asy8w90mpbwni.octoeverywhere.com/websocket] ClientState.connecting ➝ ClientState.connected]
[💡 >>>Fetching Server.Info]
[💡 >>>Fetching Printer.Info]
[💡 <<<Received Server.Info]
[💡 Jrpc Client of Ender 3 S1 PRO(ws://192.168.1.101/websocket) is connected, can Setup FCM on printer now!]
[💡 Getting fcm]
[💡 <<<Received Printer.Info]
[💡 <<<Received Server.Info]
[💡 Refreshing printer for uuid: 3e3d9c7f-33ab-4158-818f-bc07515c978b]
[💡 >>>Querying printers object list]
[💡 Jrpc Client of Hulk_Voron(ws://192.168.1.103/websocket) is connected, can Setup FCM on printer now!]
[💡 Getting fcm]
[💡 Getting settings]
[💡 <<<Received Printer.Info]
[⚠️ watchWhere just forces owner to invalidate! Ref:AutoDisposeFutureProviderElement<MachineSettings>(provider: selectedMachineSettingsProvider:AutoDisposeFutureProvider<MachineSettings>#aac05, origin: selectedMachineSettingsProvider:AutoDisposeFutureProvider<MachineSettings>#aac05)]
[💡 Getting settings]
[💡 Getting fcm.d9bf71b8-6501-4479-8ce0-53178d1394a1]
[💡 <<<Received printer objects list!]
[💡 >>>Querying Printer Objects!]
[💡 Getting fcm.3e3d9c7f-33ab-4158-818f-bc07515c978b]
[... cut ...] f a component that failed to load.  In the future this will result in a startup error.], klippyStateMessage: Printer is ready)]
[💡 Getting meta for file: `zetony do vozejku_14h50m_0,20mm_240C_ABS_ENDER3S1PRO.gcode`]
[💡 Start MJPEG - targFps: 20 - https://QSfgYEXYpv:OCGYbMyli9rgj5h3cbs9RZpwxMrFrm0T@app-sbxs5qw9eyy5w2luth2asy8w90mpbwni.octoeverywhere.com/webcam/?action=snapshot]
[💡 Getting fcm.client]
[💡 Stop MJPEG]
[💡 _AdaptiveStreamManager DISPOSED]
[💡 Show Dialog request for DialogType.logging]

Clon1998, 2023-08-22:
Thanks for the log! (I deleted it due to privacy reasons)

It looks like you added two printers in the app? Is that correct? Are both printers yours? Or is one of them your friend's printer?

Besides that I don't see a reason to why the companion would not receive notifications. Can you also provide me the log file of the companion?

Clon1998, 2023-08-22:
> Both printers are my. Ended 3s1 pro and Voron. How can I make log of companion? I think it will be problem by my friend because from his printer I get notifications. But I will send you later his log. 

So you're just receiving notifications from his printer? Your printers send you notifications just fine?

Krlesul, 2023-08-22:
My Voron printer is without problem but Ender is problem. I get notifications of my Friend printer CR10 with name MyPrinter but only when his printer printing and not getting notifications of my Ender. When is his printer switched off or when his printer don't printing then I get notifications of my Ender without problem

Krlesul, 2023-08-22:
So I don't get notifications of my printer when my friend printer is switched on too

Krlesul, 2023-08-22:
I have my private network and my friend has his private network. But maybe when we installed it before then he used my old network. I don't remember. But now I have complete new network and also I see in my network router connected only my printers and laptop. We are just now in work. We can check it later my friend log if you will have time 😇.

Clon1998, 2023-08-22:
Alright, I am at work to (lol)

We just send me the http request result from your friend as well.

Krlesul, 2023-08-22:
Hi, here is log of my friend

Krlesul, 2023-08-22:
{"result": {"namespace": "mobileraker", "key": "fcm", "value": {"client": {"lastSeen": "2023-08-11T04:17:12.156452", "version": "v0.4.0-21-g7bb9f8f"}, "9271b2b9-75c1-4ff9-8f49-8a1a2b0388fb": {"created": "2023-06-15T21:43:22.677176", "lastModified": "2023-06-15T21:43:22.677233", "fcmToken": "dU2xksbapEjpkbDDiarTmo:APA91bHYhk-FtYsJ_T4O8oIb2up0ZK0lFI9SZ0blzvaQSo7qNK-iEFVJZUrXEeOcdrk-p-I0qEtdCPsYvypbZ1oFTK6nIFIDUDy4h5BJ4_Ob-jIJ11Wzsa_hI357xR9_9U5ayYrfgHtA", "machineName": "My Printer", "language": "en", "settings": {"created": "2023-06-15T21:43:22.677178", "lastModified": "2023-06-15T21:43:22.677178", "progress": 0.25, "states": ["standby", "printing", "paused", "complete", "error"]}, "snap": {"progress": 0.0, "state": "complete", "m117": ""}}, "17f85daf-0308-4bab-9a95-951911c8c10a": {"created": "2023-08-06T13:21:14.095644", "lastModified": "2023-08-06T13:21:14.095748", "fcmToken": "fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E", "machineName": "cr10", "language": "en", "settings": {"created": "2023-08-06T13:21:14.095646", "lastModified": "2023-08-06T13:21:14.095646", "progress": -1.0, "states": ["complete", "error"]}, "snap": {"progress": 0.0, "state": "complete", "m117": ""}}}}}

Krlesul, 2023-08-22:
💡 Completed FirebaseRemote init
💡 Machine in box is cr10 (17f85daf-0308-4bab-9a95-951911c8c10a)#201357912
💡 Completed Hive init
💡 Completed PaymentService init
💡 Completed initialRoute init
💡 Started initializeAvailableMachines
💡 Selecting machine cr10
💡 Received fetchAll
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-15T14:48:30.000Z, originalAppUserId: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d, allExpirationDates: {}, requestDate: 2023-08-21T20:46:38.892Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
💡 RCat ID: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Received all machines
💡 machineProvider creation STARTED 17f85daf-0308-4bab-9a95-951911c8c10a
💡 machineProvider creation DONE 17f85daf-0308-4bab-9a95-951911c8c10a
💡 initialized all machineProviders
💡 Completed initializeAvailableMachines
💡 Notifications are permitted: true
💡 Received fetchAll
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Setup notification channels
💡 Setup ReceiverPort!
💡 Completed NotificationService init
💡 Theme selected: 1, available theme len: 4
💡 cr10(ws://192.168.10.188/websocket)  Device's FCM token: fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 An OE config is available. Can do handover in case local client fails! ref:65190293
💡 JsonRpcClient (ws://192.168.10.188/websocket , ClientType.local) CREATED!!
💡 [578913489]Trying to connect to ws://192.168.10.188/websocket
💡 [578913489-ws://192.168.10.188/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
⚠️ Returning LocalClient
💡 Token from FCM fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 Received fetchAll
💡 Printer Service received klippyState: null
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
⚠️ Got channel error SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.10.188, port: 80
⛔ [ws://192.168.10.188/websocket578913489] WS-Stream error: SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.10.188, port: 80
💡 [578913489-ws://192.168.10.188/websocket] ClientState.connecting ➝ ClientState.error
💡 Local clientState is ClientState.error. Will switch to remoteClient. ref:65190293
⚠️ Returned RemoteClient
💡 Printer Service received klippyState: KlipperState.error
💡 JsonRpcClient (wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket , ClientType.octo) CREATED!!
💡 [1045704857]Trying to connect to wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket
💡 [1045704857-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
💡 Printer Service received klippyState: null
💡 JsonRpcClient (ws://192.168.10.188/websocket, ClientType.local) disposed!
⚠️ Got channel error WebSocketException: Connection to 'https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com:0/websocket#' was not upgraded to websocket
⚠️ Sending GET to https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket to determine error reason
⛔ [wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket1045704857] WS-Stream error: SocketException: HTTP connection timed out after 0:00:05.000000, host: app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com, port: 443
💡 [1045704857-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.connecting ➝ ClientState.error
💡 Printer Service received klippyState: KlipperState.error
💡 [1045704857]Trying to connect to wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket
💡 [1045704857-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.error ➝ ClientState.connecting
💡 Using headers {}
⚠️ Got channel error WebSocketException: Connection to 'https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com:0/websocket#' was not upgraded to websocket
⚠️ Sending GET to https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket to determine error reason
👾 Got Response: 604
⛔ [wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket1045704857] WS-Stream error: OctoEverywhereHttpException{App Connection Revoked/Expired, 604, parentException: null, parentStack: null}
💡 [1045704857-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.connecting ➝ ClientState.error
💡 App in AppLifecycleState.inactive
💡 App backgrounded

Krlesul, 2023-08-22:
💡 Completed FirebaseRemote init
💡 Machine in box is cr10 (17f85daf-0308-4bab-9a95-951911c8c10a)#558659529
💡 Completed Hive init
💡 Completed PaymentService init
💡 Completed initialRoute init
💡 Started initializeAvailableMachines
💡 Selecting machine cr10
💡 Received fetchAll
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-15T14:48:30.000Z, originalAppUserId: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d, allExpirationDates: {}, requestDate: 2023-08-22T09:09:22.575Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
💡 RCat ID: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Received all machines
💡 machineProvider creation STARTED 17f85daf-0308-4bab-9a95-951911c8c10a
💡 machineProvider creation DONE 17f85daf-0308-4bab-9a95-951911c8c10a
💡 initialized all machineProviders
💡 Completed initializeAvailableMachines
💡 Notifications are permitted: true
💡 Setup notification channels
💡 Setup ReceiverPort!
💡 Completed NotificationService init
💡 Theme selected: 1, available theme len: 4
💡 cr10(ws://192.168.10.188/websocket)  Device's FCM token: fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 An OE config is available. Can do handover in case local client fails! ref:420117898
💡 JsonRpcClient (ws://192.168.10.188/websocket , ClientType.local) CREATED!!
💡 [903868526]Trying to connect to ws://192.168.10.188/websocket
💡 [903868526-ws://192.168.10.188/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
⚠️ Returning LocalClient
💡 Token from FCM fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 [903868526-ws://192.168.10.188/websocket] ClientState.connecting ➝ ClientState.connected
💡 >>>Fetching Server.Info
💡 >>>Fetching Printer.Info
💡 Received fetchAll
💡 Printer Service received klippyState: null
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 <<<Received Server.Info
💡 Printer Service received klippyState: KlipperState.ready
💡 Refreshing printer for uuid: 17f85daf-0308-4bab-9a95-951911c8c10a
💡 >>>Querying printers object list
💡 Jrpc Client of cr10(ws://192.168.10.188/websocket) is connected, can Setup FCM on printer now!
💡 Getting fcm
💡 Getting settings
💡 <<<Received Printer.Info
💡 Partial Update STARTED KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, authorization, update_manager, history, octoprint_compat], warnings: [], klippyStateMessage: null)
💡 Partial Update Done KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, authorization, update_manager, history, octoprint_compat], warnings: [], klippyStateMessage: Printer is ready)
⚠️ watchWhere just forces owner to invalidate! Ref:AutoDisposeFutureProviderElement<MachineSettings>(provider: selectedMachineSettingsProvider:AutoDisposeFutureProvider<MachineSettings>#0d5a3, origin: selectedMachineSettingsProvider:AutoDisposeFutureProvider<MachineSettings>#0d5a3)
💡 Getting settings
💡 Getting fcm.17f85daf-0308-4bab-9a95-951911c8c10a
💡 <<<Received printer objects list!
💡 >>>Querying Printer Objects!
💡 Current FCMConfig in MoonrakerDB: DeviceFcmSettings{machineId: fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E, machineName: cr10, language: en, settings: NotificationSettings{progress: -1.0, states: {complete, error}}}
💡 <<<Received queried printer objects
💡 Fetching cached temperature store data
💡 Received cached temperature store for [heater_bed, temperature_sensor Board_MCU, extruder]
💡 Subscribing printer objects for ws-updates!
💡 Updating Default Macros for "cr10(ws://192.168.10.188/websocket)"!
💡 Getting settings
💡 Also requesting an update for current_file: adapter ring_12m_0,20mm_250C_ABS_CR10SMARTPRO.gcode
💡 Getting meta for file: `adapter ring_12m_0,20mm_250C_ABS_CR10SMARTPRO.gcode`
💡 Max allowed machines for non Supporters is -1
💡 Last dismiss of Supporter AD: 2023-08-14 06:16:31.915
💡 List Webcams request...
💡 Trying to fetch all webcams from moonraker.
💡 UPDATED current_file: GCodeFile(name: adapter ring_12m_0,20mm_250C_ABS_CR10SMARTPRO.gcode, parentPath: gcodes, modified: 1692723659.066967, size: 641531, printStartTime: 1692723659.4897656, jobId: 00007C, slicer: PrusaSlicer, slicerVersion: 2.6.0-rc1+win64, gcodeStartByte: 36811, gcodeEndByte: 630362, layerCount: null, objectHeight: 7.0, estimatedTime: 743.0, nozzleDiameter: 0.4, layerHeight: 0.2, firstLayerHeight: 0.2, firstLayerTempBed: 250.0,  [... cut ...] png), GCodeThumbnail(width: 400, height: 300, size: 24455, relativePath: .thumbs/adapter ring_12m_0,20mm_250C_ABS_CR10SMARTPRO-400x300.png)])
💡 Received 1 webcams from moonraker.
💡 Got 1 webcams
💡 Start MJPEG - targFps: 15 - http://192.168.10.188/webcam/?action=snapshot
💡 App in AppLifecycleState.inactive
💡 App backgrounded
💡 Stop MJPEG
💡 App forgrounded
💡 Start MJPEG - targFps: 15 - http://192.168.10.188/webcam/?action=snapshot
💡 App in AppLifecycleState.inactive
💡 App backgrounded
💡 Stop MJPEG

Krlesul, 2023-08-22:
💡 Completed FirebaseRemote init
💡 Machine in box is cr10 (17f85daf-0308-4bab-9a95-951911c8c10a)#806733159
💡 Completed Hive init
💡 Completed PaymentService init
💡 Completed initialRoute init
💡 Started initializeAvailableMachines
💡 Selecting machine cr10
💡 Received fetchAll
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-15T14:48:30.000Z, originalAppUserId: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d, allExpirationDates: {}, requestDate: 2023-08-22T18:45:55.445Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
💡 RCat ID: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Received all machines
💡 machineProvider creation STARTED 17f85daf-0308-4bab-9a95-951911c8c10a
💡 machineProvider creation DONE 17f85daf-0308-4bab-9a95-951911c8c10a
💡 initialized all machineProviders
💡 Completed initializeAvailableMachines
💡 Notifications are permitted: true
💡 Received fetchAll
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Setup notification channels
💡 Setup ReceiverPort!
💡 Completed NotificationService init
💡 Theme selected: 1, available theme len: 4
💡 cr10(ws://192.168.10.188/websocket)  Device's FCM token: fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 An OE config is available. Can do handover in case local client fails! ref:405351592
💡 JsonRpcClient (ws://192.168.10.188/websocket , ClientType.local) CREATED!!
💡 [349116457]Trying to connect to ws://192.168.10.188/websocket
💡 [349116457-ws://192.168.10.188/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
⚠️ Returning LocalClient
💡 Token from FCM fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 Received fetchAll
💡 Printer Service received klippyState: null
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
⚠️ Got channel error SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.10.188, port: 80
⛔ [ws://192.168.10.188/websocket349116457] WS-Stream error: SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.10.188, port: 80
💡 [349116457-ws://192.168.10.188/websocket] ClientState.connecting ➝ ClientState.error
💡 Local clientState is ClientState.error. Will switch to remoteClient. ref:405351592
⚠️ Returned RemoteClient
💡 Printer Service received klippyState: KlipperState.error
💡 JsonRpcClient (wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket , ClientType.octo) CREATED!!
💡 [831087096]Trying to connect to wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket
💡 [831087096-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
💡 Printer Service received klippyState: null
💡 JsonRpcClient (ws://192.168.10.188/websocket, ClientType.local) disposed!
⚠️ Got channel error WebSocketException: Connection to 'https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com:0/websocket#' was not upgraded to websocket
⚠️ Sending GET to https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket to determine error reason
👾 Got Response: 604
⛔ [wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket831087096] WS-Stream error: OctoEverywhereHttpException{App Connection Revoked/Expired, 604, parentException: null, parentStack: null}
💡 [831087096-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.connecting ➝ ClientState.error
💡 Printer Service received klippyState: KlipperState.error
💡 [831087096]Trying to connect to wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket
💡 [831087096-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.error ➝ ClientState.connecting
💡 Using headers {}
⚠️ Got channel error WebSocketException: Connection to 'https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com:0/websocket#' was not upgraded to websocket
⚠️ Sending GET to https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket to determine error reason
👾 Got Response: 604
⛔ [wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket831087096] WS-Stream error: OctoEverywhereHttpException{App Connection Revoked/Expired, 604, parentException: null, parentStack: null}
[... cut ...]  could complete
💡 Show Dialog request for DialogType.logging
💡 ws://192.168.10.188/websocket566981790] Was unable to complete all hanging JRPC requests after 30sec...
💡 ws://192.168.10.188/websocket566981790] All hanging requests finished!
💡 JsonRpcClient (ws://192.168.10.188/websocket, ClientType.local) disposed!
Bad state: Websocket is closing, request id=33, method printer.objects.query never got an response!
⛔ Unexpected exception thrown during refresh 17f85daf-0308-4bab-9a95-951911c8c10a...

Krlesul, 2023-08-22:
💡 Completed FirebaseRemote init
💡 Machine in box is cr10 (17f85daf-0308-4bab-9a95-951911c8c10a)#35193698
💡 Completed Hive init
💡 Completed PaymentService init
💡 Completed initialRoute init
💡 Started initializeAvailableMachines
💡 Selecting machine cr10
💡 Received fetchAll
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-15T14:48:30.000Z, originalAppUserId: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d, allExpirationDates: {}, requestDate: 2023-08-21T04:28:44.595Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
💡 RCat ID: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Received all machines
💡 machineProvider creation STARTED 17f85daf-0308-4bab-9a95-951911c8c10a
💡 machineProvider creation DONE 17f85daf-0308-4bab-9a95-951911c8c10a
💡 initialized all machineProviders
💡 Completed initializeAvailableMachines
💡 Notifications are permitted: true
💡 Received fetchAll
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Setup notification channels
💡 Setup ReceiverPort!
💡 Completed NotificationService init
💡 Theme selected: 1, available theme len: 4
💡 cr10(ws://192.168.10.188/websocket)  Device's FCM token: fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 An OE config is available. Can do handover in case local client fails! ref:930741136
💡 JsonRpcClient (ws://192.168.10.188/websocket , ClientType.local) CREATED!!
💡 [13457682]Trying to connect to ws://192.168.10.188/websocket
💡 [13457682-ws://192.168.10.188/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
⚠️ Returning LocalClient
💡 Token from FCM fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 Received fetchAll
💡 Printer Service received klippyState: null
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 [13457682-ws://192.168.10.188/websocket] ClientState.connecting ➝ ClientState.connected
💡 >>>Fetching Server.Info
💡 >>>Fetching Printer.Info
💡 <<<Received Server.Info
💡 <<<Received Printer.Info
💡 Partial Update STARTED KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, authorization, update_manager, history, octoprint_compat], warnings: [], klippyStateMessage: null)
💡 Partial Update Done KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, authorization, update_manager, history, octoprint_compat], warnings: [], klippyStateMessage: Printer is ready)
💡 Printer Service received klippyState: KlipperState.ready
💡 Refreshing printer for uuid: 17f85daf-0308-4bab-9a95-951911c8c10a
💡 >>>Querying printers object list
💡 Jrpc Client of cr10(ws://192.168.10.188/websocket) is connected, can Setup FCM on printer now!
💡 Getting fcm
💡 Getting settings
⚠️ watchWhere just forces owner to invalidate! Ref:AutoDisposeFutureProviderElement<MachineSettings>(provider: selectedMachineSettingsProvider:AutoDisposeFutureProvider<MachineSettings>#e207a, origin: selectedMachineSettingsProvider:AutoDisposeFutureProvider<MachineSettings>#e207a)
💡 Getting settings
💡 Getting fcm.17f85daf-0308-4bab-9a95-951911c8c10a
💡 <<<Received printer objects list!
💡 >>>Querying Printer Objects!
💡 Current FCMConfig in MoonrakerDB: DeviceFcmSettings{machineId: fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E, machineName: cr10, language: en, settings: NotificationSettings{progress: -1.0, states: {complete, error}}}
💡 <<<Received queried printer objects
💡 Fetching cached temperature store data
💡 Received cached temperature store for [heater_bed, temperature_sensor Board_MCU, extruder]
💡 Subscribing printer objects for ws-updates!
💡 Updating Default Macros for "cr10(ws://192.168.10.188/websocket)"!
💡 Getting settings
💡 Also requesting an update for current_file: Mesh_body_02_5h42m_0,20mm_250C_ABS_CR10SMARTPRO.gcode
💡 Getting meta for file: `Mesh_body_02_5h42m_0,20mm_250C_ABS_CR10SMARTPRO.gcode`
💡 Max allowed machines for non Supporters is -1
💡 Last dismiss of Supporter AD: 2023-08-14 06:16:31.915
💡 List Webcams request...
💡 Trying to fetch all webcams from moonraker.
💡 UPDATED current_file: GCodeFile(name: Mesh_body_02_5h42m_0,20mm_250C_ABS_CR10SMARTPRO.gcode, parentPath: gcodes, modified: 1692648481.9520214, size: 12535764, printStartTime: 1692648482.7909384, jobId: 000079, slicer: PrusaSlicer, slicerVersion: 2.6.0-rc1+win64, gcodeStartByte: 65569, gcodeEndByte: 12524588, layerCount: null, objectHeight: 31.8, estim [... cut ...] 2, height: 32, size: 2059, relativePath: .thumbs/Mesh_body_02_5h42m_0,20mm_250C_ABS_CR10SMARTPRO-32x32.png), GCodeThumbnail(width: 400, height: 300, size: 44810, relativePath: .thumbs/Mesh_body_02_5h42m_0,20mm_250C_ABS_CR10SMARTPRO-400x300.png)])
💡 GCode "SET_PIN PIN=lights VALUE=0.00" executed successfully!
💡 Stop MJPEG
💡 _AdaptiveStreamManager DISPOSED
💡 Start MJPEG - targFps: 15 - http://192.168.10.188/webcam/?action=snapshot
💡 App in AppLifecycleState.inactive
💡 App backgrounded
💡 Stop MJPEG

Krlesul, 2023-08-22:
💡 Completed FirebaseRemote init
💡 Machine in box is cr10 (17f85daf-0308-4bab-9a95-951911c8c10a)#344077758
💡 Completed Hive init
💡 Completed PaymentService init
💡 Completed initialRoute init
💡 Started initializeAvailableMachines
💡 Selecting machine cr10
💡 Received fetchAll
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-15T14:48:30.000Z, originalAppUserId: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d, allExpirationDates: {}, requestDate: 2023-08-22T09:08:40.785Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
💡 RCat ID: $RCAnonymousID:a8c8237398a74105a9fa568fe3b49e3d
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Received all machines
💡 machineProvider creation STARTED 17f85daf-0308-4bab-9a95-951911c8c10a
💡 machineProvider creation DONE 17f85daf-0308-4bab-9a95-951911c8c10a
💡 initialized all machineProviders
💡 Completed initializeAvailableMachines
💡 Notifications are permitted: true
💡 Received fetchAll
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Setup notification channels
💡 Setup ReceiverPort!
💡 Completed NotificationService init
💡 cr10(ws://192.168.10.188/websocket)  Device's FCM token: fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E
💡 An OE config is available. Can do handover in case local client fails! ref:877568966
💡 JsonRpcClient (ws://192.168.10.188/websocket , ClientType.local) CREATED!!
💡 [1067721960]Trying to connect to ws://192.168.10.188/websocket
💡 [1067721960-ws://192.168.10.188/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
⚠️ Returning LocalClient
💡 Theme selected: 1, available theme len: 4
💡 Received fetchAll
💡 Printer Service received klippyState: null
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
⚠️ Got channel error SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.10.188, port: 80
⛔ [ws://192.168.10.188/websocket1067721960] WS-Stream error: SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.10.188, port: 80
💡 [1067721960-ws://192.168.10.188/websocket] ClientState.connecting ➝ ClientState.error
💡 Local clientState is ClientState.error. Will switch to remoteClient. ref:877568966
⚠️ Returned RemoteClient
💡 Printer Service received klippyState: KlipperState.error
💡 JsonRpcClient (wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket , ClientType.octo) CREATED!!
💡 [607317157]Trying to connect to wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket
💡 [607317157-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
💡 Printer Service received klippyState: null
💡 JsonRpcClient (ws://192.168.10.188/websocket, ClientType.local) disposed!
⚠️ Got channel error WebSocketException: Connection to 'https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com:0/websocket#' was not upgraded to websocket
⚠️ Sending GET to https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket to determine error reason
👾 Got Response: 604
⛔ [wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket607317157] WS-Stream error: OctoEverywhereHttpException{App Connection Revoked/Expired, 604, parentException: null, parentStack: null}
💡 [607317157-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.connecting ➝ ClientState.error
💡 Printer Service received klippyState: KlipperState.error
💡 [607317157]Trying to connect to wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket
💡 [607317157-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.error ➝ ClientState.connecting
💡 Using headers {}
⚠️ Got channel error WebSocketException: Connection to 'https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com:0/websocket#' was not upgraded to websocket
⚠️ Sending GET to https://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket to determine error reason
👾 Got Response: 604
⛔ [wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket607317157] WS-Stream error: OctoEverywhereHttpException{App Connection Revoked/Expired, 604, parentException: null, parentStack: null}
💡 [607317157-wss://1nb6E8Jodu:mcRGB8osoCgd0AdtoJuaSd05wHezJPmH@app-181lnqcnniz0hhuonyl2i7487q2r6vkb.octoeverywhere.com/websocket] ClientState.connecting ➝ ClientState.error
💡 App in AppLifecycleState.inactive
💡 App backgrounded

Clon1998, 2023-08-22:
> {"result": {"namespace": "mobileraker", "key": "fcm", "value": {"client": {"lastSeen": "2023-08-11T04:17:12.156452", "version": "v0.4.0-21-g7bb9f8f"}, "9271b2b9-75c1-4ff9-8f49-8a1a2b0388fb": [...]

Thanks,
Your phone is registered with your friend's printer. If he knows how to issue an HTTP DELETE he can run the following http call to clear out the fcm database: `DELETE http://<PRINTER-IP>/server/database/item?namespace=mobileraker&key=fcm`.
Otherwise, you guys might need to wait for a button I just added that enables users to clear the database from inside the app.

Krlesul, 2023-08-22:
You are the Best one. Where can he insert it? In to web browser or in SSH terminal?

Clon1998, 2023-08-22:
There are several options to run a HTTP-DELETE

The easiest would be to use [CURL](https://reqbin.com/req/c-1dw4uds4/curl-delete-request-example#:~:text=To%20make%20a%20DELETE%20request,HTTP%20headers%20to%20the%20server.)

Small heads up, your friend needs to run it since your pushToken is registered on his machine.

Krlesul, 2023-08-22:
Thank you so much. We Will make it tomorrow. Thank you again

Clon1998, 2023-08-22:
> Thank you so much. We Will make it tomorrow. Thank you again

No problem. I wonder if that fixes the issue with your printer not sending notifications. Just let me know if that also works!

Krlesul, 2023-08-22:
I Will let you know. Thank you so much And I bought you pizza already 👍.

Krlesul, 2023-08-23:
Good morning. We try it just now but not working. If we insert this with sein IP adress so it not working DELETE http://<PRINTER-IP>/server/database/item?namespace=mobileraker&key=fcm

Must we change namespace too?

Clon1998, 2023-08-23:
> Good morning. We try it just now but not working. If we insert this with sein IP adress so it not working DELETE http://<PRINTER-IP>/server/database/item?namespace=mobileraker&key=fcm
> 
> Must we change namespace too?

No, just run an http delete instead of an http get that the browser runs whenever you enter something in the url bar of the browser.

Clon1998, 2023-08-23:
> Good morning. We try it just now but not working. If we insert this with sein IP adress so it not working DELETE http://<PRINTER-IP>/server/database/item?namespace=mobileraker&key=fcm
> 
> Must we change namespace too?

If you guys are using iOS I can invite you to a TestFlight beta that already includes the new button once I am home again.

Krlesul, 2023-08-23:
Ok, we Will try it. And when Will you insert button in to app? 😇

Krlesul, 2023-08-23:
[image]

{
    "result": {
        "namespace": "mobileraker",
        "key": "fcm",
        "value": {
            "client": {
                "lastSeen": "2023-08-22T19:05:44.994894",
                "version": "v0.4.0-21-g7bb9f8f"
            },
            "9271b2b9-75c1-4ff9-8f49-8a1a2b0388fb": {
                "created": "2023-06-15T21:43:22.677176",
                "lastModified": "2023-06-15T21:43:22.677233",
                "fcmToken": "dU2xksbapEjpkbDDiarTmo:APA91bHYhk-FtYsJ_T4O8oIb2up0ZK0lFI9SZ0blzvaQSo7qNK-iEFVJZUrXEeOcdrk-p-I0qEtdCPsYvypbZ1oFTK6nIFIDUDy4h5BJ4_Ob-jIJ11Wzsa_hI357xR9_9U5ayYrfgHtA",
                "machineName": "My Printer",
                "language": "en",
                "settings": {
                    "created": "2023-06-15T21:43:22.677178",
                    "lastModified": "2023-06-15T21:43:22.677178",
                    "progress": 0.25,
                    "states": ["standby", "printing", "paused", "complete", "error"]
                },
                "snap": {
                    "progress": 0.0,
                    "state": "printing",
                    "m117": ""
                }
            },
            "220fd959-a125-4e70-bc93-eab380edb0e8": {
                "created": "2023-08-22T21:10:45.569088",
                "lastModified": "2023-08-22T21:10:45.569156",
                "fcmToken": "fcTmyXMcTjitcmHhfrX01I:APA91bHW1K9KfCFp8AbTRuTzCLsIUikeyjsa5-dIkJEqVWi9JKeT0NDgHAX5VZgeDVMNCZyhUnO9SqGrZhn7fM5HYN0EkPFykHF9Z1GiYQ8irzc6AWbAxbVO498CkbIGLKCw6-0EzQ-E",
                "machineName": "Cr10 smart pro",
                "language": "en",
                "settings": {
                    "created": "2023-08-22T21:10:45.569092",
                    "lastModified": "2023-08-22T21:35:34.291039",
                    "progress": 0.25,
                    "states": ["complete", "error", "paused", "cancelled", "printing", "standby"]
                },
                "snap": {
                    "progress": 0.0,
                    "state": "printing",
                    "m117": ""
                }
            }
        }
    }
}

Krlesul, 2023-08-23:
Is it good ?

Clon1998, 2023-08-23:
> Is it good ?
By the result of the command this looks good to me.
However, validate that you can try to open this URL again: http://your-printer-ip/server/database/item?namespace=mobileraker&key=fcm

If its empty/just one `"snap"` element exists it good.

Krlesul, 2023-08-23:
{"error": {"code": 404, "message": "Key 'fcm' in namespace 'mobileraker' not found", "traceback": "Traceback (most recent call last):\n\n  File \"/home/biqu/moonraker/moonraker/app.py\", line 695, in _process_http_request\n    result = await self._do_request(args, conn)\n\n  File \"/home/biqu/moonraker/moonraker/app.py\", line 671, in _do_local_request\n    return await self.callback(\n\n  File \"/home/biqu/moonraker/moonraker/components/database.py\", line 762, in _handle_item_request\n    val = await self.get_item(namespace, key)\n\n  File \"/usr/lib/python3.9/concurrent/futures/thread.py\", line 52, in run\n    result = self.fn(*self.args, **self.kwargs)\n\n  File \"/home/biqu/moonraker/moonraker/components/database.py\", line 196, in func_wrapper\n    return command_func(*args)\n\n  File \"/home/biqu/moonraker/moonraker/components/database.py\", line 359, in _get_impl\n    ns = self._get_record(namespace, key_list[0])\n\n  File \"/home/biqu/moonraker/moonraker/components/database.py\", line 674, in _get_record\n    raise self.server.error(\n\nmoonraker.utils.ServerError: Key 'fcm' in namespace 'mobileraker' not found\n\n\nThe above exception was the direct cause of the following exception:\n\n\nTraceback (most recent call last):\n\n  File \"/home/biqu/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1713, in _execute\n    result = await result\n\n  File \"/home/biqu/moonraker/moonraker/app.py\", line 658, in get\n    await self._process_http_request()\n\n  File \"/home/biqu/moonraker/moonraker/app.py\", line 697, in _process_http_request\n    raise tornado.web.HTTPError(\n\ntornado.web.HTTPError: HTTP 404: Key 'fcm' in namespace 'mobileraker' not found\n"}}

Krlesul, 2023-08-23:
I don't know but maybe is problem with IP adress of printer because before has my friend other IP adress of his printer And he don't know what for IP it was

Clon1998, 2023-08-23:
> {"error": {"code": 404, "message": "Key 'fcm' in namespace 'mobileraker' not found", "traceback": "Traceback (most recent call last):\n\n File "/home/biqu/moonraker/moonraker/app.py", line 695, in [...]

This looks good! 
The entire cache of FCM token is clear now and your phone should not receive any notifications anymore. Your friend might need to restart his printer, mobileraker_comapnion and the app to ensure he receives himself notifications again.

Clon1998, 2023-08-24:
I will close this issue as complete. Let me know if something is missing.

closed 2023-08-24 by Clon1998