Clon1998/mobileraker #229: Publish on F-Droid

kozross, 2023-08-23:
## Feature Request

### Problem Description

As someone who doesn't use the Play Store (for many reasons), I would much prefer to get this app from F-Droid.

### Proposed Solution

This app should be available on F-Droid.

### Alternatives Considered

Running Google Play, but that's not happening.

### Additional Context

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [X] Provided a clear description of the problem you're facing.
- [X] Clearly outlined the proposed solution or feature.
- [X] Mentioned any alternative solutions or ideas you've considered.
- [X] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

kozross, 2023-08-23:
I realize that this will affect push notifications. Personally, I wouldn't mind them being missing in the F-Droid build: it won't affect my use case in any way.

Clon1998, 2023-08-23:
Duplicate of #20.
As stated in there. The app is using a lot of services that do not comply with the requirements of F-Droid.
Also since I am the only Dev, I don't have the capacity to convert the current app and maintain a fork of it on F-Droid.
If you don't want to use Google Play you can always just download the newest version directly from GitHub

closed 2023-08-23 by Clon1998

Clon1998, 2023-08-23:
In addition, I just noticed that due to the license change of the project, a release to F Droid is not possible.

Jookia, 2023-08-24:
It seems like the license change was only recently. Maybe it could be forked for F-Droid: https://github.com/Clon1998/mobileraker/commit/c1323faf7ffadfcaf82bc23b3622ef6e5735571a