Clon1998/mobileraker #230: KIAUH/Moonraker Managed Updates

fuzelet, 2023-08-24:
## Feature Request

### Problem Description
I would love to assist in helping test various aspects of this application/project, but I am not able to find a way to switch github repos of mobileraker when installed with KIAUH.

### Proposed Solution
Possibly splitting mobileraker and mobileraker_companion into their own installs that can have various git commands ran against the folder to change repos or allow both to be controlled by moonraker.

### Alternatives Considered
I tried adding my own entry to the moonraker.conf file to see if i could use it instead of KIAUH to update, however it does not work the way i tried.

### Additional Context
When using KIAUH and selecting the option to install Mobileraker, it appears to actually be for mobileraker_companion which depends on mobileraker as part of the install process. It would be great if we could install both separately or somehow be able to control their respective updates

**moonraker.conf**
[update_manager mobileraker]
type: git_repo
path: /home/jeffrey/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker.git
primary_branch: dev
managed_services: mobileraker
env: /home/jeffrey/mobileraker-env/bin/python
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install-mobileraker-companion.sh

[update_manager mobileraker-companion]
type: git_repo
path: /home/jeffrey/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch: rewrite
managed_services: mobileraker
env: /home/jeffrey/mobileraker-env/bin/python
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install-mobileraker-companion.sh

[image]

Clon1998, 2023-10-19:
Hey,
I am sorry that it took so long for me to get back to you.

I am a little confused about what you are trying to achieve with these `update_manager` sections.

The repo **mobileraker** (https://github.com/Clon1998/mobileraker.git) represents the flutter app. Therefore, it contains the code for the mobile app. There is no need to include it inside of moonraker since it won't have any effect. In case you want to compile the app yourself you will need to initialize your own firebase project, replace the firebase init file, and also mock the private lib `mobileraker_pro`.

The repo **mobileraker_companion** (https://github.com/Clon1998/mobileraker_companion.git) contains the python application that enables notifications. It does not depend on the core mobileraker repository. In fact, any other app could adapt it and modify the backend server URL to send json data with the notification content to their backend server.

In addition, it seems like you are still using the `rewrite` branch which is not necessary anymore. You can now move back to the normal branch. Please refer to this snippet for the config:

```cfg
[update_manager mobileraker]
type: git_repo
path: ~/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch:main
managed_services: mobileraker
virtualenv: ~/mobileraker-env
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install-mobileraker-companion.sh
```

fuzelet, 2023-10-21:
The idea here was to allow someone that uses kiauh to have the ability to quickly change repos to help the development of either application. Right now the only repo you can change when using kiauh is the mobileraker_companion application.

Clon1998, 2023-10-21:
> The idea here was to allow someone that uses kiauh to have the ability to quickly change repos to help the development of either application. Right now the only repo you can change when using kiauh is the mobileraker_companion application.

There is no need to check out this repo nor add it to KIHAU since it is not related to Klipper directly.

fuzelet, 2023-10-21:
> There is no need to check out this repo nor add it to KIHAU since it is not related to Klipper directly.

Understood. Thank you for your time and additional information regarding this.

closed 2023-10-21 by fuzelet