Clon1998/mobileraker #231: Documentation doesn't clarify licensing

Jookia, 2023-08-24:
The README and website list the project as free and open source, which are terms used to refer to software under an OSI-approved or FSF-approved open source license. This is not the case for this project.

Additionally no where in the documentation does it state this this software cannot be used for commercial use. It is quite possible people are violating this license unknowingly.

Changing the documentation be clear and unambiguous that this software is freeware and only usable for non-commercial use would solve this issue.

Clon1998, 2023-08-24:
> The README and website list the project as free and open source, which are terms used to refer to software under an OSI-approved or FSF-approved open source license. This is not the case for this [...]

I agree that I might need to update some claims on the project's website.
However, I want to make clear that the app core codebase is still open source. I also partly disagree with your concern that the readme does not State anything about the license. The repository itself contains a license file which clearly states that. However, for clarity I can change the readme in this direction.

Jookia, 2023-08-24:
On Thu, Aug 24, 2023 at 12:19:42AM -0700, Patrick Schmidt wrote:
> I agree that I might need to update some claims on the project's website.
> However, I want to make clear that the app core codebase is still open source. I also partly disagree with your concern that [...]

Saying it's open source is a little misleading as it doesn't fit the open source definition: https://opensource.org/osd/
If you mean 'source available' then that is more accurate. But it isn't open source.

Clon1998, 2023-08-24:
@Jookia I adjusted the readme in #232.

PR #232 (2023-08-24): Adjusted readme to reflect the new license
Closes  #231
commit 3af35436 (2023-09-05): Adjusted readme to reflect the new license (#232)

* Adjusted readme to reflect the new license

Closes  #231

* Update README.md
closed 2023-08-24 by Clon1998