Clon1998/mobileraker #234: Add temperature notification options

Jameson42, 2023-08-25:
## Feature Request

### Problem Description
As someone who prints ABS in an enclosure, it is usually best if I wait to remove prints until the enclosure temperature has cooled to near room temperature.

### Proposed Solution
I would like the option to set up a notification based on a temperature sensor dropping to some target value after print completion. For example, setting a target of 25° C would send a notification to my phone after the print has completed and the chamber thermistor value has dropped to 25° C or lower. 

Setting the notifier to watch a temperature sensor of my choice would be preferable, but I would settle for just watching the bed temperature as this gives a reasonable approximation of chamber temp post-print.

### Alternatives Considered
While I could add a temperature wait command to my PRINT_END macro to achieve this, that would artificially inflate the print times in the printer history.

Clon1998, 2023-08-25:
Hey,
Implementing that in GCode with the help of Mobileraker's custom_notifcations might be the best way to keep it flexible.

Here is with what I can up with. I tested it and it seems to be working just fine :).

E.g. (Adjust the senor and more to your parameters): 
```
[gcode_macro PRINT_END]
#   Use PRINT_END for the slicer ending script - please customise for your slicer of choice
gcode:
...
    UPDATE_DELAYED_GCODE ID=cooled_down DURATION=10

[delayed_gcode cooled_down]
gcode:
  TEMPERATURE_WAIT SENSOR=extruder MAXIMUM=90
  M117 $MR$:$printer_name|I am cool af now!
```

Jameson42, 2023-08-26:
Ah, that will work! I didn't realize there was a custom notification functionality. I see it now in the mobileraker_companion repo. 

Perhaps it would be a good idea to link to the companion docs from this project's README?