Clon1998/mobileraker #237: Config Files  - Add Toggle for Backup Files

DrFate09, 2023-08-27:
To make it easier to find config files quickly, it would be great to toggle if I want to see my backup printer.cfg files or not. This functionality is present in Mainsail.

Clon1998, 2023-08-27:
Just to confirm, you are talking about a backup file that is using the file ending .bak?

DrFate09, 2023-08-28:
No, it's the backup printer.cfg files that are auto-generated.

[image]

Clon1998, 2023-08-28:
Gotcha.
And you want to be able to hide them/toggle hidden/visible?

DrFate09, 2023-08-28:
Exactly, same functionality as in Mainsail.

[image]

jamincollins, 2025-09-13:
Yes please!

closed 2026-10-31 by Clon1998