Clon1998/mobileraker #238: Live Activities on iOS

needs-coffee, 2023-08-27:
## Feature Request
A request for live activities notifications on iOS similar to how octopod shows live printing data on the lock screen and similar to the android progress notification that was previously available in mobileraker. 

There is a flutter package for live activities  (https://pub.dev/packages/live_activities) however the issue i currently see is that the actual widget needs to be created in swift as a native iOS widget, not in dart. 

I will try and get round to implementing this but wanted to leave this here as a feature request as it is something i miss from octopod.

Clon1998, 2023-08-27:
Hey,
I am aware of that package. 
Small heads up:
The main problem is getting the information from Firebase to this live activity. I'd assume it could work by injecting some logic into the [MobilerakerNotificationServiceExtension](https://github.com/Clon1998/mobileraker/tree/master/ios/MobilerakerNotificationServiceExtension). However, I still want to look into IOS/Swift myself.

Clon1998, 2023-09-08:
Hey,
I actually wanted to try out Swift and swiftUI so I figured that this might be a good start.
My current UI looks like that
[image]

 Any input & feedback would be appreciated. Especially what kind of info might be handy to show in the live activity?

jMaddrell, 2023-09-09:
It’d be great if there was somewhere to show M117 messages.

Clon1998, 2023-09-30:
First POC for the live activities is now included in v2.6.4

flexzuu, 2023-10-10:
It would be really cool to allow using the widget that was created for the live activity also as normal widget. While no print is going on it could show some simple status msg instead.

Clon1998, 2023-10-10:
> It would be really cool to allow using the widget that was created for the live activity also as normal widget. While no print is going on it could show some simple status msg instead.

There is a separate issue for that #141.

siyu6974, 2023-11-15:
The ETA on the widget is not the same ETA within the app. It's more like a time remaining thing. Can we rename it?

Clon1998, 2023-11-15:
> The ETA on the widget is not the same ETA within the app. It's more like a time remaining thing. Can we rename it?

It depends. But you are right I should use the correct label.

It shows the eta if the print takes longer than 2.5hrs while it uses a timer if only 2.5hre are left.

siyu6974, 2023-11-15:
> It shows the eta if the print takes longer than 2.5hrs while it uses a timer if only 2.5hre are left.

Love it! I can make a PR to fix the label if you like / don't have the time.

Clon1998, 2023-11-15:
> > It shows the eta if the print takes longer than 2.5hrs while it uses a timer if only 2.5hre are left.
> 
> 
> 
> Love it! I can make a PR to fix the label if you like / don't have the time.

Feel free 😉
I think I already included the remaining label/translation and therefore, only a small change in swift is required.

Clon1998, 2023-11-15:
> > It shows the eta if the print takes longer than 2.5hrs while it uses a timer if only 2.5hre are left.
> 
> Love it! I can make a PR to fix the label if you like / don't have the time.

Added in 2cc48b4982460cef20663a00c4ed1b8b88387c53

ihrapsa, 2023-12-16:
I really like this feature, however it feels a bit text crowded. I recently stumbled upon [Joint Craft App](https://apps.apple.com/us/app/joint-craft-voron-companion/id6455888249) and I really like how they did it: 
| App | Joint | Mobileraker |
|-----|-------|---- |
| | [image] |[image] |

Would be nice to:
-  either **limit the height** of the banner or reorder/resize text elements, 
- have an **stl preview** 
- have a **progress bar** instead of the **progress wheel** (only graphic element that feels way too small)
- make the banner follow the system appearance or manual toggle (Dark/Light mode)

What do you think?

Clon1998, 2023-12-16:
> I really like this feature, however it feels a bit text crowded. I recently stumbled upon [Joint Craft App](https://apps.apple.com/us/app/joint-craft-voron-companion/id6455888249) and I really like how they did it:
> 
> App	Joint	Mobileraker
> <img alt="" width="250" src="https://private-user-images.githubusercontent.com/40600040/291017969-7fb9d84e-708a-4802-84c1-6b753fc72e41.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3Njc2NTEsIm5iZiI6MTcwMjc2NzM1MSwicGF0aCI6Ii80MDYwMDA0MC8yOTEwMTc5NjktN2ZiOWQ4NGUtNzA4YS00ODAyLTg0YzEtNmI3NTNmYzcyZTQxLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxNlQyMjU1NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYjkyMTdkNWExMWU1YzIyYWQ1OTdmZWI0ZDJiMDQ0MjdmYzc5YzA1MmY3ZWRjNmQ3YmQxMjg0ZjZjZGY1ODQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.EKHVPAS5Qokoyp-wz5cH9I2KS23mep-v_VxMpDRnfjA">	<img alt="" width="250" src="https://private-user-images.githubusercontent.com/40600040/291018024-8c624cb7-d94e-4c80-bc40-eda467c2fbb8.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3Njc2NTEsIm5iZiI6MTcwMjc2NzM1MSwicGF0aCI6Ii80MDYwMDA0MC8yOTEwMTgwMjQtOGM2MjRjYjctZDk0ZS00YzgwLWJjNDAtZWRhNDY3YzJmYmI4LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxNlQyMjU1NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjJmZTEzY2E1MDIxN2UzNTMxMzZlMWQ3NjU5NzI3ODJlNTgyMWQyNTA1OTZhNTcyZTYzMGM1YTZhNDJkYzBhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.YFqRB0pGLaBWrGd8tExSFq0pcuPei7h-W9CL4FumUt4">
> Would be nice to:
> 
> * either **limit the height** of the banner or reorder/resize text elements,
> * have an **stl preview**
> * have a **progress bar** instead of the **progress wheel** (only graphic element that feels way too small)
> * make the banner follow the system appearance or manual toggle (Dark/Light mode)
> 
> What do you think?

I actually really like their design. However, I am sure they could be a bit upset if I copied it 1:1. 
However, I agree that I still need to work a bit on the activities, especially the background stuff should be a rather fast fix.
Also, their design might not work as well with mobileraker since I currently have no way to share the STL/Preview to the Platform, and I also need to incorporate the machine name due to multi-printer support.

ihrapsa, 2023-12-17:
For sure, we don’t have to copy their design. I can come up with a poc visual if needed and go from there. What limits the sharing of the STL/preview to the Platform?

Clon1998, 2023-12-17:
> For sure, we don’t have to copy their design. I can come up with a poc visual if needed and go from there. What limits the sharing of the STL/preview to the Platform?

I thought the lib I am using currently does not support it. However, it seems like [I am wrong](https://github.com/istornz/flutter_live_activities?tab=readme-ov-file#access-flutter-picture-from-native-), and it actually does (Well, we would still need to download the stl image onto the device since using a URL is not safe due to proxy setups with custom headers and certificates).

Also, images can look bad because it depends on the slicer if 1. a preview is available and 2. the preview is using a transparent/nearly transparent background.

Clon1998, 2023-12-17:
[image]
As of now, I went back to the drawing board and defaulted to a black background, switched to a linear progress bar and made it more compact.

Clon1998, 2023-12-17:
[image]
[image]

I think using this layout is a bit cleaner and provides the most important information right at the top. It will switch between the ETA and Timer depending on the remaining time (3 hours for now).

ihrapsa, 2023-12-17:
I really like what you did. This looks way cleaner and pleasant to the eye. 

Regarding the STL preview, I see the app is able to preview them in `Files` tab. Aren't those locally downloaded? I think we can also use [base64](https://github.com/istornz/flutter_live_activities/issues/30). As long as the size is below [4KB](https://developer.apple.com/documentation/activitykit/displaying-live-data-with-live-activities) ActivityKit should allow that.  For gcode files with no preview a placeholder image or the mobileraker icon could be used instead.

I'm happy for now with what you did so thank you and good job! 🙂

kennydataml, 2024-01-08:
This feature was working for me up until 2 days ago when I restarted my phone. I have no idea what happened. Live activities is enabled in both app settings and iPhone settings.

Clon1998, 2025-02-10:
> Found this thread after thinking about suggesting Live Action support, but found out it’s already being worked on. I also don’t see Live Action despite having it turned on in both the iPhone settings and the app’s settings.

Will look into it since I am able to see the LA getting started normally.

However, did you also install the companion? I forgot if the companion is a requirement to get them working or not.

Besides that, what iOS version are you using?

JarTheHead, 2025-02-11:
> Besides that, what iOS version are you using?

Currently on iOS 18.3.1, I'll try downloading the companion when I get the chance to since I don't believe I installed it (unless kiauh auto installs it with mobileraker)

JarTheHead, 2025-02-27:
Got the chance to print with the companion installed, but live activities are still not starting.

Clon1998, 2025-03-06:
> Got the chance to print with the companion installed, but live activities are still not starting.

Can you share a the app logs?
You can do that at the very bottom of the settings page.

JarTheHead, 2025-03-07:
[mobileraker_2025-03-06T17-43-47.log](https://github.com/user-attachments/files/19117803/mobileraker_2025-03-06T17-43-47.log)