Clon1998/mobileraker #242: Parsing of heater_generic config is broken

Clon1998, 2023-08-30:
This is still happening to me with 2.5.3-249.

I've added a heater_generic and now I can no longer use the app to control my printer.

Exception:
```
 type 'Null' is not a subtype of type 'String' in type cast

#0      _$$_ConfigHeaterGenericFromJson (package:mobileraker/data/dto/config/config_heater_generic.g.dart:14)
#1      new _$_ConfigHeaterGeneric.fromJson (package:mobileraker/data/dto/config/config_heater_generic.freezed.dart:196)
#2      _$ConfigHeaterGenericFromJson (package:mobileraker/data/dto/config/config_heater_generic.freezed.dart:18)
#3      new ConfigHeaterGeneric.fromJson (package:mobileraker/data/dto/config/config_heater_generic.dart:45)
#4      new ConfigFile.parse (package:mobileraker/data/dto/config/config_file.dart:102)
#5      PrinterService._updateConfigFile (package:mobileraker/service/moonraker/printer_service.dart:697)
#6      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:529)
#7      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:478)
#8      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#9      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:614)
#10     PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:773)
<asynchronous suspension>
#11     PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:203)
<asynchronous suspension>
```

Failed-Key: configfile 
Raw Json:
```json
{
    "configfile": {
        "config": {
            "virtual_sdcard": {
                "path": "/home/bush/sv01_data/gcodes"
            },
            "pause_resume": {},
            "display_status": {},
            "menu __main __octoprint": {
                "type": "disabled"
            },
            "menu __main __sdcard": {
                "type": "disabled"
            },
            "menu __main __macros": {
                "type": "list",
                "name": "Macros"
            },
            "menu __main __macros __resume": {
                "type": "command",
                "enable": "{printer.print_stats.state == \"paused\"}",
                "name": "Resume printing",
                "gcode": "\nRESUME"
            },
            "menu __main __macros __pause": {
                "type": "command",
                "enable": "{printer.print_stats.state == \"printing\"}",
                "name": "Pause printing",
                "gcode": "\nPAUSE"
            },
            "menu __main __macros __cancel": {
                "type": "command",
                "enable": "{(printer.print_stats.state == \"printing\" or printer.print_stats.state == \"paused\")}",
                "name": "Cancel printing",
                "gcode": "\nCANCEL_PRINT"
            },
            "stepper_x": {
                "step_pin": "PF0",
                "dir_pin": "PF1",
                "enable_pin": "!PD7",
                "microsteps": "16",
                "rotation_distance": "40",
                "endstop_pin": "^PE5",
[... 835 lines cut ...]
        "extruder": "extruder",
        "position": [
            0.0,
            0.0,
            0.0,
            0.0
        ],
        "max_velocity": 300.0,
        "max_accel": 3000.0,
        "max_accel_to_decel": 1500.0,
        "square_corner_velocity": 5.0
    },
    "extruder": {
        "temperature": 22.36,
        "target": 0.0,
        "power": 0.0,
        "can_extrude": false,
        "pressure_advance": 0.0,
        "smooth_time": 0.04,
        "motion_queue": null
    }
}
```

_Originally posted by @busheezy in https://github.com/Clon1998/mobileraker/issues/205#issuecomment-1699846159_

busheezy, 2023-08-30:
Sorry, I wasn't sure if I should make a new issue or not. Let me know if I can help in any way.

Clon1998, 2023-08-31:
No worries at all, I can always just create a new issue based of an comment, just like it did.

However,
I am unable to reproduce that issue and parsing of the JSON works just fine.

Could you send me the latest log file of the app (App Settings -> Scroll to The Btoom -> Debug logs)

busheezy, 2023-08-31:
I didn't know that was a feature. Very neat.

```
Exception:
 type 'Null' is not a subtype of type 'String' in type cast

#0      _$$_ConfigHeaterGenericFromJson (package:mobileraker/data/dto/config/config_heater_generic.g.dart:14)
#1      new _$_ConfigHeaterGeneric.fromJson (package:mobileraker/data/dto/config/config_heater_generic.freezed.dart:196)
#2      _$ConfigHeaterGenericFromJson (package:mobileraker/data/dto/config/config_heater_generic.freezed.dart:18)
#3      new ConfigHeaterGeneric.fromJson (package:mobileraker/data/dto/config/config_heater_generic.dart:45)
#4      new ConfigFile.parse (package:mobileraker/data/dto/config/config_file.dart:102)
#5      PrinterService._updateConfigFile (package:mobileraker/service/moonraker/printer_service.dart:697)
#6      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:529)
#7      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:478)
#8      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#9      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:614)
#10     PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:773)
<asynchronous suspension>
#11     PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:203)
<asynchronous suspension>

Failed-Key: configfile 
Raw Json:
{"configfile":{"config":{"virtual_sdcard":{"path":"/home/bush/sv01_data/gcodes"},"pause_resume":{},"display_status":{},"menu __main __octoprint":{"type":"disabled"},"menu __main __sdcard":{"type":"disabled"},"menu __main __macros":{"type":"list","name":"Macros"},"menu __main __macros __resume":{"type":"command","enable":"{printer.print_stats.state == \"paused\"}","name":"Resume printing","gcode":"\nRESUME"},"menu __main __macros __pause":{"type":"command","enable":"{printer.print_stats.state == \"printing\"}","name":"Pause printing","gcode":"\nPAUSE"},"menu __main __macros __cancel":{"type":"command","enable":"{(printer.print_stats.state == \"printing\" or printer.print_stats.state == \"paused\")}","name":"Cancel printing","gcode":"\nCANCEL_PRINT"},"stepper_x":{"step_pin":"PF0","dir_pin":"PF1","enable_pin":"!PD7","microsteps":"16","rotation_distance":"40","endstop_pin":"^PE5","position_endstop":"0","position_max":"300","homing_speed":"50"},"stepper_y":{"step_pin":"PF6","dir_pin":"PF7","enable_pin":"!PF2","microsteps":"16","rotation_distance":"40","endstop_pin":"^PJ1","position_endstop":"0","position_max":"255","homing_speed":"50"},"stepper_z":{"step_pin":"PL3","dir_pin":"!PL1","enable_pin":"!PK0","microsteps":"16","rotation_distance":"8","endstop_pin":"probe:z_virtual_endstop","position_max":"300"},"extruder":{"max_extrude_only_distance":"100.0","max_extrude_cross_section":"5.0","step_pin":"PA4","dir_pin":"!PA6","enable_pin":"!PA2","microsteps":"16","rotation_distance":"7.680","nozzle_diameter":"0.400","filament_diameter":"1.750","heater_pin":"PB4","sensor_type":"EPCOS 100K B57560G104F","sensor_pin":"PK5","min_temp":"0","max_temp":"265","control":"pid","pid_kp":"30.170","pid_ki":"3.409","pid_kd":"66.751"},"heater_bed":{"heater_pin":"PH5","sensor_type":"EPCOS 100K B57560G104F","sensor_pin":"PK6","min_temp":"0","max_temp":"110","control":"pid","pid_kp":"72.630","pid_ki":"1.664","pid_kd":"792.579"},"fan":{"pin":"PH6"},"bltouch":{"sensor_pin":"^PD3","control_pin":"PB5","x_offset":"-34.0","y_offset":"-3.5","z_offset":"2.30","samples":"3","samples_result":"median","sample_retract_dist":"5.0","samples_tolerance":"0.01","samples_tolerance_retries":"3"},"gcode_macro CANCEL_PRINT":{"description":"Cancel the actual running print","rename_existing":"CANCEL_PRINT_BASE","gcode":"\nTURN_OFF_HEATERS\nCANCEL_PRINT_BASE"},"gcode_macro PAUSE":{"description":"Pause the actual running print","rename_existing":"PAUSE_BASE","variable_extrude":"1.0","gcode":"\n\n{% set E = printer[\"gcode_macro PAUSE\"].extrude|float %}\n\n\n{% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}\n{% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}\n\n{% set max_z = printer.toolhead.axis_maximum.z|float %}\n{% set act_z = printer.toolhead.position.z|float %}\n{% if act_z < (max_z - 2.0) %}\n{% set z_safe = 2.0 %}\n{% else %}\n{% set z_safe = max_z - act_z %}\n{% endif %}\n\nPAUSE_BASE\nG91\n{% if printer.extruder.can_extrude|lower == 'true' %}\nG1 E-{E} F2100\n{% else %}\n{action_respond_info(\"Extruder not hot enough\")}\n{% endif %}\n{% if \"xyz\" in printer.toolhead.homed_axes %}\nG1 Z{z_safe} F900\nG90\nG1 X{x_park} Y{y_park} F6000\n{% else %}\n{action_respond_info(\"Printer not homed\")}\n{% endif %}"},"gcode_macro RESUME":{"description":"Resume the actual running print","rename_existing":"RESUME_BASE","gcode":"\n\n{% set E = printer[\"gcode_macro PAUSE\"].extrude|float %}\n\n{% if 'VELOCITY' in params|upper %}\n{% set get_params = ('VELOCITY=' + params.VELOCITY)  %}\n{%else %}\n{% set get_params = \"\" %}\n{% endif %}\n\n{% if printer.extruder.can_extrude|lower == 'true' %}\nG91\nG1 E{E} F2100\n{% else %}\n{action_respond_info(\"Extruder not hot enough\")}\n{% endif %}\nRESUME_BASE {get_params}"},"gcode_macro M600":{"gcode":"\n{% set X = params.X|default(50)|float %}\n{% set Y = params.Y|default(0)|float %}\n{% set Z = params.Z|default(10)|float %}\nSAVE_GCODE_STATE NAME=M600_state\nPAUSE\nG91\nG1 E-.8 F2700\nG1 Z{Z}\nG90\nG1 X{X} Y{Y} F3000\nG91\nG1 E [... cut ...]
serial: /dev/serial/by-id/usb-Klipper_rp2040_E6605838836F9A30-if00

[multi_pin heater_pin]
pins: pico:gpio9,pico:gpio26

[multi_pin fan_pin]
pins: pico:gpio5,pico:gpio27

[heater_generic my_heater]
sensor_type: BME280
gcode_id: ENC
i2c_mcu: pico
i2c_speed: 400000
heater_pin: multi_pin:heater_pin
max_power: 1
control: watermark
min_temp: 0
max_temp: 55

[verify_heater my_heater]
check_gain_time: 120

[heater_fan my_heater_fan]
pin: multi_pin:fan_pin
max_power: 1
heater: my_heater
heater_temp: 30.0
fan_speed: 1.0
```

Clon1998, 2023-08-31:
Sorry I input your JSON into the wrong file I am using for testing <.<

The App is complaining because your heater `heater_generic` does not use a sensor_pin?
Is that expected? As far as I understand the klipper docs the sensor_pin should always be present?

Clon1998, 2023-08-31:
Since you are using the `BME280` that uses i2c you can not define a sensor_pin (lol)...
I will adapt my implementation since I don't really care about most of the config entries I parse. I just used the Klipper config for reference and sometimes it simply is outdated.

Clon1998, 2023-08-31:
However, I won't trigger a new patch release since going by my reporting, you are the only users affected by that issue.

Clon1998, 2023-08-31:
However, if you are on IOS I can trigger a TestFlight version that resolves the issue.

Clon1998, 2023-08-31:
Or provide an Android APK :)

busheezy, 2023-08-31:
I'm an android user. An APK would be perfect. Thanks.

busheezy, 2023-08-31:
I added you on Discord.

closed 2023-09-05 by Clon1998