Clon1998/mobileraker #243: Notification snapshot shown in wrong orientation

leitenky, 2023-08-31:
## Bug Report

### Description
A What notifications are pushed to mobile raker the web camera images are flipped ( despite displaying correctly in the app when looking at the stream)

### Steps to Reproduce
1. Start a print
2. Wait for notifications to be pushed to the phone
3. The image received will be the original default orientation of the web camera not the modified/rotated version

### Expected Behavior
The notification snapshot should appear in the same orientation as in the app

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information
- Mobileraker:  @v0.4.0-21

- Klipper: v0.11.0-271-g5f990f93
- Moonraker: v0.8.0-138-gfe120952

### Debug Logs
[mobileraker_2023-08-31T19-03-45.log](https://github.com/Clon1998/mobileraker/files/12491087/mobileraker_2023-08-31T19-03-45.log)
[mobileraker_2023-08-31T07-46-03.log](https://github.com/Clon1998/mobileraker/files/12491088/mobileraker_2023-08-31T07-46-03.log)
[mobileraker_2023-08-31T18-28-10.log](https://github.com/Clon1998/mobileraker/files/12491089/mobileraker_2023-08-31T18-28-10.log)
[mobileraker_2023-08-30T23-45-20.log](https://github.com/Clon1998/mobileraker/files/12491090/mobileraker_2023-08-30T23-45-20.log)
[mobileraker_2023-08-30T23-38-55.log](https://github.com/Clon1998/mobileraker/files/12491091/mobileraker_2023-08-30T23-38-55.log)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-08-31:
You will need TK configure the rotation inside the companion:

https://github.com/Clon1998/mobileraker_companion#companion---config

Clon1998, 2023-09-01:
Let me know if further assistance is required.

closed 2023-09-01 by Clon1998

leitenky, 2023-09-01:
I'll follow up if I have any further issues.  Thanks for the info!