Clon1998/mobileraker #248: Unable to parse ConfigFile with multiple extruders

shermzm, 2023-09-08:
## Bug Report

### Description
Since the latest update to 

### Steps to Reproduce
Add a new printer
Test connection, succeed
Click finish
Error appears

### Expected Behavior
Click finish, view dashboard.

### Version Information
- Mobileraker: 2.6
- Klipper: v0.11.0-271-g5f990f93
- Moonraker: v0.8.0-138-gfe120952

### Debug Logs
[mobileraker_2023-09-07T20-30-28.log](https://github.com/Clon1998/mobileraker/files/12554419/mobileraker_2023-09-07T20-30-28.log)

Exception:
 type 'Null' is not a subtype of type 'num' in type cast

#0      _$$_ConfigExtruderFromJson (package:common/data/dto/config/config_extruder.g.dart:12)
#1      new _$_ConfigExtruder.fromJson (package:common/data/dto/config/config_extruder.freezed.dart:205)
#2      _$ConfigExtruderFromJson (package:common/data/dto/config/config_extruder.freezed.dart:18)
#3      new ConfigExtruder.fromJson (package:common/data/dto/config/config_extruder.dart:26)
#4      new ConfigFile.parse (package:common/data/dto/config/config_file.dart:71)
#5      PrinterService._updateConfigFile (package:common/service/moonraker/printer_service.dart:703)
#6      PrinterService._parseObjectType (package:common/service/moonraker/printer_service.dart:533)
#7      PrinterService._temperatureStore.<anonymous closure> (package:common/service/moonraker/printer_service.dart:482)
#8      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#9      PrinterService._parseQueriedObjects (package:common/service/moonraker/printer_service.dart:620)
#10     PrinterService._printerObjectsQuery (package:common/service/moonraker/printer_service.dart:779)
<asynchronous suspension>
#11     PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:207)
<asynchronous suspension>

Clon1998, 2023-09-08:
Hey,
thanks for reporting.
Could you please share the output of the following website if you visit it on your PC/Browser:

http://<YOUR PRINTER IP>/printer/objects/query?configfile

shermzm, 2023-09-08:
Sure thing, here you go @Clon1998 

[query.json.txt](https://github.com/Clon1998/mobileraker/files/12560741/query.json.txt)

shermzm, 2023-09-08:
Thanks for your quick resolution! I'm on Android, so I'll just wait for the next version to be released.

Clon1998, 2023-09-08:
Wait...
I thought you are on IOS.
Android is even easier. I will upload the APK to the 2.6.2 Tag that includes the fix

shermzm, 2023-09-08:
Even better, thank you!

shermzm, 2023-09-08:
That one works again. Thank you!

closed 2023-09-08 by shermzm

Clon1998, 2023-09-14:
Included in 2.6.3