Clon1998/mobileraker #250: Config files not syncing

DrFate09, 2023-09-09:
## Bug Report

### Description
I have been encountering a strange issue today where my local Klipper and Mobileraker instances have differing config files. I have been trying to save changes to my Start macro but the changes are not being sync to my local Klipper instance and vice versa with syncing my local Klipper to Mobileraker.

Here’s an image of the same Start macro from the same printer, notice that the G28 Z & G28 X Y Z are not the same, enough though they should be.

https://imgur.com/a/pQBDuSK

### Version Information
- Mobileraker: 2.6.2-276

Clon1998, 2023-09-09:
Hey,
thanks for reporting the issue.
I replaced how mobileraker Downloads Files by making the download a background task to ensure the Ui does not flicker.
It seems like I implemented it in a way where the app downloads the file and keeps it in temporary storage. If you reopen the same file it will use the temporary file instead of downloading the latest config file.
I've changed that to ensure xo fig files are always downloaded again to ensure they stay up to date.

Clon1998, 2023-09-14:
Included in 2.6.3

closed 2023-09-14 by Clon1998