Clon1998/mobileraker #251: Possible memory leak in mobileraker.py

austinrdennis, 2023-09-09:
## Bug Report

### Description
I've noticed a slow and gradual increase in the amount of memory mobileraker.py uses that points to a possible memory leak.

### Steps to Reproduce
Boot the host machine and start the mobileraker service. Not sure what exactly the mechanism is.

### Expected Behavior
Memory usage remains relatively stable over time and is reasonable.

### Screenshots
[image]
[image]

### Version Information
- Mobileraker: v0.4.0-25-g2808855d
- Klipper: v0.11.0-271-g5f990f93
- Moonraker: v0.8.0-138-gfe120952

### Debug Logs
[mobileraker_2023-09-08T21-06-33.log](https://github.com/Clon1998/mobileraker/files/12566802/mobileraker_2023-09-08T21-06-33.log)

### Additional Context
After about 26 hours of runtime, it's now using around 2.1 GB of system memory which seems like way too much.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [X] Provided a clear bug description.
- [X] Listed detailed steps to reproduce the issue.
- [X] Described the expected behavior.
- [X] Included the Mobileraker version you are using.
- [X] Attached Mobileraker's debug log files.
- [X] Specified the version numbers of Klipper and Moonraker if applicable.

austinrdennis, 2023-09-09:
I've also posted this as bug in the mobileraker_companion repo as issue 57 as I wasn't sure which one to choose. Feel free to close this issue if it's not applicable this repo.

austinrdennis, 2023-09-09:
`sudo systemctl restart mobileraker.service` clears out the memory usage at least temporarily. I'll see if it builds up again.

Clon1998, 2023-09-09:
'll leave this open since other users might start to look for the issue on mobileraker rather than the mobileraker_companion repo.

The original issue for this problem is https://github.com/Clon1998/mobileraker_companion/issues/27

austinrdennis, 2023-09-10:
> 'll leave this open since other users might start to look for the issue on mobileraker rather than the mobileraker_companion repo.
> 
> The original issue for this problem is [Clon1998/mobileraker_companion#27](https://github.com/Clon1998/mobileraker_companion/issues/27)

Doh! I feel like a total dork for not looking for a duplicate issue in the other repo, only this one. My bad! I'll report back if the issue is solved for me sometime on Monday. I got a long print going, can't update just yet.

austinrdennis, 2023-09-11:
Looks like the latest commit on mobileraker_companion squashed this bug. Thanks for all your awesome work! 😁

closed 2023-09-11 by austinrdennis