Clon1998/mobileraker #252: Error while trying to fetch printer...

Bupher, 2023-09-10:
## Bug Report

### Description
Cannot connect to one of my printers with mobileraker, but the other connects just fine. Already verified no IP address issues as the web client works just fine from the same device. I can also access a different printer on the same network.

This issue persists across multiple android devices. 

### Steps to Reproduce
1. Go to 'Manage Printers' 
2. Click 'Bender' (the name of the printer in question) 
3. Wait for it to throw an error

### Expected Behavior
Mobileraker connects to the printer and allows me to access and control it. 

### Screenshots
[image]
[image]

### Version Information
- Mobileraker: 2.6.2-276
- Klipper: v0.11.0-271-g5f990f93
- Moonraker: v0.8.0-138-gfe120952

### Debug Logs

[dl3.pushbulletusercontent.com_3uobvivYy2hu6uKZBXHzyeshBsOI8wAZ_mobileraker_2023-09-09T22-15-52.log.txt](https://github.com/Clon1998/mobileraker/files/12567065/dl3.pushbulletusercontent.com_3uobvivYy2hu6uKZBXHzyeshBsOI8wAZ_mobileraker_2023-09-09T22-15-52.log.txt)
[dl3.pushbulletusercontent.com_B7nratUHZ1xCjK7Xe1OgEKIsorZ8k0SE_mobileraker_2023-09-09T21-50-30.log.txt](https://github.com/Clon1998/mobileraker/files/12567061/dl3.pushbulletusercontent.com_B7nratUHZ1xCjK7Xe1OgEKIsorZ8k0SE_mobileraker_2023-09-09T21-50-30.log.txt)
[dl3.pushbulletusercontent.com_wvzO7E9EvF7PoVX0401lmqA6kaG0gJGt_mobileraker_2023-09-09T21-11-26.log.txt](https://github.com/Clon1998/mobileraker/files/12567062/dl3.pushbulletusercontent.com_wvzO7E9EvF7PoVX0401lmqA6kaG0gJGt_mobileraker_2023-09-09T21-11-26.log.txt)
[dl3.pushbulletusercontent.com_E6vb0EoJI41C38BAwJBJUNKC2oJiaX2o_mobileraker_2023-09-09T21-24-38.log.txt](https://github.com/Clon1998/mobileraker/files/12567063/dl3.pushbulletusercontent.com_E6vb0EoJI41C38BAwJBJUNKC2oJiaX2o_mobileraker_2023-09-09T21-24-38.log.txt)
[dl3.pushbulletusercontent.com_uBW0g2GURBTcJGpXfE0aeqclQA1bHT0H_mobileraker_2023-09-09T22-16-22.log.txt](https://github.com/Clon1998/mobileraker/files/12567064/dl3.pushbulletusercontent.com_uBW0g2GURBTcJGpXfE0aeqclQA1bHT0H_mobileraker_2023-09-09T22-16-22.log.txt)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [X] Provided a clear bug description.
- [X] Listed detailed steps to reproduce the issue.
- [X] Described the expected behavior.
- [X] Included the Mobileraker version you are using.
- [X] Attached Mobileraker's debug log files.
- [X] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-09-10:
Hey thanks for reporting!
are you by any chance using multiple extruders?
The logs look pretty similar to the ones from this issue https://github.com/Clon1998/mobileraker/issues/248

You can find a new APK with the fix in the mentioned issue. I will probably release the fix via google play and AppStore during next week.

Bupher, 2023-09-10:
Yes, this is a mutli extruder machine.

I tried to just update the Google Play installed app using the new apk, but it gave me an error. I thought I've done this before with other projects, but I'll just wait a few days for the Google play release and not have to redo my configs.

I appreciate the assistance!

Clon1998, 2023-09-14:
Closed as duplicate of #248. 
Bug fix included in 2.6.3, should be available either tomorrow or during next week.

Bupher, 2023-09-23:
App updated, issue resolved

closed 2023-09-23 by Bupher