Clon1998/mobileraker #253: Multiple Printers - File view not updating correctly between printers

fuzelet, 2023-09-11:
## Bug Report

### Description
Open printer.cfg on one printer then switch to another printer and open that printer.cfg. The second printer's file is actually the first. The file being viewed is actually from a printer that is "off" as well.

### Steps to Reproduce
1. Go to 'Files'
2. Click on 'printer.cfg'
3. Close file, save, etc 
4. Change to another printer that is running.
5. Repeat steps above.
6. File opener second time is actually file from other printer.

### Expected Behavior
File being opened is actually the proper file from the respective printer.

### Screenshots
[image]
[image]
[image]
[image]
[image]

### Version Information
- Mobileraker: [v2.6.2-276 - android]
- Klipper: [v0.11.0-275-g8ef0f7d7]
- Moonraker: [0.8.0-138-gfe120952]

### Debug Logs
[mobilerakerapplogs.zip](https://github.com/Clon1998/mobileraker/files/12580952/mobilerakerapplogs.zip)

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

fuzelet, 2023-09-12:
> Hey! Thanks for reporting this. I need to double check but I think I already fixed that! The fix will be included in 2.6.3

Appreciate the quick response as always. Let me know if you need anything else from me.

fuzelet, 2023-09-12:
Wanted to post this here as well in the event it's related for some reason. If not I can open a new issue for it.

I noticed that the information in the "print preview" is not updating correctly and not sure if it will lead to issues with preheating, etc. The hotend and bed temps are flipped?
The thumbnail of the print is also pixelated even with a 32x32 and 400x400 thumbnail.

[image]
[image]

Clon1998, 2023-09-12:
> In the future should I try the TestFlight iOS version prior to reporting issues? I primarily use Beta Android as I am a supporter on that platform, but can always give the iOS TestFlight version a go PRIOR to reporting an issues. I ask I see the iOS version is typically "newer" than the android equivalent? I'd rather limit the amount of extra work you do cause I am just using an older version of the application. Is there a better place to bring these questions up as well for development purposes? Discord? Discussion Thread?
> 
> Not sure if this issue is even related to the above, just want to make sure I am not reporting duplicates or stuff that was already fixed in a newer version, etc.

No that's fine,
also I just checked. 023d4471ea7eb2abb4ccd30b8982ccda9549f322 Fixed the downloading of config files :).

Clon1998, 2023-09-14:
Included in 2.6.3

closed 2023-09-14 by Clon1998

fuzelet, 2023-09-19:
Not sure if this warrants a new issue or not, but noticed the same issue with gcode file thumbnails not syncing now.

[image]
[image]