Clon1998/mobileraker #254: Add webcam save function not working properly

Notjohnsnow, 2023-09-16:
## Bug Report

### Description
Able to add printer successfully but when I go to add a web camera and enter all the required info I am able to see it when I hit the preview button. Hitting the save icon after goes to the dashboard but doesn't not show the camera view. 
Going back into settings to verify the camera info but it is set back to "default" with no info saved. 

### Steps to Reproduce
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See the error.

### Expected Behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information
- Mobileraker: [version]
- Klipper: [version]
- Moonraker: [version]

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Bigscrb15, 2023-09-17:
> ## Bug Report
> ### Description
> Able to add printer successfully but when I go to add a web camera and enter all the required info I am able to see it when I hit the preview button. Hitting the save icon after goes to the dashboard but doesn't not show the camera view. Going back into settings to verify the camera info but it is set back to "default" with no info saved.
> 
> ### Expected Behavior
> Http info saved and camera view displayed in dashboard
> 
> ### Version Information
> * Mobileraker: current
> 
> ### Debug Logs
> ### Additional Context
> Add any other relevant context about the problem here.
> 
> #### Checklist
> To help us diagnose the issue, please ensure you've completed the following steps:
> 
> * [ ]  Provided a clear bug description.
> * [ ]  Listed detailed steps to reproduce the issue.
> * [ ]  Described the expected behavior.
> * [ ]  Included the Mobileraker version you are using.
> * [ ]  Attached Mobileraker's debug log files.
> * [ ]  Specified the version numbers of Klipper and Moonraker if applicable.

I am having the exact same result. iOS 17

Clon1998, 2023-09-17:
Could either of you please share the app logs ( App settings -> scroll to bottom -> Debug logs).
In addition please provide me a screenshot of the dashboard and the version number of moonraker and mobileraker.

Notjohnsnow, 2023-09-17:
I tried to upload the logs when making the report but it says it can't except the file type. Hopefully you can see it this way.

[image]
[image]

Clon1998, 2023-09-17:
Thanks,
it seems like you are running an outdated Moonraker version. 
Could you please post the version of Moonraker you are using? There was a bug that prevented the usage prior to Moonraker version `v0.8.0-127`.

Notjohnsnow, 2023-09-17:
How/where do I locate the version?
I noticed in the Moonraker config under update manager  everything was # out. Someone in another outlet suggested that I add unhash everything and add "enable_auto_refresh: True" which I did. I also enabled auto_system_updates.
I assume this would update to the latest version after some time?
I just did this so not sure how the change will take effect

Clon1998, 2023-09-18:
You should be able to lookup the version in mainsail as you mentioned. I am not sure why the update panel in mainsail is not showing the version correctly tho. Therefore, the moonraker version should also be stated in the moonraker log files.

furrypotato, 2023-09-18:
I too have same issue. Using Neptune 4 pro. I think Elegoo use a customised version so I can't update Moonraker to the required version. I'll check with their Devs to see if they intend to update. 
[image]

Notjohnsnow, 2023-09-19:
> You should be able to lookup the version in mainsail as you mentioned. I am not sure why the update panel in mainsail is not showing the version correctly tho. Therefore, the moonraker version should also be stated in the moonraker log files.

Version is 0.7.1-609-gbdd0222-dirty

Clon1998, 2023-09-19:
In that case the moonraker version is outdated.

I will check what is required to support older versions again.

furrypotato, 2023-09-19:
OK thanks, appreciated. So far loving the app.

RaFl81, 2023-09-20:
Hi,
I still can't add a WebCam. Is there a trick to activate the legacy implementation?
Using a Neptune 4 pro
Klipper: v0.10.0-530-g3387a9c2-dirty-20230308_184216-mkspi
mooraker: v0.7.1-609-gbdd0222-dirty
Mobileraker: 2.6.3-282

💡 BULK ADD/MODIFY Webcams "1" request...
08:20:19.619 (+0:20:23.423436)
💡 ADD/MODIFY Webcam "Default" request...
08:20:19.619 (+0:20:23.423543)
💡 Trying to update or add webcam with uuid:"".
MobilerakerException{Unable to add or update webcam with uuid:, parentException: JRpcError{code: 400, message: Invalid value for parameter 'rotate'}, parentStack: null}
08:20:19.668 (+0:20:23.472689)
⛔ Error while saving cam
08:20:19.671 (+0:20:23.475312)
⛔ isRefreshing SKIPPED
08:20:19.687 (+0:20:23.491390)

Clon1998, 2023-09-20:
> Hi,
>
> I still can't add a WebCam. Is there a trick to activate the legacy implementation?
>
> Using a Neptune 4 pro
>
> Klipper: v0.10.0-530-g3387a9c2-dirty-20230308_184216-mkspi
>
> mooraker: [...]

It will work as soon as I release 2.6.4 of Mobileraker

richar43, 2023-09-28:
Is there any updates to when you will releasing 2.6.4

Clon1998, 2023-09-28:
> Is there any updates to when you will releasing 2.6.4

I hope I get other stuff done tomorrow.
In any case I should be able to release it this weekend.

richar43, 2023-09-28:
That is great to hear. Thank you for all your hard work.

Clon1998, 2023-09-30:
Fixed in v2.6.4

closed 2023-09-30 by Clon1998

furrypotato, 2023-09-30:
Can confirm it camera now works for me. Thanks for your work.

richar43, 2023-09-30:
This might sound like a dumb question,  but where can I find 2.6.4 at

Clon1998, 2023-09-30:
An apk is available on GitHub under the releases.

Google and Apple are currently reviewing the app and it should hopefully be available by Monday.

richar43, 2023-09-30:
Ok, thanks for the update.

RaFl81, 2023-09-30:
Hi. Could Not Test it yet but i can Download it on Google Play Store (Germany) now.

richar43, 2023-10-01:
Got it working now. Thank you for fixing it.