Clon1998/mobileraker #257: Dashboard error with "none" type printer

peterkeen, 2023-09-20:
## Bug Report

### Description
I have a `none` type printer configured like this:

```
# printer.cfg
[mcu]
serial: /dev/serial/by-id/usb-Klipper_rp2040_E661AC8863625425-if00

[printer]
kinematics: none
max_velocity: 1000
max_accel: 1000

[output_pin relay1]
pin: gpio4
```

```
# moonraker.cfg
[server]
host: 0.0.0.0
port: 7126
klippy_uds_address: /home/pete/printer_2_data/comms/klippy.sock

[authorization]
trusted_clients:
    0.0.0.0/0
cors_domains:
    *

[update_manager]
channel: dev
refresh_interval: 168
```

When I set up this printer in Mobileraker I get this error when viewing the dashboard:

```
Exception:
 MobilerakerException{Missing field: motionReport, parentException: null, parentStack: null}

#0      PrinterBuilder.build (package:common/data/dto/machine/printer.dart:88)
#1      PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:212)
<asynchronous suspension>
```

### Steps to Reproduce
1. Set up a printer as above
2. Add it to Mobileraker
3. View the dashboard
4. See the error

### Expected Behavior
I expect the Dashboard to show no information but not throw an error.

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information
- Mobileraker: 2.6.3-282
- Klipper: 8ef0f7d7e3d3b2ac7bc1e80ed3295ceca6bba4e7
- Moonraker: 3d9052d711464cb5ea81cde99769c8f1fd7d3f8d

### Debug Logs
[mobileraker_2023-09-19T19-55-47.log](https://github.com/Clon1998/mobileraker/files/12678397/mobileraker_2023-09-19T19-55-47.log)
[mobileraker_2023-09-20T03-19-24.log](https://github.com/Clon1998/mobileraker/files/12678398/mobileraker_2023-09-20T03-19-24.log)
[mobileraker_2023-09-20T08-48-30.log](https://github.com/Clon1998/mobileraker/files/12678399/mobileraker_2023-09-20T08-48-30.log)
[mobileraker_2023-09-20T08-55-16.log](https://github.com/Clon1998/mobileraker/files/12678400/mobileraker_2023-09-20T08-55-16.log)
[mobileraker_2023-09-20T14-51-31.log](https://github.com/Clon1998/mobileraker/files/12678401/mobileraker_2023-09-20T14-51-31.log)
[text.txt](https://github.com/Clon1998/mobileraker/files/12678402/text.txt)

### Additional Context

I do not have the companion set up for this printer.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-09-21:
Hey,
I will check that. But it seems like the problem is caused by the fact that you set `kinematics: none `. The app always expects a `motionReport` object. I'd assume if no kinematics are set this is also not available.

I also think that even if I would fix this parsing error, you'd run into more problems since there might be other components and data missing.

Any particular reason you're not running a kinematic setting?

peterkeen, 2023-09-21:
Thanks!

I'm using a second Klipper/Moonraker to run power control for my printer via that relay. The primary Moonraker is configured to use the secondary via an HTTP [power] entry.

Mostly I want to use the Mobileraker console and macro list with the second instance. 

If this is not a use case you want to support I completely understand.

Clon1998, 2023-09-21:
> Thanks!
>
> I'm using a second Klipper/Moonraker to run power control for my printer via that relay. The primary Moonraker is configured to use the secondary via an HTTP [power] entry.
>
> Mostly I want [...]

Interesting.
I will have a review what might happen if I make tue Motionreport optional and let you know if I can adapt the app for this use case.

Clon1998, 2023-10-19:
I will assign wont fix since I currently don't have the capacity to support this edge case.