Clon1998/mobileraker #258: Ukrainian localization

iZonex, 2023-09-21:
## Feature Request

### Problem Description

Currently, the Mobileraker app lacks Ukrainian localization, which can hinder the user experience for Ukrainian-speaking users. The absence of a native language interface can be frustrating and may lead to misunderstandings or misuse of the app's features.

### Proposed Solution
Integrate Ukrainian localization into the Flutter app. A **uk.yaml** file containing translation keys for the Ukrainian language has been created and needs to be incorporated into the app settings through **supportedLocales**. This will allow Ukrainian-speaking users to navigate and utilize the app more efficiently and comfortably, enhancing user satisfaction and expanding the app's user base.

### Alternatives Considered
An alternative solution could be the utilization of third-party translation apps or plugins, but these often do not provide contextually accurate translations and can disrupt the user experience. Having integrated, contextually accurate Ukrainian localization is the most effective solution to address the needs of Ukrainian-speaking users.

### Additional Context
The proposed Ukrainian localization has been meticulously translated, considering the grammatical and stylistic nuances of the Ukrainian language and the contextual specificity of the app. It is crucial to test this localization in real-world app usage conditions and gather user feedback to ensure the quality and reliability of the translations.

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x] Provided a clear description of the problem you're facing.
- [x] Clearly outlined the proposed solution or feature.
- [x] Mentioned any alternative solutions or ideas you've considered.
- [x] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

iZonex, 2023-09-21:
Pull Request: https://github.com/Clon1998/mobileraker/pull/256

Clon1998, 2023-09-30:
#256 merged

closed 2023-09-30 by Clon1998