Clon1998/mobileraker #259: Camera url parser cutting everything after & in url

moookino, 2023-09-21:
## Bug Report

### Description
I've camera added with go2rtc and I use mjpeg stream, which works fine with Fluidd, however it's stuck on connection spinner in mobile app and when I try to edit and save camera it creates new camera entry with missing url part after &

URL which works in Fluidd
http://192.168.1.1:1984/stream.html?src=camera1&mode=mjpeg

New camera added when editing from app
http://192.168.1.1:1984/stream.html?src=camera1

Also added from app is lower case with the same name
[image]

### Steps to Reproduce
1. Go to camera settings
2. Click on save with original url from moonraker/fluidd
3. Go to fluidd
4. See another camera entity created with cut everything after & in url for stream

### Expected Behavior

- [ ] application parsing camera url correctly
- [ ] application shouldn't create additional camera entry on save with lower case, but update existing one

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information
- Mobileraker: 2.6.3-282
- Klipper: 
- Moonraker: 

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.
[mobileraker_2023-09-21T09-18-25.log](https://github.com/Clon1998/mobileraker/files/12683456/mobileraker_2023-09-21T09-18-25.log)
[mobileraker_2023-09-21T10-01-51.log](https://github.com/Clon1998/mobileraker/files/12683457/mobileraker_2023-09-21T10-01-51.log)
[mobileraker_2023-09-21T10-09-47.log](https://github.com/Clon1998/mobileraker/files/12683458/mobileraker_2023-09-21T10-09-47.log)
[mobileraker_2023-09-21T11-25-09.log](https://github.com/Clon1998/mobileraker/files/12683459/mobileraker_2023-09-21T11-25-09.log)
[mobileraker_2023-09-21T12-04-07.log](https://github.com/Clon1998/mobileraker/files/12683460/mobileraker_2023-09-21T12-04-07.log)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-09-21:
Hey,
just to confirm.

- You have a MJPEG webcam added via fluid with url http://192.168.1.1:1984/stream.html?src=camera1&mode=mjpeg
- Mobileraker is unable to show the webcam
- You try to edit the webcam from within the app?
- A new webcam is created rather than the current one is being edited?

Can you send me a screenshot of the webcam settings:
1. IN Fluidd
2. In the app when you try to edit it the first time
3. In the app after you saved it with your changes.

moookino, 2023-09-21:
-     You have a MJPEG webcam added via fluid with url http://192.168.1.1:1984/stream.html?src=camera1&mode=mjpeg - Yes
-     Mobileraker is unable to show the webcam - Yes
-     You try to edit the webcam from within the app? - Yes and check if smth is wrong
-     A new webcam is created rather than the current one is being edited? - Yes

1. [image]

1. [image] I noticed camera name and url is incorrect already even without doing any changes, only opening camera in printer settings

1. This one has the same values as 2nd one

AlexxIT, 2023-09-21:
You using wrong url. `stream.html` is not MJPEG format, it is HTML format

Clon1998, 2023-09-21:
Thanks,
I feel like your Moonraker version is a little bit out of date.
Can you provide me the version of moonraker you are using?

moookino, 2023-09-21:
I have it as part of https://github.com/mkuf/prind stack, I've tried to follow https://moonraker.readthedocs.io/en/latest/web_api/#version-information and call it from fluidd container as I don't have it exposed in local network

Maybe I'm missing smth in the application setup, but I realized camera setup between moonraker/fluidd and application aren't synced. So I had camera setup here, which had still old settings, compared to Fluidd. I've tried to delete and add my printer again and camera is missing, when printer was readded.
So I proceed to add camera in mobile app and with correct mjpeg adaptive stream url it finally worked. Now I have "Camera" I've created in Fluidd and "Default" created by mobilraker.

Let me know if that was my dumb mistake or it's some issue 😅

Clon1998, 2023-09-21:
So I had a look at the logs of the app.
The app itself correctly read the input/URL you entered:
```
12:11:30.773 (+0:07:23.334276)
💡 BULK ADD/MODIFY Webcams "1" request...
12:11:30.773 (+0:07:23.334519)
💡 ADD/MODIFY Webcam "camera" request...
12:11:30.773 (+0:07:23.334657)
💡 Trying to update or add webcam with uuid:"camera".
12:11:30.826 (+0:07:23.387402)
⛔ WebcamInfo(uuid: camera, name: camera, service: WebcamServiceType.mjpegStreamer, streamUrl: http://192.168.1.1:1984/stream.html?src=camera1, snapshotUrl: http://192.168.1.1:1984/stream.html?src=camera1, enabled: true, icon: mdiWebcam, aspectRatio: 4:3, targetFpsIdle: 15, location: printer, targetFps: 15, flipHorizontal: false, flipVertical: false, rotation: 0, source: database)
12:11:30.826 (+0:07:23.387626)

```

In addition, there is nothing going on in terms of transforming the URL/input you entered into the textfield on the edit page. This [test](https://github.com/Clon1998/mobileraker/blob/562b19c261aaa4ca2e91253b4f3486ee6d5bf546/common/test/unit/util/misc_test.dart#L947)  also is still fine and covers all cases I could think of.

I wonder what the data in the moonraker DB and the output of the webcam API is after you added the webcam that shows the wrong URL.

To fetch the webcams in moonraker DB: `GET http://<YOUR PRINTER ADDR/IP>/server/database/item?namespace=webcams`

To fetch the webcams via the WebcamAPI: `GET http://<YOUR PRINTER ADDR/IP>//server/webcams/list`

moookino, 2023-09-21:
moonraker DB:
Default is the one I've added through app and Camera is the original one
`{
    "result": {
        "namespace": "webcams",
        "key": null,
        "value": {
            "290c0e7c-bda6-4727-b37c-f57d7cd6cdef": {
                "name": "Default",
                "service": "mjpegstreamer-adaptive",
                "targetFps": 15,
                "urlStream": "http://192.168.1.1:1984/api/stream.mjpeg?src=camera1",
                "urlSnapshot": "http://192.168.1.1:1984/api/frame.jpeg?src=camera1",
                "flipX": false,
                "flipY": false,
                "enabled": false,
                "targetFpsIdle": 5,
                "aspectRatio": "4:4",
                "icon": "",
                "location": "unknown",
                "rotation": 0,
                "extra_data": {}
            },
            "d057c44e-94f6-4901-8962-b76b00de7349": {
                "enabled": true,
                "flipX": false,
                "flipY": false,
                "name": "Camera",
                "rotation": 0,
                "service": "mjpegstreamer-adaptive",
                "targetFps": 15,
                "targetFpsIdle": 5,
                "urlStream": "http://192.168.1.1:1984/api/stream.mjpeg?src=camera1",
                "urlSnapshot": "http://192.168.1.1:1984/api/frame.jpeg?src=camera1"
            }
        }
    }
}`

Webcam API:

`{
    "result": {
        "webcams": [
            {
                "enabled": false,
                "icon": "",
                "aspect_ratio": "4:4",
                "target_fps_idle": 5,
                "name": "Default",
                "location": "unknown",
                "service": "mjpegstreamer-adaptive",
                "target_fps": 15,
                "stream_url": "http://192.168.1.1:1984/api/stream.mjpeg?src=camera1",
                "snapshot_url": "http://192.168.1.1:1984/api/frame.jpeg?src=camera1",
                "flip_horizontal": false,
                "flip_vertical": false,
                "rotation": 0,
                "source": "database",
                "extra_data": {}
            }
        ]
    }
}`

Clon1998, 2023-09-21:
Thanks, seems like the FLuidd version you are using is not yet using the WebcamAPI since the cam added via FLuidd is missing.
Mobileraker now uses the API rather than the Database to ensure a unified API for all interfaces :)

Clon1998, 2023-09-21:
You could also just save the initial/fluidd webcam again inside of fluid to see if the webcam API is updated :)

moookino, 2023-09-21:
Got it, thanks for your help 👍🏻

closed 2023-09-21 by moookino