Clon1998/mobileraker #261: Progress  percentage not the same with mainsail

jpbnto, 2023-09-23:
## Bug Report

### Description
Progress percentage is not the same with mainsail.
It was working before i uodated mobileraker.

### Steps to Reproduce
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See the error.

### Expected Behavior
Is should be the same.

### Screenshots
[image]
[image]

### Version Information
- Mobileraker: v0.4.0-27-g21e70b33
- Klipper: v0.11.0-276-ge9bf2d4c
- Moonraker: v0.8.0-139-g3d9052d7

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-09-23:
Hey,
I will look i to this but I suspect that the app got stuck.
Did you try to refresh the page in the app or force close and reopen the app?

jpbnto, 2023-09-23:
Unfortunately i didn't try that. I just saw the notifications then i compared the app and mainsail. 

I will try it when i will print again.

jpbnto, 2023-09-24:
That's weird. They are now the same. 

[image]
[image]

jpbnto, 2023-09-25:
Another gcode 1% difference
[image]
[image]

Clon1998, 2023-10-12:
> Another gcode 1% difference [image] [image]

Sorry about my late response.

1% is good. It can happen that due to precision error, there is a small deviation between the two UIs.

closed 2023-10-12 by Clon1998