Clon1998/mobileraker #262: Issues with Creality K1 Max

TaylorN15, 2023-10-02:
## Bug Report

### Description
I have a (rooted) Creality K1 Max. When using MobileRaker on my local network, I get an error. I’ve tried adding my wifi SSID to the list, I can see this is being picked up (according to the logs) but makes no difference. When I’m off my LAN, it works fine. 

`Error while fetching inital KlippyObject: Future not completed `

[image]

I’m also getting duplicate notifications at times, with incorrect information (e.g. layer 0/0, time is wrong). 
[image]

And finally, the webcam stream doesn’t seem to work. I can’t even see a widget for it in the app?

Clon1998, 2023-10-02:
Hey thanks for reporting.
Let's address the issues one by one.
Therefore, could you please provide me the mobileraker logs and the app version and moonraker version you are currently using?

1. For the future not complete issue, I should have Fixed that in  the latest version 2.6.4
2. Regarding the notifications, I'd assume you tried to add the printer multiple times/reinstall the app. Therefore, it might be required to reset the notification device registry. You can do that in the printer settings at the very bottom (If the app is connected to the printer, so we might need to fix 1. first)
3. For the webcams, please verify that webcams are shown in the printer edit page. Otherwise go ahead and add one.

TaylorN15, 2023-10-02:
Thanks. I probably should’ve checked for updates first but I only downloaded it 1-2 days ago, seems I just missed the latest version by a few hours! I’ve installed 2.6.4 now. 

Is there any issue with me attaching the logs here? I saw some tokens/keys in the logs. 

1. Not getting the Dart error anymore, but still getting an error when connecting via LAN. Now I’m getting 
`jRpcMethod timed out after 0:00:05.000000` 
2. I only tried to add the printer once. And I don’t get duplicate notifications always, just sometimes. 
3. Thanks, I didn’t see these settings before. I assume these were in the latest release? I’ve added the webcam settings and it’s working now :)

Clon1998, 2023-10-02:
> Thanks. I probably should’ve checked for updates first but I only downloaded it 1-2 days ago, seems I just missed the latest version by a few hours! I’ve installed 2.6.4 now.
>
> Is there any issue [...]

Ohh yeah, 2.6.4 is out since Saturday. 
1. It seems like the timeout for any command is set to short. You could try to increase it to 10-30 sec in the printer's settings to see if that helps. However, since you are connected to the same LAN I suspect something else is wrong with your printer address setup. 
2. I would still suggest you clear the device notification registry. 
3. Well webcam support was there since day one. However, I recently migrated to Moonrakers WebcamAPI which provides a unified interface to interact with webcam settings on multiple UIs. However, this requires Moonraker version 0.8.0-30+. Therefore, with Mobileraker 2.6.4 I reintroduced a legacy webcam system :).

TaylorN15, 2023-10-02:
1. I've increased it to 30 seconds, same issue. Not sure what could be going wrong here. I can only assume there's a different URL or port that you use when connecting locally? I don't know Dart very well, so I probably won't be of much help.
2. I have done this, will let you know if I get any more duplicates. However, the time is still wrong e.g. the ETA shown is in the wrong timezone. But the time in the setting shows correctly. As for the layer count, not sure what's going on here. Fluidd isn't showing a layer count either, but if I hit the Moonraker API from my browser I can see a layer count/total layers etc.
3.

TaylorN15, 2023-10-02:
Ah, I think i worked it out. It was the WebSocket address. In the app it was configured to use ws://10.10.10.30/websocket, but it should be ws://10.10.10.30:7125/websocket. Once I changed this, it seems to work now. However, the webcam still doesn't work on the local network, but I think that's because it's trying to connect on port 80. It's returning a 404 error trying to access http://10.10.10.30/webcam?action=stream, it should be using http://10.10.10.30:4408 as the host, as this is what's specified in my settings.

Clon1998, 2023-10-02:
In case your webcam uses a different port. You need to enter the full URI in the Webcam's settings. Otherwise the app will concat the printer address with `/webcam...`

So in your case: `http://10.10.10.30:4408/webcam?action=stream`

TaylorN15, 2023-10-02:
Yes that’s what I did. But my point was that the printer address I have entered in the app does have the port included. So wouldn’t it use that address as the base URI?

Clon1998, 2023-10-03:
In most cases the port of the printer (Default 7125) is different from the port of the webcam server (80). Therefor, the app does the following things when contacting a relative webcam URL with the machine's websocket uri:
```dart
///Returns a URI that is either based from the machineURI or the camURI if it is absolute.
Uri buildWebCamUri(Uri machineUri, Uri camUri) {
  if (camUri.isAbsolute) return camUri;
  return machineUri.toHttpUri().removePort().resolveUri(camUri);
}
```

TaylorN15, 2023-10-03:
Ok. Makes sense I suppose. Thanks 

Any ideas about the layer/filament info?

Clon1998, 2023-10-03:
> Ok. Makes sense I suppose. Thanks
> 
> Any ideas about the layer/filament info?

You might wanna check that you are running the latest Moonraker version. I think I might use some newer fields to calculate the layers. [Here](https://github.com/Clon1998/mobileraker_companion/blob/21e70b332e41004692b940a3d0e1c76cac5b1786/mobileraker/data/dtos/moonraker/printer_snapshot.py#L175C3-L205) is the code that is used to calculate it.

TaylorN15, 2023-10-03:
Thanks. I’m thinking the issue is elsewhere as it doesn’t show the layer count in the Fluidd UI either, but I can see it’s being set with the print stats macro. I’ll keep digging :)

closed 2023-10-03 by TaylorN15

Clon1998, 2023-10-03:
> Thanks. I’m thinking the issue is elsewhere as it doesn’t show the layer count in the Fluidd UI either, but I can see it’s being set with the print stats macro. I’ll keep digging :)

You might want to check if you added the `[virtual_sdcard]` config. As stated in the [Status_Reference](https://www.klipper3d.org/Status_Reference.html#print_stats)

TaylorN15, 2023-10-04:
> You might want to check if you added the `[virtual_sdcard]` config. As stated in the [Status_Reference](https://www.klipper3d.org/Status_Reference.html#print_stats)

Yep, it's already there.
```
[virtual_sdcard]
path: /usr/data/printer_data/gcodes
```

TaylorN15, 2023-10-06:
FYI for anyone else with this issue, it was something to do with the custom firmware and Moonraker version or something. 

I reinstalled the firmware and latest versions of Fluidd/Moonraker etc following [this](https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki)  wiki, now everything is good :)

jb510, 2024-01-22:
FYI for anyone else with this issue.  I'd used the Guilouz script for installs and apparently forgot on this printer to install the very last module from that script "Mobileraker Companion".  Once I reinstalled that, connection worked through Octoeverywhere.