Clon1998/mobileraker #263: JRpcTimeoutError

siboj, 2023-10-03:
I try to use zerotier tunnel from my mobile to my printer and sometime works sometimes no error error that i recive from mobileraker is 

Exception:
 JRpcTimeoutError: JRpcMethod timed out after 0:00:10.000000

#0      JsonRpcClient.sendJRpcMethod.<anonymous closure> (package:common/network/json_rpc_client.dart:226)
#1      FutureExtensions.onError.onError (dart:async/future.dart:1069)
#2      _RootZone.runBinary (dart:async/zone.dart:1666)
#3      _FutureListener.handleError (dart:async/future_impl.dart:171)
#4      Future._propagateToListeners.handleError (dart:async/future_impl.dart:852)
#5      Future._propagateToListeners (dart:async/future_impl.dart:873)
#6      Future._completeError (dart:async/future_impl.dart:649)
#7      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:927)
#8      Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
#9      _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398)
#10     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429)
#11     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:189)

I change time our to 30 s but still same error

Clon1998, 2023-10-03:
This indicates that the printer does not answer in time. The next time this happens, could you verify if mainsail/fluidd is also broken?

siboj, 2023-10-03:
> This indicates that the printer does not answer in time. The next time this happens, could you verify if mainsail/fluidd is also broken?

Its not broken everithing works fine but i don't know why need tooì much time to open

siboj, 2023-10-03:
If i waiting in open app he will connect, he will try a lot of time and it will pass
[image]

TaylorN15, 2023-10-06:
You could try increasing the value in Printer Settings -> Client timeout in the app.

siboj, 2023-10-09:
> You could try increasing the value in Printer Settings -> Client timeout in the app.

I try that first but still same

Clon1998, 2023-10-09:
> > You could try increasing the value in Printer Settings -> Client timeout in the app.
> 
> I try that first but still same

I am looking into that.
It seems like I made the app throw `JRpcTimeoutError `  to aggressively. I already adjusted it in some places for the upcoming release. Lets see if that helps :)

siboj, 2023-10-09:
> I am looking into that. It seems like I made the app throw `JRpcTimeoutError ` to aggressively. I already adjusted it in some places for the upcoming release. Lets see if that helps :)

Thanks man

I would like to test new version

siboj, 2023-11-04:
Hi

I tested new version of app and everithing is fine, its working very well

closed 2023-11-04 by siboj