Clon1998/mobileraker #264: Detailed info about current and specific layer

apostolovski, 2023-10-05:
## Feature Request

### Problem Description
I regularly print in 2 colors. I switch between filaments at a specific layer. 

### Proposed Solution
1 Show an ETA (time or countdown) until I reach the start of a layer and/or
2 Pause the printer at the start of that layer and/or
3 Notify me when the printer is paused (or is about to pause) and/or
4 Show the “completion %” of the current layer

Clon1998, 2023-10-05:
Hey,
I will look into this. 
However, I think you can achieve what you want by adding custom notifications to your Gcode that allows mobileraker to send you notifications with messages you like.

Clon1998, 2023-10-19:
Hey, 
I had a look and as of now, the app is unable to reliably calculate the current layer and trigger notifications based on that value. The reason for that is that some printers might not include the correct layer or use Z-Hop which would throw off the calculation.

Because of that I suggest you make use of the [Custom Notification](https://github.com/Clon1998/mobileraker_companion/blob/main/docs/Custom_Notifications.md) feature that the [mobileraker_companion ](https://github.com/Clon1998/mobileraker_companion/) offers to get the most reliable result. Just include it in your Slicer's GCode macros.