Clon1998/mobileraker #265: "Multipliers" not changing value when typing, just with scrollers.

thomazff, 2023-10-06:
## Bug Report

### Description
When I try to change some values under Multipliers or Limits this does not reflect onthe machine. With the sliders it works realtime, but if I try a more precise way (typing) this does not work.

### Expected Behavior
As there is no ENTER or Confirm button, I'm assuming the info is not sent to the printer.

### Screenshots
Not aplicable, but on the app it looks like it worked, but in klipperscreen or mainsail there is no change to those values.

### Version Information
- Mobileraker: 2.6.3-282
- Klipper: v0.11.0-295-g83ef0e13
- Moonraker: v0.8.0-142-gee62d07c

### Debug Logs
10:26:45.395 (+0:00:00.067773)
💡 Completed FirebaseRemote init
10:26:45.395 (+0:00:00.068554)
💡 FirebaseRemote values are fetched and activated!
10:26:45.466 (+0:00:00.138758)
💡 Machine in box is My Printer (179c2a40-5ec3-44bf-8ea3-54c16f3040e0)#267415128
10:26:45.466 (+0:00:00.138826)
💡 Completed Hive init
10:26:45.487 (+0:00:00.160076)
💡 Completed PaymentService init
10:26:45.487 (+0:00:00.160204)
💡 Selecting machine My Printer
10:26:45.487 (+0:00:00.160218)
💡 Received fetchAll
10:26:45.490 (+0:00:00.163354)
💡 Theme selected: 0, available theme len: 5
10:26:45.490 (+0:00:00.163372)
💡 Theme selected: 0, available theme len: 5
10:26:45.491 (+0:00:00.164394)
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-05-31T22:07:45Z, originalAppUserId: $RCAnonymousID:325da517bb584c2b8637552886334c04, allExpirationDates: {}, requestDate: 2023-10-06T13:22:11Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
10:26:45.491 (+0:00:00.164419)
💡 RCat ID: $RCAnonymousID:325da517bb584c2b8637552886334c04
10:26:45.491 (+0:00:00.164447)
💡 Received isSupporter false
10:26:45.491 (+0:00:00.164458)
💡 Max allowed machines for non Supporters is -1
10:26:45.491 (+0:00:00.164478)
💡 Completed initialRoute init
10:26:45.491 (+0:00:00.164487)
💡 Started initializeAvailableMachines
10:26:45.491 (+0:00:00.164493)
💡 Received all machines
10:26:45.491 (+0:00:00.164512)
💡 machineProvider creation STARTED 179c2a40-5ec3-44bf-8ea3-54c16f3040e0
10:26:45.491 (+0:00:00.164532)
💡 machineProvider creation DONE 179c2a40-5ec3-44bf-8ea3-54c16f3040e0
10:26:45.491 (+0:00:00.164544)
💡 initialized all machineProviders
10:26:45.491 (+0:00:00.164549)
💡 Completed initializeAvailableMachines
10:26:45.496 (+0:00:00.168997)
💡 Notifications are permitted: true
10:26:45.500 (+0:00:00.172722)
💡 Setup notification channels
10:26:45.500 (+0:00:00.172746)
💡 Setup ReceiverPort!
10:26:45.501 (+0:00:00.174256)
💡 Completed NotificationService init
10:26:45.502 (+0:00:00.175013)
💡 Completed init for workManager
10:26:45.508 (+0:00:00.181205)
💡 Theme selected: 0, available theme len: 4
10:26:45.548 (+0:00:00.220906)
[... 232 lines cut ...]
💡 <<<Received printer objects list!
10:35:00.981 (+0:08:15.653770)
💡 >>>Querying Printer Objects!
10:35:01.095 (+0:08:15.767690)
💡 <<<Received queried printer objects
10:35:01.098 (+0:08:15.770905)
💡 Fetching cached temperature store data
10:35:01.211 (+0:08:15.884578)
💡 Received cached temperature store for [heater_bed, temperature_sensor Raspberry_Pi, extruder]
10:35:01.212 (+0:08:15.885374)
💡 Subscribing printer objects for ws-updates!
10:35:01.213 (+0:08:15.885995)
💡 Updating Default Macros for "My Printer(ws://192.168.68.105/websocket)"!
10:35:01.213 (+0:08:15.886056)
💡 Getting settings
10:35:01.213 (+0:08:15.886503)
💡 Also requesting an update for current_file: rabbit_40m_210C_PLA.gcode
10:35:01.213 (+0:08:15.886553)
💡 Getting meta for file: `rabbit_40m_210C_PLA.gcode`
10:35:01.244 (+0:08:15.917201)
💡 UPDATED current_file: GCodeFile(name: rabbit_40m_210C_PLA.gcode, parentPath: gcodes, modified: 1696595186.4649673, size: 2718158, printStartTime: 1696598400.9502227, jobId: 000069, slicer: PrusaSlicer, slicerVersion: 2.6.1+win64, gcodeStartByte: 87014, gcodeEndByte: 2707775, layerCount: null, objectHeight: 42.6, estimatedTime: 2398.0, nozzleDiameter: 0.4, layerHeight: 0.2, firstLayerHeight: 0.2, firstLayerTempBed: 60.0, firstLayerTempExtruder: 210.0, chamberTemp: null, filamentName: PLA BRANCO PRINTALOT, filamentType: PLA, filamentTotal: 1920.33, filamentWeightTotal: 5.73, thumbnails: [GCodeThumbnail(width: 32, height: 32, size: 1724, relativePath: .thumbs/rabbit_40m_210C_PLA-32x32.png), GCodeThumbnail(width: 128, height: 128, size: 10750, relativePath: .thumbs/rabbit_40m_210C_PLA-128x128.png), GCodeThumbnail(width: 400, height: 400, size: 51788, relativePath: .thumbs/rabbit_40m_210C_PLA-400x400.png)])
10:38:57.340 (+0:12:12.012615)
💡 Getting fcm.client
10:39:41.687 (+0:12:56.359790)
💡 Show Dialog request for DialogType.logging

Clon1998, 2023-10-06:
Thanks for reporting.
Are you using Android or iOS?

thomazff, 2023-10-06:
iOS Version.

Just updated the app to 2.6.4 and it is still there.

closed 2023-10-29 by Clon1998