Clon1998/mobileraker #266: Automatically switch connection without adding SSIDs

TaylorN15, 2023-10-06:
## Feature Request

### Problem Description
I have a few SSIDs that I had to add manually to the app (long pressing doesn’t work on iOS it seems)

### Proposed Solution
Try accessing the web socket or IP locally first, and if this fails, switch to remote?

Clon1998, 2023-10-06:
> ## Feature Request
> ### Problem Description
> I have a few SSIDs that I had to add manually to the app (long pressing doesn’t work on iOS it seems)
> 
> ### Proposed Solution
> Try accessing the web socket or IP locally first, and if this fails, switch to remote?

Long pressing works fine on my IPhone. You will need to grant fine location access since WiFi SSID access requires this permission.
In case you don't want to rapidly switch to the remote connection, you can simply remove all SSIDs from the list and the app will simply try the local connection first and then the remote one in case local fails.

TaylorN15, 2023-10-06:
Hmmm. I’ve tried on my iPhone and iPad, and have granted location access…

Also, I didn’t know about the local check first. If I didn’t have my SSID in the list, it was coming up with the OctoEverywhere logo and said “using remote connection”.

closed 2023-10-06 by TaylorN15

Clon1998, 2023-10-06:
> Hmmm. I’ve tried on my iPhone and iPad, and have granted location access…
> 
> Also, I didn’t know about the local check first. If I didn’t have my SSID in the list, it was coming up with the OctoEverywhere logo and said “using remote connection”.

Did you grant fine location access?

TaylorN15, 2023-10-06:
Yes, “Precise Location” is enabled. All good though, thanks.

Clon1998, 2023-10-06:
> Yes, “Precise Location” is enabled. All good though, thanks.

That is interesting,
could you do me a favor and retry to add a wifi ssid and then upload the latest log here for me to have a look?
You can find the log at the very bottom of the app settings page.

TaylorN15, 2023-10-06:
Apologies. I misunderstood. I was long pressing in the field where you type in the SSID, as that’s where I see the message. I had to close that box, and long press the Add button, maybe the message should be updated.