Clon1998/mobileraker #268: Extrusion feed rate and filament length

Henk72, 2023-10-07:
## Feature Request

### Problem Description
The app supports extruding, but lacks the option to set feed rate and filament length for the extrusion.

### Proposed Solution
Add these parameters like how they are used in Mainsail (see screen shot attached).

### Alternatives Considered
If I need to extrude a bit of filament after a filament change, I can use Mainsail, but prefer using this great app.

### Additional Context
See screenshot.
---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [ ] Provided a clear description of the problem you're facing.
- [ ] Clearly outlined the proposed solution or feature.
- [ ] Mentioned any alternative solutions or ideas you've considered.
- [ ] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

[image]

Clon1998, 2023-10-07:
Hey,
both of these options are available.

To set a extrusion amount you just select any of the step values you configured.

[image]

To configure the steps and also the Feedrate you will need to go thr the printer settings.

[image]

Henk72, 2023-10-07:
Patrick, thank you for the fast reply and great pointer. I didn't know it was possible to define my own extruder steps.

It would be awesome if it would be possible to also define multiple extruder feed rates, like how the extruder steps are working.

Clon1998, 2023-10-07:
Mhhh...
I will think about it. 
However, I think it might be beneficial to simply add the ability to modify a Feedrate directly in that card.

I will have a look to see what works on the ui and different screen sizes

Clon1998, 2024-01-12:
Added in android2.6.9 & ios2.6.9

closed 2024-01-12 by Clon1998