Clon1998/mobileraker #270: "Parsing 'Configfile' failed" when using "temperature_combined" on heater bed

ChuiVanfleet, 2023-10-09:
## Bug Report

### Description
When using `sensor_type: temperature_combined` as the temperature sensor for the heated bed, mobilraker cannot be used.

### Steps to Reproduce
1. Configure your heater bed to use the temperature_combined sensor type. Something like:

```
[temperature_sensor bed_center_temperature]
sensor_type: Generic 3950
sensor_pin: PF3

[heater_bed]
heater_pin: PA3
sensor_type: temperature_combined
sensor_list: temperature_sensor bed_center_temperature
combination_method: mean
maximum_deviation: 120
```
2. Open mobileraker
3. mobileraker fails to parser printer configuration.

### Expected Behavior
I would expect mobileraker to work even, with this configuration since klipper works with it.

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information
- Mobileraker: 2.6.4 (android)
- Klipper: v0.11.0-291-g6adff395
- Moonraker: v0.8.0-142-gee62d07c

### Debug Logs
```
19:43:09.862 (+0:00:00.564036)
💡 [ClientType.local@ws://192.168.1.129/websocket #574943590] JsonRpcClient CREATED!!
19:43:09.862 (+0:00:00.564085)
💡 [ClientType.local@ws://192.168.1.129/websocket #574943590] Trying to connect
19:43:09.862 (+0:00:00.564140)
💡 [ClientType.local@ws://192.168.1.129/websocket #574943590] ClientState.disconnected ➝ ClientState.connecting
19:43:09.862 (+0:00:00.564239)
💡 [ClientType.local@ws://192.168.1.129/websocket #574943590] Using headers {}
19:43:09.862 (+0:00:00.564298)
💡 [ClientType.local@ws://192.168.1.129/websocket #574943590] Using timeout 0:00:06.000000
19:43:09.863 (+0:00:00.565208)
💡 Printer Service received klippyState: null
19:43:09.863 (+0:00:00.565511)
💡 Received isSupporter false
19:43:09.863 (+0:00:00.565583)
💡 Max allowed machines for non Supporters is -1
19:43:09.902 (+0:00:00.604772)
💡 [ClientType.local@ws://192.168.1.129/websocket #574943590] ClientState.connecting ➝ ClientState.connected
19:43:09.903 (+0:00:00.605142)
💡 >>>Fetching Server.Info
19:43:09.903 (+0:00:00.605876)
💡 >>>Fetching Printer.Info
19:43:09.904 (+0:00:00.606441)
💡 INIT QUEUE STATUS...
19:43:09.904 (+0:00:00.606561)
💡 Queue status request...
19:43:09.918 (+0:00:00.620014)
💡 <<<Received Server.Info
19:43:09.918 (+0:00:00.620381)
💡 Printer Service received klippyState: KlipperState.ready
19:43:09.918 (+0:00:00.620445)
💡 Refreshing printer for uuid: 0067d3a3-0618-469b-9bc3-82f461eef6da
19:43:09.918 (+0:00:00.620478)
💡 >>>Querying printers object list
19:43:09.919 (+0:00:00.621010)
💡 Getting settings
19:43:09.929 (+0:00:00.631002)
💡 <<<Received Printer.Info
19:43:09.929 (+0:00:00.631175)
💡 Partial Update STARTED KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, history, update_manager, authorization, octoprint_compat, spoolman], registeredDirectories: [config, logs, gcodes, config_examples, docs], warnings: [], moonrakerVersion: MoonrakerVersion(major: 0, minor: 8, patch: 0, commits: 142, commitHash: gee62d07), klippyStateMessage: null)
19:43:09.929 (+0:00:00.631290)
💡 Partial Update Done KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, history, update_manager, authorization, octoprint_compat, spoolman], registeredDirectories: [config, logs, gcodes, config_examples, docs], warnings: [], moonrakerVersion: MoonrakerVersion(major: 0, minor: 8, patch: 0, commits: 142, commitHash: gee62d07), klippyStateMessage: Printer is ready)
19:43:09.929 (+0:00:00.631558)
💡 NExt status is :JobQueueStatus(queuedJobs: [], queueState: QueueState.paused)
19:43:09.937 (+0:00:00.639719)
💡 <<<Received printer objects list!
19:43:09.938 (+0:00:00.640322)
💡 >>>Querying Printer Objects!
19:43:10.047 (+0:00:00.748994)
💡 <<<Received queried printer objects
type 'Null' is not a subtype of type 'String' in type cast
#0   _$$_ConfigHeaterBedFromJson (package:common/data/dto/config/config_heater_bed.g.dart:13)
#1   new _$_ConfigHeaterBed.fromJson (package:common/data/dto/config/config_heater_bed.freezed.dart:193)
#2   _$ConfigHeaterBedFromJson (package:common/data/dto/config/config_heater_bed.freezed.dart:18)
#3   new ConfigHeaterBed.fromJson (package:common/data/dto/config/config_heater_bed.dart:25)
#4   new ConfigFile.parse (package:common/data/dto/config/config_file.dart:61)
#5   PrinterService._updateConfigFile (package:common/service/moonraker/printer_service.dart:695)
#6   PrinterService._parseObjectType (package:common/service/moonraker/printer_service.dart:532)
#7   PrinterService._temperatureStore.<anonymous closure> (package:common/service/moonraker/printer_service.dart:481)
#8   _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#9   PrinterService._parseQueriedObjects (package:common/service/moonraker/printer_service.dart:617)
#10   PrinterService._printerObjectsQuery (package:common/service/moonraker/printer_service.dart:768)
#11   <asynchronous suspension>
#12   PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:207)
#13   <asynchronous suspension>
19:43:10.051 (+0:00:00.753357)
⛔ Error while parsing configfile object
19:43:10.060 (+0:00:00.762615)
💡 Fetching cached temperature store data
type 'Null' is not a subtype of type 'String' in type cast
#0   _$$_ConfigHeaterBedFromJson (package:common/data/dto/config/config_heater_bed.g.dart:13)
#1   new _$_ConfigHeaterBed.fromJson (package:common/data/dto/config/config_heater_bed.freezed.dart:193)
#2   _$ConfigHeaterBedFromJson (package:common/data/dto/config/config_heater_bed.freezed.dart:18)
#3   new ConfigHeaterBed.fromJson (package:common/data/dto/config/config_heater_bed.dart:25)
#4   new ConfigFile.parse (package:common/data/dto/config/config_file.dart:61)
#5   PrinterService._updateConfigFile (package:common/service/moonraker/printer_service.dart:695)
#6   PrinterService._parseObjectType (package:common/service/moonraker/printer_service.dart:532)
#7   PrinterService._temperatureStore.<anonymous closure> (package:common/service/moonraker/printer_service.dart:481)
#8   _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#9   PrinterService._parseQueriedObjects (package:common/service/moonraker/printer_service.dart:617)
#10   PrinterService._printerObjectsQuery (package:common/service/moonraker/printer_service.dart:768)
#11   <asynchronous suspension>
#12   PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:207)
#13   <asynchronous suspension>
19:43:10.062 (+0:00:00.764758)
⛔ Error in Dash
```
### Additional Context

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-10-09:
Thank you for reporting the error. 
In the current state, Mobileraker does not support this kind of configuration, but I will investigate what is needed to support it.

ChuiVanfleet, 2023-10-13:
I'm trying to be proactive about this. Though admittedly I have next to no experience with mobile app development. But it appears that  changing `required String sensorPin,` to `String? sensorPin` inside common/lib/data/dto/config/config_heater_bed.dart (and likely a similar change in config_extruder.dart ) may do the trick. But I'm unable to test anything since (as far as I can tell) I can't get all the source code (moonraker_pro?). If you're willing to give me access, or explain what I'm doing wrong, I'd love to test this and put up a pull request for it.

Clon1998, 2023-10-19:
> I'm trying to be proactive about this. Though admittedly I have next to no experience with mobile app development. But it appears that changing `required String sensorPin,` to `String? sensorPin` [...]

I'm sorry for not getting back to you sooner. I hope I can have a look and fix it later today.
You are already in the correct code section. However, I want to assess the blast radius of the change to ensure nothing else breaks :).
Also this reminds me to create a public version of `mobileraker_pro` package.

closed 2023-10-29 by Clon1998