Clon1998/mobileraker #273: Live Activity takes up too much space in the Dynamic Island

CobreDev, 2023-10-12:
## Bug Report

### Description
When the Live Activity is active, it takes the entire width of my status bar. This makes viewing the time or other status bar info very hard 

### Steps to Reproduce
1. Start Live Activity by exiting the app
2. Observe

### Expected Behavior
There should be less blank space between the Island, the countdown, and the circle graph to allow the time and other info to be visible

### Screenshots
[image]
[image]

### Version Information
- Mobileraker: 2.6.4

### Debug Logs
N/A

### Additional Context
Device - iPhone 15 Pro, iOS 17.0.3

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [X] Provided a clear bug description.
- [X] Listed detailed steps to reproduce the issue.
- [X] Described the expected behavior.
- [X] Included the Mobileraker version you are using.
- N/A Attached Mobileraker's debug log files.
- N/A Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-10-13:
Hey,
I've also noticed that on the emulator and pushed a fix for that . It will be included in the next release alongside some other changes to the live activity.

CobreDev, 2023-10-13:
Fantastic, thank you!

Clon1998, 2023-10-19:
Here is the updated size, sadly due to a bug with the ActivityKit this is the best I can do

[image]

CobreDev, 2023-10-19:
That's not bad!

closed 2023-10-29 by Clon1998

jordanborth, 2026-02-14:
It's the timer taking up the whole view right?