Clon1998/mobileraker #274: Webcam return request 404 status

DrFailbucket, 2023-10-18:
## Bug Report

### Description
I was adding my K1 Max to Mobileraker and noticed that the Cam doesn't show up. In the config it is set up (i think so) correctly. When i open the cam url in the browser (on phone) it works fine. 

### Screenshots
[image]
[image]

### Version Information
- Mobileraker: `2.6.4-300`
- Klipper: `09faed31-dirty`
- Moonraker: `v0.7.1-780-gdde9bcc-dirty` (as per logfile) 

### Debug Logs
[mobileraker_2023-10-18T02-53-00.log](https://github.com/Clon1998/mobileraker/files/12973762/mobileraker_2023-10-18T02-53-00.log)

### Additional Context
What i noticed is that on Browser it requires to enter the URL like this `XXX.XXX.XXX.XXX:4408/webcam/?action=stream` , so only when adding the port it works. Can't tell if Mobileraker does this to or not. 

I already tried to add ':4408' in front of the 'camera Url' setting like this: `:4408/webcam/?action=stream` but when i hit save it changes back to `/webcam/?action=stream`

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-10-18:
Hey!
I should probably rework that input or add an info text about the behavior of the webcam input.

When you enter a URL into the app, it will be processed differently depending on the type of URL you provide. 

1. If you enter a relative URL, the app will combine the printer IP (without the port) with the path you have provided. For example, if you enter  `/webcam/?action=stream`, the app will combine it with the printer IP `192.168.178.111` to create `192.168.178.111/webcam/?action=stream`.

2. If you enter an absolute URL, the app will use the URL directly without any modifications. For instance, if you enter `132.144.1.99:1234/webcam/?action=stream`, the app will use it as is.

So in your case, you will need to enter the full URL to ensure the app uses the port you specified.

DrFailbucket, 2023-10-18:
Thanks for the fast answer! :)

It wont let me input anything exepct for `/webcam/?action=stream`.
When i try to put in the complete adress and hit save or even preview it is showing the error like the one in the screenshot i provided in the first post

please take a look on how i do it and tell me if i doing something worng please

https://github.com/Clon1998/mobileraker/assets/30552450/ad334961-4895-488b-99a0-3e58fb48c151

DrFailbucket, 2023-10-18:
Okey i got i by myself! 

i have missed the `http://` in front of the ip adress 
Now its working fine! 

to be clear, the URL should look like this: `https://yourip:port/webcam/?action=stream`

closed 2023-10-18 by DrFailbucket

Punto45, 2024-06-26:
These help me two thx