Clon1998/mobileraker #276: remote access Unauthorized

shom, 2023-10-20:
## Bug Report

### Description
I can't save config file and view timelapse on remote access.

### Screenshots
[image]

### Version Information
- Mobileraker: [2.6.4-298]
- Klipper: [v0.11.0-295]
- Moonraker: [v0.8.0-146]

### Debug Logs
10:48:06.171 (+0:00:00.018346)
💡 Completed FirebaseRemote init
10:48:06.171 (+0:00:00.018912)
💡 FirebaseRemote values are fetched and activated!
10:48:06.175 (+0:00:00.022681)
💡 Machine in box is Voron (5d177218-89be-43e9-b7a5-9997d50d6a2a)#571158801
10:48:06.175 (+0:00:00.022709)
💡 Completed Hive init
10:48:06.181 (+0:00:00.028799)
💡 Completed PaymentService init
10:48:06.181 (+0:00:00.028853)
💡 Completed initialRoute init
10:48:06.181 (+0:00:00.028865)
💡 Started initializeAvailableMachines
10:48:06.181 (+0:00:00.028914)
💡 Selecting machine Voron
10:48:06.181 (+0:00:00.028928)
💡 Received fetchAll
10:48:06.182 (+0:00:00.029486)
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-10-04T01:55:45Z, originalAppUserId: $RCAnonymousID:9dbca79bfb324914b59c48012b75cd8b, allExpirationDates: {}, requestDate: 2023-10-20T02:47:07Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
10:48:06.182 (+0:00:00.029504)
💡 RCat ID: $RCAnonymousID:9dbca79bfb324914b59c48012b75cd8b
10:48:06.182 (+0:00:00.029527)
💡 Received isSupporter false
10:48:06.182 (+0:00:00.029539)
💡 Max allowed machines for non Supporters is -1
10:48:06.182 (+0:00:00.029556)
💡 Received all machines
10:48:06.182 (+0:00:00.029587)
💡 machineProvider creation STARTED 5d177218-89be-43e9-b7a5-9997d50d6a2a
10:48:06.182 (+0:00:00.029607)
💡 machineProvider creation DONE 5d177218-89be-43e9-b7a5-9997d50d6a2a
10:48:06.182 (+0:00:00.029621)
💡 initialized all machineProviders
10:48:06.182 (+0:00:00.029630)
💡 Completed initializeAvailableMachines
10:48:06.183 (+0:00:00.030506)
💡 Theme selected: 0, available theme len: 5
10:48:06.183 (+0:00:00.030524)
💡 Theme selected: 0, available theme len: 5
10:48:06.188 (+0:00:00.035623)
💡 Notifications are permitted: true
10:48:06.188 (+0:00:00.035656)
💡 Started to listen for LiveActivity updates
10:48:06.192 (+0:00:00.039695)
💡 Setup notification channels
10:48:06.192 (+0:00:00.039759)
💡 Returning LocalClient
10:48:06.192 (+0:00:00.039784)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #322159786] JsonRpcClient CREATED!!
10:48:06.192 (+0:00:00.039795)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #322159786] Trying to connect
10:48:06.192 (+0:00:00.039807)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #322159786] ClientState.disconnected ➝ ClientState.connecting
10:48:06.192 (+0:00:00.039826)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #322159786] Using headers {X-Api-Key: b788f4d0d23448daad9b62exxxxxxxxx}
[... 713 lines cut ...]
💡 Received fetchAll
11:08:51.658 (+0:20:45.506081)
💡 Received isSupporter false
11:08:51.658 (+0:20:45.506136)
💡 Max allowed machines for non Supporters is -1
11:08:55.866 (+0:20:49.713500)
💡 Received fetchAll
11:08:55.866 (+0:20:49.713802)
💡 Received isSupporter false
11:08:55.866 (+0:20:49.713865)
💡 Max allowed machines for non Supporters is -1
11:08:56.685 (+0:20:50.533037)
💡 Getting fcm.client
11:08:59.846 (+0:20:53.693548)
💡 Show Dialog request for DialogType.logging

### Moonraker log

2023-10-20 10:59:53,697 [web.py:log_exception()] - 401 GET /server/files/timelapse/timelapse_sb%E9%80%82%E9%85%8D%20%E5%90%ABx%E9%99%90%E4%BD%8D%E5%9B%BA%E5%AE%9A%E4%BD%8D_ABS_1h59m.gcode_20231019_2116.mp4 (125.45.x.x): Unauthorized
2023-10-20 10:59:53,704 [app.py:log_request()] - 401 GET /server/files/timelapse/timelapse_sb%E9%80%82%E9%85%8D%20%E5%90%ABx%E9%99%90%E4%BD%8D%E5%9B%BA%E5%AE%9A%E4%BD%8D_ABS_1h59m.gcode_20231019_2116.mp4 (125.45.x.x) [No User] 17.68ms

2023-10-20 11:07:25,839 [app.py:log_request()] - 101 GET /websocket (125.45.x.x) [_API_KEY_USER_] 7.36ms
2023-10-20 11:07:25,841 [websockets.py:open()] - Websocket Opened: ID: 548217931280, Proxied: False, User Agent: Dart/3.1 (dart:io), Host Name: klipper.xxxxx.top
2023-10-20 11:07:31,837 [web.py:log_exception()] - 401 POST /server/files/upload (125.45.x.x): Unauthorized
2023-10-20 11:07:31,843 [app.py:log_request()] - 401 POST /server/files/upload (125.45.x.x) [No User] 15.24ms

Clon1998, 2023-10-20:
Hey,
Thanks for reporting this error.

I had a look at your logs and the app is constantly using the local connection and not ans remote connection?
However, it seems like the local client does not include the API Key in the file download requests.
This should be a quick fix.

shom, 2023-10-20:
I reconfigured, I don't want to use octoeverywhere so I use manual.
This is a new log, the result is still the same.

2023-10-20 19:46:17,267 [web.py:log_exception()] - 401 GET /server/files/timelapse/timelapse_%5Ba%5D_filter_access_cover_ABS_1h0m.gcode_20231020_1659.mp4? (125.45.x.x): Unauthorized
2023-10-20 19:46:17,275 [app.py:log_request()] - 401 GET /server/files/timelapse/timelapse_%5Ba%5D_filter_access_cover_ABS_1h0m.gcode_20231020_1659.mp4? (125.45.x.x) [No User] 21.51ms

16:55:16.899 (+0:00:00.021529)
💡 Completed FirebaseRemote init
16:55:16.901 (+0:00:00.023089)
💡 FirebaseRemote values are fetched and activated!
16:55:16.908 (+0:00:00.029734)
💡 Machine in box is Voron (5d177218-89be-43e9-b7a5-9997d50d6a2a)#11095032
16:55:16.908 (+0:00:00.029795)
💡 Completed Hive init
16:55:16.918 (+0:00:00.040704)
💡 Completed PaymentService init
16:55:16.919 (+0:00:00.040826)
💡 Completed initialRoute init
16:55:16.919 (+0:00:00.040853)
💡 Started initializeAvailableMachines
16:55:16.919 (+0:00:00.040980)
💡 Selecting machine Voron
16:55:16.919 (+0:00:00.041014)
💡 Received fetchAll
16:55:16.922 (+0:00:00.044255)
💡 Theme selected: 0, available theme len: 6
16:55:16.922 (+0:00:00.044288)
💡 Theme selected: 0, available theme len: 6
16:55:16.923 (+0:00:00.044855)
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-10-04T01:55:45Z, originalAppUserId: $RCAnonymousID:9dbca79bfb324914b59c48012b75cd8b, allExpirationDates: {}, requestDate: 2023-10-19T08:50:40Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
16:55:16.923 (+0:00:00.044910)
💡 RCat ID: $RCAnonymousID:9dbca79bfb324914b59c48012b75cd8b
16:55:16.923 (+0:00:00.044972)
💡 Received isSupporter false
16:55:16.923 (+0:00:00.045001)
💡 Max allowed machines for non Supporters is -1
16:55:16.923 (+0:00:00.045028)
💡 Received all machines
16:55:16.923 (+0:00:00.045088)
💡 machineProvider creation STARTED 5d177218-89be-43e9-b7a5-9997d50d6a2a
16:55:16.923 (+0:00:00.045132)
💡 machineProvider creation DONE 5d177218-89be-43e9-b7a5-9997d50d6a2a
16:55:16.923 (+0:00:00.045168)
💡 initialized all machineProviders
16:55:16.923 (+0:00:00.045186)
💡 Completed initializeAvailableMachines
16:55:16.925 (+0:00:00.047515)
💡 Notifications are permitted: true
16:55:16.925 (+0:00:00.047560)
💡 Started to listen for LiveActivity updates
16:55:16.933 (+0:00:00.055067)
💡 Setup notification channels
16:55:16.933 (+0:00:00.055200)
💡 A manual-RemoteClient is available. Can do handover in case local client fails! ref:723728264
16:55:16.933 (+0:00:00.055227)
💡 [Smart-Switching] Local SSID list is empty. Smart switching disabled
16:55:16.933 (+0:00:00.055288)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #699815072] JsonRpcClient CREATED!!
16:55:16.933 (+0:00:00.055311)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #699815072] Trying to connect
16:55:16.933 (+0:00:00.055333)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #699815072] ClientState.disconnected ➝ ClientState.connecting
16:55:16.933 (+0:00:00.055379)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #699815072] Using headers {X-Api-Key: b788f4d0d23448daad9b62exxxxxxxxx}
16:55:16.933 (+0:00:00.055402)
💡 [ClientType.local@ws://klipper.xxx.top:7010/websocket #699815072] Using timeout 0:00:05.000000
16:55:16.934 (+0:00:00.055745)
💡 Returning LocalClient
16:55:16.934 (+0:00:00.055985)
💡 Printer Service received klippyState: null
[... 3266 lines cut ...]
10:21:58.211 (+17:26:41.333528)
💡 INIT QUEUE STATUS...
10:21:58.211 (+17:26:41.333571)
💡 Queue status request...
10:21:58.257 (+17:26:41.378782)
💡 <<<Received Server.Info
10:21:58.257 (+17:26:41.378936)
💡 NExt status is :JobQueueStatus(queuedJobs: [], queueState: QueueState.paused)
10:21:58.257 (+17:26:41.378989)
💡 Printer Service received klippyState: KlipperState.ready
10:21:58.257 (+17:26:41.379017)
💡 Refreshing printer for uuid: 5d177218-89be-43e9-b7a5-9997d50d6a2a
10:21:58.257 (+17:26:41.379039)
💡 >>>Querying printers object list
10:21:58.271 (+17:26:41.392730)
💡 <<<Received Printer.Info
10:21:58.271 (+17:26:41.392844)
💡 Partial Update STARTED KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, history, timelapse, octoprint_compat, authorization, update_manager], registeredDirectories: [config, logs, gcodes, timelapse, timelapse_frames, config_examples, docs], warnings: [], moonrakerVersion: MoonrakerVersion(major: 0, minor: 8, patch: 0, commits: 146, commitHash: g52ee49e), klippyStateMessage: null)
10:21:58.271 (+17:26:41.392949)
💡 Partial Update Done KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, history, timelapse, octoprint_compat, authorization, update_manager], registeredDirectories: [config, logs, gcodes, timelapse, timelapse_frames, config_examples, docs], warnings: [], moonrakerVersion: MoonrakerVersion(major: 0, minor: 8, patch: 0, commits: 146, commitHash: g52ee49e), klippyStateMessage: Printer is ready)
10:21:58.305 (+17:26:41.427430)
💡 <<<Received printer objects list!
10:21:58.306 (+17:26:41.428078)
💡 >>>Querying Printer Objects!
10:21:58.374 (+17:26

Clon1998, 2023-11-08:
Gotcha,
will have a look. Seems like the web client that downloads files (Configs and time lapse) does not use the correct config/ headers.

Clon1998, 2023-11-08:
Just fixed the issue. It will be included in the upcoming release.

Clon1998, 2023-12-06:
Fix is included in 2.6.7