Clon1998/mobileraker #277: Live Aktivität und Push-Benachrichtigung

Experte9, 2023-10-20:
## Bug Report
Hallo ;)
Ich mal wieder, habe mal 3 Probleme.
### Description
In der Live Aktivität steht keine Zeit und/oder kein Name der Druckdatei.
Ich habe mehrere Drucker in Mobileraker, manchmal werden mir 2 Live Aktivitäten angezeigt und manchmal nur 1. (logischerweise drucken beide gleichzeitig).
Seit dem Update mit der Live Aktivität bekomme ich keine Push-Benachrichtigung mehr.

### Steps to Reproduce
1. Datei um Drucken auswählen
2. Drucken
3. Bildschirm sperrt sich (meist noch beim Aufheizen oder wenn der Fortschritt bei 0% steht)
4. Im Sperrbildschirm die Live Aktivität anzeigen
5. (Wenn man dann später mal ein schaut ist die Wahrscheinlichkeit das es richtig angezeigt wird höher)

### Screenshots
[image]
[image]
[image]

### Version Information
- Mobileraker App: 2.6.4-298
- Mobileraker: 0.4.0-29-g5a4cae4a
- Klipper: 0.11.0-299-gb1f597c5
- Moonraker: 0.8.0-186-g2641fc54

### Debug Logs
[mobileraker_2023-10-19T08-53-18.log](https://github.com/Clon1998/mobileraker/files/13053924/mobileraker_2023-10-19T08-53-18.log)
[mobileraker_2023-10-19T12-40-05.log](https://github.com/Clon1998/mobileraker/files/13053927/mobileraker_2023-10-19T12-40-05.log)
[mobileraker_2023-10-19T12-52-59.log](https://github.com/Clon1998/mobileraker/files/13053928/mobileraker_2023-10-19T12-52-59.log)
[mobileraker_2023-10-19T16-13-10.log](https://github.com/Clon1998/mobileraker/files/13053929/mobileraker_2023-10-19T16-13-10.log)
[mobileraker_2023-10-20T13-26-46.log](https://github.com/Clon1998/mobileraker/files/13053930/mobileraker_2023-10-20T13-26-46.log)

Clon1998, 2023-10-20:
Hey,
bei den Live Activities war mir das auch schon aufgefallen. In der neusten Version sollten die bisschen öfters updaten.
Das Problem mit den mehreren Druckern höre ich zum ersten Mal. Ich werde mir das mal anschauen. 

Die normalen Push Mitteilungen sollten eigentlich ohne Probleme gehen. Könntest du mir die mobileraker.log Datei vom companion hochladen? Ich schaue mal ob der was sagt.

Experte9, 2023-10-21:
Alles klar, dann warte ich freudig auf die neue Version.

Anbei noch die .log Datei
[mobileraker.log](https://github.com/Clon1998/mobileraker/files/13060805/mobileraker.log)

Clon1998, 2023-10-21:
> Alles klar, dann warte ich freudig auf die neue Version.
> 
> Anbei noch die .log Datei [mobileraker.log](https://github.com/Clon1998/mobileraker/files/13060805/mobileraker.log)

Mhh die logs sehen eigentlich gut aus, sowohl deine client logs sowie die Backend Server logs :(

Clon1998, 2023-10-21:
Könntest du CustomNotifications mal testen. Dafür einfach `M117 $MR$:Hey, I am a notification` in die GCode Konsole eingeben.
Diese sollten eigentlich direkt eine Push schicken.

Nur zur Sicherheit, prüf auch mal die IOS einstellungen bezüglich der Notifications für MR.

Experte9, 2023-10-21:
Hab den Befehl getestet, kam auch keine Meldung. 
Die Einstellungen hab ich auch geprüft: 
[image]
[image]

Experte9, 2023-10-28:
Nachdem heute das Update kam (App) geht das mit den Push wieder, zumindest bei einem von beiden. 
Bei dem 2. zeigt er mir dafür eine Fehlermeldung. Kiauh sagt aber dass es installiert ist. 
[image]
Ich beobachte noch ob das mit den Live Aktivitäten klappt und gebe dann noch mal Bescheid.

Clon1998, 2023-10-28:
> Nachdem heute das Update kam (App) geht das mit den Push wieder, zumindest bei einem von beiden. 
> Bei dem 2. zeigt er mir dafür eine Fehlermeldung. Kiauh sagt aber dass es installiert ist. 
> [image]
> Ich beobachte noch ob das mit den Live Aktivitäten klappt und gebe dann noch mal Bescheid. 

prüfe am besten Mal in der Service läuft

Audi systemctl status mobileraker.service

Experte9, 2023-10-28:
Das ist der Output:

● mobileraker.service - Companion app to enable push notifications on mobileraker
     Loaded: loaded (/etc/systemd/system/mobileraker.service; enabled; vendor preset: e>
     Active: active (running) since Sat 2023-10-28 20:06:53 CEST; 2h 47min ago
   Main PID: 200517 (python)
      Tasks: 1 (limit: 1389)
        CPU: 1min 50.188s
     CGroup: /system.slice/mobileraker.service
             └─200517 /home/pi/mobileraker-env/bin/python /home/pi/mobileraker_companio>

Oct 28 22:49:06 mainsailos python[200517]: 2023-10-28 22:49:06 mobileraker._Default INF>
Oct 28 22:49:06 mainsailos python[200517]: 2023-10-28 22:49:06 mobileraker._Default INF>
Oct 28 22:49:06 mainsailos python[200517]: 2023-10-28 22:49:06 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
lines 1-19/19 (END)...skipping...
● mobileraker.service - Companion app to enable push notifications on mobileraker
     Loaded: loaded (/etc/systemd/system/mobileraker.service; enabled; vendor preset: e>
     Active: active (running) since Sat 2023-10-28 20:06:53 CEST; 2h 47min ago
   Main PID: 200517 (python)
      Tasks: 1 (limit: 1389)
        CPU: 1min 50.188s
     CGroup: /system.slice/mobileraker.service
             └─200517 /home/pi/mobileraker-env/bin/python /home/pi/mobileraker_companio>

Oct 28 22:49:06 mainsailos python[200517]: 2023-10-28 22:49:06 mobileraker._Default INF>
Oct 28 22:49:06 mainsailos python[200517]: 2023-10-28 22:49:06 mobileraker._Default INF>
Oct 28 22:49:06 mainsailos python[200517]: 2023-10-28 22:49:06 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
Oct 28 22:49:29 mainsailos python[200517]: 2023-10-28 22:49:29 mobileraker._Default INF>
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
[... 353 lines cut ...]
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
lines 1-19/19 (END)

Experte9, 2023-10-31:
Es bleibt Übrigens dabei, dass es zu 90% nur eine Live Aktivität anzeigt und diese unzuverlässig aktualisiert (App im Hintergrund geöffnet) und ich nur von einem Drucker Push bekomme.

Clon1998, 2023-10-31:
> Es bleibt Übrigens dabei, dass es zu 90% nur eine Live Aktivität anzeigt und diese unzuverlässig aktualisiert (App im Hintergrund geöffnet) und ich nur von einem Drucker Push bekomme.

Alles klar. Ich schaue mir das ganze nochmal an. Die Aktualisierung funktioniert bei mir zuverlässig. Allerdings habe ich das bis jetzt immer nur mit einem Drucker getestet.

Experte9, 2023-11-18:
#Update zu Version 2.6.7-340
Es werden bei mehrfachen Druckern die Live Aktivitäten immer zuverlässig angezeigt. Die Aktualisierung funktioniert nur bei öffnen der App. 
Push Benachrichtigungen für den 2. Drucker, unverändert.

Clon1998, 2023-11-19:
> #Update zu Version 2.6.7-340 Es werden bei mehrfachen Druckern die Live Aktivitäten immer zuverlässig angezeigt. Die Aktualisierung funktioniert nur bei öffnen der App. Push Benachrichtigungen für den 2. Drucker, unverändert.

Gude,
bei den live activities war das Problem, dass die lib die ich verwendet habe nur immer eine activity mit Daten versorgen konnte. Das ist jetzt aber behoben.

Bezüglich der push meldungen, könntest du mal per [Testflight ](https://testflight.apple.com/join/h9JwuyR0) die Version 
2.6.8 (342) testen (Sobald die freigeschaltet ist). Hatte einen Fehler in der Registrierung zwischen Drucker und Handy für die Push Meldungen.

Experte9, 2023-11-22:
Kann es sein das die Live Aktivitäten nur mit der Push aktualisiert werden oder wenn man die App noch mal öffnet?

Für den 2. Drucker bekomme ich auch mit der Beta keine Push. Die Fehlermeldung in der App ist aber weg.

Clon1998, 2023-11-22:
> Kann es sein das die Live Aktivitäten nur mit der Push aktualisiert werden oder wenn man die App noch mal öffnet?
> 
> Für den 2. Drucker bekomme ich auch mit der Beta keine Push. Die Fehlermeldung in der App ist aber weg. 

Die live activity wird nur aktualisiert wenn du die App öffnest oder wenn eine Push-Meldung geschickt wird. Das hast du soweit richtig beschrieben.
Bzw. Bezüglich der Push Meldung, bei den Updates der live activity handelt es sich um andere Push Updates die im 2% Schritt gemacht werden.

Experte9, 2023-11-23:
Dann scheinen die Push aller 2% bei mir nicht zu funktionieren. 

Und ich muss mich korrigieren, die Fehlermeldung für den 2. Drucker ist nicht weg. Steht nach wie vor an und bekomme da auch mit der Test Benachrichtigung nichts.

Clon1998, 2023-11-23:
> Dann scheinen die Push aller 2% bei mir nicht zu funktionieren.
> 
> Und ich muss mich korrigieren, die Fehlermeldung für den 2. Drucker ist nicht weg. Steht nach wie vor an und bekomme da auch mit der Test Benachrichtigung nichts.

Habe gerade nochmal geschaut. Die Updates für live activities gehen ebenfalls in 5% Schritten.

Schick mal den Datenbank Dump von dem Drucker, bei dem die Meldung angezeigt wird.
Dafür einfach die URL im Browser öffnen:
`GET http://<IP/URL>/server/database/item?namespace=mobileraker&key=fcm`

Experte9, 2023-11-23:
Es läuft halt Multiklipper, heißt auf dem Raspberry sind 2 Drucker angeschlossen. der erste geht und der 2. halt nicht
Hier ist das Ergebnis:

`{"result": {"namespace": "mobileraker", "key": "fcm", "value": {"client": {"lastSeen": "2023-11-21T19:13:55.695678", "version": "v0.4.0-31-g81ae9a0"}, "25920149-fdd6-4f4b-a8b4-8876c4e8e430": {"created": "2023-10-28T13:42:43.820262", "lastModified": "2023-10-28T13:42:43.820301", "fcmToken": "cGtTbkOSaEZDl37Ez7OO8W:APA91bF34YMNUp_19nhllyx8Ip3O_2e4AMlMxX2t1depeR6J-Di1vRVCCXFnynrPovOdgTZzXgJkkqAkqlUYn9uWB4XFFB_scJAIH_RWRG0TW_aYz-5zNeh64dgsSRZZMnV6-G4MHqmJ", "machineName": "Schwarz", "language": "en", "settings": {"created": "2023-10-28T13:42:43.820264", "lastModified": "2023-10-28T13:53:05.459795", "progress": -1.0, "states": ["printing", "paused", "complete", "error", "cancelled"]}, "snap": {"progress": 0.0, "progress_live_activity": 0.0, "state": "error", "m117": "", "gcode_response": "cdf2989f37836af7475f86c35c1b0f031fdd6c1eb9707ef06db12da2167e967a"}}, "10c03b32-38e9-475e-b233-4cf06ce87261": {"created": "2023-11-03T16:44:45.928082", "lastModified": "2023-11-03T16:44:45.928196", "fcmToken": "cBjKpErcckNDoldGOeVN29:APA91bENtfUo2OYdulLN8xlNLxeagxjFDbp-84Jj430uZr8wCPz-wk8KjHmJl-9vQjUUERYGseOLNkfgaav-tHGb2-ydbsyeXbRpGAFdiOk0WdLOcNPNKd0LmQ6y9wULW5RrzJCnBqLP", "machineName": "Ender Wei\u00df", "language": "en", "settings": {"created": "2023-11-03T16:44:45.928085", "lastModified": "2023-11-23T09:16:34.303871", "progress": 0.5, "states": ["printing", "paused", "complete", "error"]}, "snap": {"progress": 0.0, "progress_live_activity": 0.0, "state": "error", "m117": "", "gcode_response": "cdf2989f37836af7475f86c35c1b0f031fdd6c1eb9707ef06db12da2167e967a"}}}}}`

Clon1998, 2023-11-23:
> Es läuft halt Multiklipper, heißt auf dem Raspberry sind 2 Drucker angeschlossen. der erste geht und der 2. halt nicht Hier ist das Ergebnis:
>
> `{"result": {"namespace": "mobileraker", "key": "fcm", [...]

Danke dafür!
Nur nochmal zu deinem Setup. Du hast 2 Drucker an einem Pi hängen. Auf dem PI laufen 2 Klipper + 2 Moonraker Instanzen.
Wie sieht es mit dem companion aus? Ein companion für beide Drucker oder zwei companion Instanzen?

Könntest du mir nochmal sagen von welchem Drucker der Datenbank dump ist? Bzw. schick mir am besten nochmal den dump vom anderen Drucker. Und verwendest du nur dein IPhone oder ebenfalls ein IPad/Tablet? Im ersten dump stehen nähmlich zwei Geräte mit jeweils anderen Push-Tokens (Und drucker Namen?).

Experte9, 2023-11-23:
Genau es sind 2 Drucker an einem Pi, 2x Klipper und 2x Moonraker, für companion läuft glaube ich nur eine Instanz. Ich hab das mit dem [Kihau ](https://github.com/dw-0/kiauh) Tool gemacht. das hat dann dein companion installiert.
der dump ist von `http://192.168.178.66/server/database/item?namespace=mobileraker&key=fcm` ich weiß nicht wie ich das dann Drucker bezogen mach. Gebe ja die Adresse von dem pi ein und dann wähle ich den jeweiligen Drucker. Oder soll ich noch die Ports mit rein schreiben? Die Drucker Sollten "Ender Weiß" und "Ender Schwarz" heißen.
Ich verwende nur mein iphone mit der App, kann sein das es durch den Wechsel auf die Beta App kommt?

Anbei noch 2 Übersichten
[image]
[image]

Experte9, 2023-11-24:
Also erster Drucker (Weiß):
`{"result": {"namespace": "mobileraker", "key": "fcm", "value": {"client": {"lastSeen": "2023-11-21T19:13:55.695678", "version": "v0.4.0-31-g81ae9a0"}, "25920149-fdd6-4f4b-a8b4-8876c4e8e430": {"created": "2023-10-28T13:42:43.820262", "lastModified": "2023-10-28T13:42:43.820301", "fcmToken": "cGtTbkOSaEZDl37Ez7OO8W:APA91bF34YMNUp_19nhllyx8Ip3O_2e4AMlMxX2t1depeR6J-Di1vRVCCXFnynrPovOdgTZzXgJkkqAkqlUYn9uWB4XFFB_scJAIH_RWRG0TW_aYz-5zNeh64dgsSRZZMnV6-G4MHqmJ", "machineName": "Schwarz", "language": "en", "settings": {"created": "2023-10-28T13:42:43.820264", "lastModified": "2023-10-28T13:53:05.459795", "progress": -1.0, "states": ["printing", "paused", "complete", "error", "cancelled"]}, "snap": {"progress": 0.7, "progress_live_activity": 0.7, "state": "printing", "m117": "", "gcode_response": "16594d648342d79a7818da67e4fb9f5e1751bf06a382ac1fb3e144536285e579"}}, "10c03b32-38e9-475e-b233-4cf06ce87261": {"created": "2023-11-03T16:44:45.928082", "lastModified": "2023-11-03T16:44:45.928196", "fcmToken": "cBjKpErcckNDoldGOeVN29:APA91bENtfUo2OYdulLN8xlNLxeagxjFDbp-84Jj430uZr8wCPz-wk8KjHmJl-9vQjUUERYGseOLNkfgaav-tHGb2-ydbsyeXbRpGAFdiOk0WdLOcNPNKd0LmQ6y9wULW5RrzJCnBqLP", "machineName": "Ender Wei\u00df", "language": "en", "settings": {"created": "2023-11-03T16:44:45.928085", "lastModified": "2023-11-23T09:16:34.303871", "progress": 0.5, "states": ["printing", "paused", "complete", "error"]}, "snap": {"progress": 0.5, "progress_live_activity": 0.7, "state": "printing", "m117": "", "gcode_response": "16594d648342d79a7818da67e4fb9f5e1751bf06a382ac1fb3e144536285e579"}}}}}`

zweiter Drucker (Schwarz):
`{"result": {"namespace": "mobileraker", "key": "fcm", "value": {"60c869e6-5fea-4b41-aeb3-979ff1624c07": {"apns": {"created": "2023-10-28T13:31:03.997453", "lastModified": "2023-10-28T13:31:03.997453", "liveActivity": "401dec8fb6bbca824c8b2edcdfa0850735d0b76bdabff76d6ecd0a1731930d8566de22c6dc637be4e85cf152a1f9afb8a7c0d14c4c6c7e5907a5b773ceb062ef6b43bf5de61040f29490114961a94a16"}}, "822e5e46-2a1a-487d-8d47-7ef20bb16556": {"settings": {"lastModified": "2023-10-28T13:53:05.461431", "states": ["printing", "paused", "complete", "error", "cancelled"], "progress": -1.0}}, "0b33b29b-5c21-40e0-b7aa-168c4d962982": {"settings": {"lastModified": "2023-11-23T09:16:34.302127", "states": ["printing", "paused", "complete", "error"], "progress": 0.5}, "apns": {"created": "2023-11-23T20:16:33.402303", "lastModified": "2023-11-23T20:16:33.402303", "liveActivity": "401d4f61a36d6426c2d64aba316a1cd1bfb95b3d845e229ba3467b37488579c52d8d09bd67b8be593d3bc44bcaac913d58635dc4d1218955d4ad6255a55f7a3b661ebe8fa1ab61810add75c64cd827b2"}}}}}`

Wo finde ich denn die `Mobileraker.conf` ? Habe unter `/home/pi/mobileraker_companion` heschaut aber nichts gefunden.

Clon1998, 2023-11-24:
> Also erster Drucker (Weiß):
>
> `{"result": {"namespace": "mobileraker", "key": "fcm", "value": {"client": {"lastSeen": "2023-11-21T19:13:55.695678", "version": "v0.4.0-31-g81ae9a0"}, [...]

Standardmäßig wird keine Mobileraker Config angelegt, da man die in der Regel nicht braucht. Aber in der Regel kann man die bei dem anderen configs unter printer_data/configs oder in den Ordner vom companion ablegen.
Wenn du bis jetzt keine hast, wird der companion aber bis jetzt immer nur vom 7125 Drucker die Push Mitteilungen senden können.

Schau Mal hier: https://github.com/Clon1998/mobileraker_companion/tree/main#companion---config

Du musst bei dir zwei printer configs anlegen und den companion Neustarten😉.
Kannst du dann auch hier posten und ich schaue nochmal drüber.

Experte9, 2023-11-24:
Also die 2 sind gerade fertig geworden. Es gab für beides ein Update. Die Push geht also für die Statusänderungen. 
Was immer noch nicht so recht geht ist das Update für die Live Aktualität. Es standen noch über 20 min bei dem einen und ca ne halbe Stunde bei dem anderen und plötzlich waren sie fertig. Als die Fertig Meldung kam wurde auch die Live Aktivität auf grün aktualisiert. 
Sonst von 0-99,9% kommt da kein Update. 
[image]

Clon1998, 2023-11-24:
> Also die 2 sind gerade fertig geworden. Es gab für beides ein Update. Die Push geht also für die Statusänderungen. Was immer noch nicht so recht geht ist das Update für die Live Aktualität. Es standen noch über 20 min bei dem einen und ca ne halbe Stunde bei dem anderen und plötzlich waren sie fertig. Als die Fertig Meldung kam wurde auch die Live Aktivität auf grün aktualisiert. Sonst von 0-99,9% kommt da kein Update.
> 
> Edit: version 2.6.8-342

Sehr gut! Bin da einfach selber nicht drauf gekommen mal zu Fragen ob der companion überhaupt zwei Drucker konfiguriert hat.

Die Live-Act. werden wie gesagt in 5% Schritten über den Companion aktualisiert und sind noch relativ neu. D.h. ETA/Remaining Time sind nur Schätzungen zu dem Zeitpunkt der Aktualisierung.  Dadurch kann es durchaus sein, dass am Ende noch zu viel Zeit auf dem Timer ist. Generell sind die Live-Act relativ statisch. Die Live-Act. selber machen eigentlich nichts außer die Daten anzeigen die ich ihnen gebe. 
Generell ist noch bisschen was für die Live-Act. zu tun. Gelegentlich passiert es zum Beispiel, dass die App 5 Live-Act. vom selben Drucker startet (Und ich bis jetzt nicht weiß warum lol).

Experte9, 2023-11-24:
Also jetzt ist das passiert was ich meine. Der Drucker ist fertig und ich bekomme auch die Meldung dafür aber die Live Aktivität weiß davon nichts und zählt ihren Countdown runter. 
Brauchst du davon irgendwelche Logs bevor ich was neues starte?
[image]

Ja das hatte ich auch schon, musste dann die Live Aktivitäten beenden und die App Neustarten. Ist aber nur 1x bisher passiert.

Clon1998, 2023-11-24:
> Also jetzt ist das passiert was ich meine. Der Drucker ist fertig und ich bekomme auch die Meldung dafür aber die Live Aktivität weiß davon nichts und zählt ihren Countdown runter. Brauchst du davon [...]

Kannst du gerne hochladen

Experte9, 2023-11-24:
Hoffe es hilft.
[mobileraker.log](https://github.com/Clon1998/mobileraker/files/13462302/mobileraker.log)

Ansonsten kannst du das Ticket auch gern zu machen. Push geht ja vernünftig und an den Live Aktivitäten arbeitest du ja noch.
(Falls du mal jemand brauchst der was testet kannst du dich auch gern melden)

Experte9, 2024-01-11:
Mit den neuen Updates geht's für mich wunderbar!
Danke für deine Arbeit

closed 2024-01-11 by Experte9