Clon1998/mobileraker #278: Flow rate interface in iOS keeps changing lines

thomazff, 2023-10-24:
Hi, I use mobileraker on em iphone 13 mini. Maybe it is my small screen. But flow rates don't fit in one line when above a few digits. So the screen keeps adjusting and moving everything below down and up.

If it could stay fixed at the "larger" position it would be really great.

Will try to screenshot or record the screen on my next print.

cramsvik, 2024-01-15:
https://github.com/Clon1998/mobileraker/assets/7917520/f14f094c-32da-4286-83bc-cf9ca5b88465

I see the same issue. Some adjustment to the widths of the three columns would probably fix the issue. The rightmost one doesn't need much space now that it is renamed to ETA

spitfire, 2024-03-07:
I am seeing the same issue. Maybe it is because I have increased system font size. If there's a risk of value not fitting could you just move it to another row instead of having it switch it all the time?

Clon1998, 2024-03-21:
Just pushed an update that makes the cells in the table enforce at most 1 text line. It also tries to adjust the fontsize

commit 69e4b608 (2024-03-21): Fix #278 - Ensured Toolhead info tables cells have an adaptive font size
closed 2024-03-25 by Clon1998