Clon1998/mobileraker #279: Connection issue

kenjk, 2023-10-24:
I installed this app on my Samsung S23 Ultra, entered the IP address as shown on my Neptune 4 Pro printer, as well as the API key found in the settings in Fluidd but it will not connect. What else do I need to do? Error is "SocketException: HTTP connection timed out after 0:00:20.000000, host: 169.254.243.159, port:443." What do I need to do? Any help would be greatly appreciated.

Clon1998, 2023-10-25:
Hey
Are you using the simple or advanced mode to add a printer?

kenjk, 2023-10-25:
I've tried both. Could you possibly walk me through the steps in detail? When it asks for the HTTP endpoint of Moonraker and I click on it, it brings up it's own details: ws://169.254.243.159/websocket. Is that what I leave in there? If not, where do I find the correct info? At the bottom under HTTP HEADERS, do I need to add any? And if so, what are they? When I click on Test Connection these are the results: Http Endpoint - Test: Invalid  SocketException: HTTP connection timed out after 0:00:5.000000, host: 169.254.243.159, port:80.  Websocket Enpoint - Test: Invalid  SocketException: HTTP connection timed out after 0:00:5.000000, host: 169.254.243.159, port:80.  What am I doing wrong? I'd really like to get this to work but I am confused. Thanks for you help.

kenjk, 2023-10-25:
I guess I should add one more thing: does my printer have to be on wifi? The Neptune 4 Pro doesn't have wifi, so I am only logged into it through Moonraker through it's IP address.  If I need wifi, do you know how to add that to this printer, or how I can find out how? Thanks again.

Clon1998, 2023-10-25:
Hey,
WiFi on the printer is not required to run the app. So that you know – the only requirement is that the printer is available in your local area network. You can test that by simply trying to open the web interface of your printer via your phone (E.g. [http://169.254.243.159](http://169.254.243.159)). If your phone cannot open the web interface the app won't be able to connect either. If your phone can open the web interface, you can proceed with adding it to the app. I suggest you completely remove all of your previous tries to add the printer. Afterward, you use the **Simple**  flow of the add printer page since it will take care of configuring the defaults for you. To stick to the example from before, you just need to insert `169.254.243.159` into the text field and proceed with pressing the Test connection button.

The advanced mode is not recommended. It offers more options that can be configured to connect to the printer. Like a specific HTTP and WebSocket address or HTTP Headers. All of this is only required for special printer and network setups.

kenjk, 2023-10-25:
Hi, That clarifies a lot, thank you. I have tried to open the web interface on my phone but it always says connection timed out. Any idea why I can't open it on my phone? Is there only a specific web browser that can open it or is there something else you might know of? Thanks again for all your assistance Sent from my Bell Samsung device over Canada's largest network.

Clon1998, 2023-10-25:
Your printer needs to be in the same network as the device that tries to open the web interface.

Based on your statement that you have directly connected your printer to your computer using an Ethernet cable, I presume that your printer is not connected to the network. To connect the printer to the network, you have two options: Firstly, you could connect the printer to your Router or a switch via Ethernet. Secondly, you could install a Wi-Fi antenna on the printer's host computer, which will enable it to connect to your network via WiFi.

kenjk, 2023-10-25:
Ok, so even though I can access my printer via its IP address on my desktop computer, which is connected to the web via WiFi, that doesn't mean my printer is connected to my network? The printer itself doesn't have WiFi capabilities, so instead of connecting its Ethernet port to my computer's Ethernet port, I need to connect it to a Wi-Fi /Ethernet adapter in order for network connection? Sent from my Bell Samsung device over Canada's largest network.

kenjk, 2023-10-25:
> Exactly.
> Imagine a telephone line connecting House B to your house. You can call House B because it is directly connected to your house. But if someone else tries to call House B, they won't be able to get through unless House B is connected to a central hub or call center.

Thank you so much for all your help. I will try what you suggested and see if that will work. I appreciate your timely responses. Thanks, KenSent from my Bell Samsung device over Canada's largest network.

kenjk, 2023-10-25:
I set it up with a wifi/Ethernet adapter connected to my home network wifi. Everything works perfectly now. Thank you so much for your help! Now I can control my printer remotely, especially handy if something goes haywire during a print I can shut it down easily. Any suggestions for the best camera setup? Webcam or basic indoor camera? Suggestions? Thanks a bunch! Ken Sent from my Bell Samsung device over Canada's largest network.

milos975, 2023-10-30:
> I guess I should add one more thing: does my printer have to be on wifi? The Neptune 4 Pro doesn't have wifi, so I am only logged into it through Moonraker through it's IP address. If I need wifi, do you know how to add that to this printer, or how I can find out how? Thanks again.

You should try your IP adress and then :4408, because this is the port for communication. Example 192.168.1.19:4408
Try this.

milos975, 2023-10-30:
Hi there,
I would like to know how to connect printer when I am not at home/same LAN network. How can I setup app to connect to printer?
Thanks.

Clon1998, 2023-10-30:
> Hi there, I would like to know how to connect printer when I am not at home/same LAN network. How can I setup app to connect to printer? Thanks.

See the FAQ:
https://github.com/Clon1998/mobileraker/blob/master/docs/faq.md#%EF%B8%8F-remote-printer-access

rallegade, 2023-11-04:
just to chip in on this as well.
@kenjk It seems like your printer is having an APIPA address (an address the machine gives itself when one cannot be obtained from a DHCP server) which means that it is not able to communicate with your other devices on the network.
Start by troubleshooting if the printer can get an IP from your router first.

kenjk, 2023-11-11:
@Clon1998 , my Mobilraker app isn't working anymore, says my trial has ended. Is a subscription plan the only way this will work?

Clon1998, 2023-11-11:
> @Clon1998 , my Mobilraker app isn't working anymore, says my trial has ended. Is a subscription plan the only way this will work? 

Hey,
I assume you are talking about the remote access provided via Ocotoeverywhere. 
Ocotoeverywhere is not affiliated with mobileraker and requires a separate subscription in order to unlock all features (As stated within mobileraker on the linking sheet).
I also recently added obico as an alternative to Ocotoeverywhere (Also another service that might require a subscription for certain features/unlimited network access).

The last option is a custom remote access solution either via a reverse proxy or via your own VPN.

Ocotoeverywhere: Ocotoeverywhere.com
Obico: obico.io
Private VPN: https://youtu.be/vq-VE-9t13k?si=Mbu953--vsSB6GNU

kenjk, 2023-11-11:
Thanks for the reply. 
So right now when I open my Mobilraker app, I don't get any info. It won't connect to my printer and I can't do anything with it. Is that normal without the Octoeverywhere plan?
Yes, I was referring to the access provided by my trial with Octoeverywhere. 
But do I need Octoeverywhere, or any alternative for that matter, in order to use Mobilraker? 
What can I do with Mobilraker by itself as a standalone app?

[bc1e7649-8f28-4e14-8c36-35129b98c597]

Clon1998, 2023-11-11:
> Thanks for the reply. 
> 
> So right now when I open my Mobilraker app, I don't get any info. It won't connect to my printer and I can't do anything with it. Is that normal without the Octoeverywhere plan?
> 
> Yes, I was referring to the access provided by my trial with Octoeverywhere. 
> 
> But do I need Octoeverywhere, or any alternative for that matter, in order to use Mobilraker? 
> 
> What can I do with Mobilraker by itself as a standalone app?
> 
> 
> @Clon1998 [https://github.com/Clon1998] , my Mobilraker app isn't working anymore, says my trial has ended. Is a subscription plan the only way this will work?
> 
> Hey,
> 
> I assume you are talking about the remote access provided via Ocotoeverywhere.
> 
> Ocotoeverywhere is not affiliated with mobileraker and requires a separate subscription in order to unlock all features (As stated within mobileraker on the linking sheet).
> 
> I also recently added obico as an alternative to Ocotoeverywhere (Also another service that might require a subscription for certain features/unlimited network access).
> 
> The last option is a custom remote access solution either via a reverse proxy or via your own VPN.
> 
> Ocotoeverywhere: Ocotoeverywhere.com
> 
> Obico: obico.io
> 
> Private VPN: https://youtu.be/vq-VE-9t13k?si=Mbu953--vsSB6GNU [https://youtu.be/vq-VE-9t13k?si=Mbu953--vsSB6GNU]
> 
> 
> 
> 
> [bc1e7649-8f28-4e14-8c36-35129b98c597]

You can use the app normally while you are at the same network as your printer. 
My printers don't even have any display anymore since I got my phone on me anyway.
In addition I am using a VPN to remotely connect to my network and check on my printers.

kenjk, 2023-11-11:
So in order to view my printer when I am away from home using Mobilraker I will need some kind of access such as Octoeverywhere or Obico?

[58e31dd8-7c24-42df-9e2b-7b8967b8316d]

Clon1998, 2023-11-11:
Yes, you will always need some kind of connection to your home network/printer. Either via the two mentioned services (obico or octoeverywhere) or you set up your own solution to tunnel back to your own network. There are plenty of options and also a lot of tutorials on how to set up either a reverse proxy or a VPN to connect to your home network.
I already posted one video on how to setup a VPN for remote printer access. [Mainsail](https://docs.mainsail.xyz/overview/quicktips/remote-access) also has a nicely written article that covers this topic.

However, the app will continue to send you push notifications even if you are not at home.

kenjk, 2023-11-11:
Aha, good to know. 
Thanks for the help again.

Cheers.

[3d6a53cd-c815-4fe3-9a1f-ea73e39951b7]

kenjk, 2023-11-16:
Hey Patrick, I am trying to link Obico with Klipper for use in Mobileraker but when I SSH into Klipper and install the package, when I get to the part where it asks for Moonraker's configurations, the file "home/mks/printer_data/config/moonraker.conf" comes up as an error saying that it doesn't exist or can't be changed. Any idea what the problem is and how to fix it? Everything goes fine right up until that section of my ssh.

[46d08328-f9ec-4e63-be8d-ef6c3b15504d]

Clon1998, 2023-11-16:
You might want to get in contact with the obico team to get that issue resolved since their installation program is unable to locate the moonraker config file.

kenjk, 2023-11-17:
Ok so I have Obico linked with my Klipper through Fluiid now. But now how do I link Obico with Mobileraker? Everytime I open Mobileraker it just goes to a screen that says I'm not supporting Octoeverywhere anymore. 

[20195fd4-46d3-4e35-8072-bb281e0190f5]

Clon1998, 2023-11-17:
> Ok so I have Obico linked with my Klipper through Fluiid now. But now how do I link Obico with Mobileraker? Everytime I open Mobileraker it just goes to a screen that says I'm not supporting Octoeverywhere anymore. 
> 
> 
> 
> You might want to get in contact with the obico team to get that issue resolved since their installation program is unable to locate the moonraker config file.
> 
> 
> 
> 
> [20195fd4-46d3-4e35-8072-bb281e0190f5]

You will need to remove OE and add Obico. See attached video.
https://github.com/Clon1998/mobileraker/assets/10357775/2770e346-5e8a-4bdd-95d0-d4090a0d95b7