Clon1998/mobileraker #280: CA issue with HTTPS even with real certificates

rallegade, 2023-11-04:
## Bug Report

### Description
No matter if I am using a self-signed certificate with the given CA which has issued the certificate installed on the device or even a certificate issued by Lets Encrypt which is in the root CA folder of all devices, i still get the error "HandshakeException: Handshake error in client (OS Error: CERTIFICATE_VERIFY_FAILED: unable to get local issuer certificate(handshake.cc:393))

### Steps to Reproduce
- Install a certificate issued by either a local CA which the devices trusts or even a certificate issued by Lets Encrypt for mainsail nginx instance
- connect to the printer with https through mobileraker
- I can get the connection to work if the "trust self signed certificates" is checked, even though this shouldn't be the case as I am using a non self signed certificate or a self signed certificate which is trusted by the device.
- If I use trust self signed certificates I still have the same error for my connection to the webcam

### Expected Behavior
There should not be any issues with trusting the certificates in the app when both I have tried are trusted in my browsers.

### Version Information

- Device-OS (Android/iOS): Android 13
- Mobileraker: 2.6.6-324
- Klipper: v0.11.0-307-gbee7ec72
- Moonraker: v0.8.0-190-ge03dbe32

###Mobileraker log
[mobileraker_2023-11-04T14-46-49.log](https://github.com/Clon1998/mobileraker/files/13256884/mobileraker_2023-11-04T14-46-49.log)

Clon1998, 2023-11-04:
Related to #193.

If I recall it correctly, flutter always uses the default trust store of the device when handling certificates without any installed custom certificates. That is the reason why you need to disable certificate checks in order to get the installed custom certificate to work aka. trust self-signed certificates. 
Since this is rather an edge case I did not invest any time into adding logic to actually install/trust custom CAs on mobileraker (#193).

Regarding the webcams, I will have a look to ensure these web clients also ignore cert-validation errors if the "Trust self-signed" is enabled.

rallegade, 2023-11-04:
That's what I expected too which is why I tried installing a certificate which is issued by letsencrypt, and which I know should be in the standard trusted CA store. 
Everything else I have running through reverse proxies are using these certs and are accepted by all the clients as valid as they are issued by letsencrypt r3 CA.

It might be niche with the self signed but it's not niche to have certificates issued by letsencrypt supported.

closed 2023-12-08 by Clon1998