Clon1998/mobileraker #282: Lifetime supporter between IOS and Android

reapola, 2023-11-12:
Purchased Lifetime supporter on iPhone and wish to use the same benefits on my Android tablet.  

How can this be done?

Clon1998, 2023-11-12:
Hey,
as of now this is not possible within the app.
However, I can manually unlock the Android version.

For that I will need logs from both of your devices. You can either upload them here or mail them to me via help@mobileraker.com

reapola, 2023-11-12:
Thanks for the quick reply.

 Both files attached, let me know if you need anything else.

[IOS_mobileraker_2023-11-12T10-09-48.log.txt](https://github.com/Clon1998/mobileraker/files/13327718/IOS_mobileraker_2023-11-12T10-09-48.log.txt)
[android_mobileraker_2023-11-12T10-13-37.log](https://github.com/Clon1998/mobileraker/files/13327720/android_mobileraker_2023-11-12T10-13-37.log)

reapola, 2023-11-12:
Mistake

Clon1998, 2023-11-12:
Both accounts should be active now.
Enojy and thanks for your support!

closed 2023-11-12 by Clon1998