Clon1998/mobileraker #283: Wencam : erreur 404

mick69000, 2023-11-13:
## Bug Report

### Description
imossible de connecter la webcam

### Description
J'ai une erreur 404 au niveau de la webcam

### Steps to Reproduce

### Expected Behavior
affichage de la webcam

### Screenshots
[image]
[image]

### Version Information

- Device-OS (iOS): [iOS 17.1.1]
- Mobileraker: mobile=> [2.6.6-234] klipper =>[v0.4.0-29-g5a4cae4a]
- Klipper: [v0.11.0-311-g9e765dae]
- Moonraker: [v0.8.0-197-g975011d1]

### Debug Logs

[mobileraker_2023-11-12T19-03-04.log](https://github.com/Clon1998/mobileraker/files/13339720/mobileraker_2023-11-12T19-03-04.log)

### Additional Context
Système installé sur un mini pc (Ubuntu 23.04 lunar).
Je passe par un nom de domaine ( elf.mickahome.fr)

Clon1998, 2023-11-13:
I don't speak nor understand a single line of French.
However, judging by your screenshots the webcam you configured is not pointing to the correct endpoint.
You can either define a webcam via its relative URI (To the printer address) or use an absolut URI (E.g. `http://my-cool-printer-cam.io/some/path?alsoAQuery`).

Without more details about the webcam config inside mobileraker and also the configs on your host system, I can not do much.

rumym, 2023-12-16:
Looks related to the bug I just filed - https://github.com/Clon1998/mobileraker/issues/299

Clon1998, 2023-12-16:
> Looks related to the bug I just filed - #299

Not quite; on this issue, the webcam was not found. Most likely, the cam is pointing to the wrong URL.
In #299 you are having an SSL issue.

rumym, 2023-12-16:
I am not using SSL when connecting to local. I have attached the
configuration in the app as well as the error I am getting.

Clon1998, 2024-01-19:
Closed due to inactivity for two months. Feel free to reopen if additional information is provided.

closed 2024-01-19 by Clon1998