Clon1998/mobileraker #284: Webcam display doesn't reflect server settings

jamincollins, 2023-11-16:
## Bug Report

### Description
The mobile app display of the webcam is not flipped as configured in the server settings.

### Steps to Reproduce
1. Configure webcam flipped vertically.
2. Access printer in mobile 

### Expected Behavior
Display of webcam in mobile should be the same as the server.

### Screenshots
[image]

[image]

### Version Information

- Device-OS (Android/iOS): Android 14
- Mobileraker: 2.6.6-324
- Klipper: v0.12.0-0-g0d67d9c4
- Moonraker: v0.8.0-197-g975011d1

### Debug Logs
Will have to upload separately

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

jamincollins, 2023-11-16:
[mobileraker_2023-11-11T19-03-11.log](https://github.com/Clon1998/mobileraker/files/13371091/mobileraker_2023-11-11T19-03-11.log)
[mobileraker_2023-11-11T19-54-37.log](https://github.com/Clon1998/mobileraker/files/13371092/mobileraker_2023-11-11T19-54-37.log)
[mobileraker_2023-11-12T17-47-51.log](https://github.com/Clon1998/mobileraker/files/13371093/mobileraker_2023-11-12T17-47-51.log)
[mobileraker_2023-11-15T12-52-44.log](https://github.com/Clon1998/mobileraker/files/13371094/mobileraker_2023-11-15T12-52-44.log)
[mobileraker_2023-11-15T17-56-55.log](https://github.com/Clon1998/mobileraker/files/13371095/mobileraker_2023-11-15T17-56-55.log)

Clon1998, 2023-11-16:
Hey,
thanks for reporting this issue.
This was already brought to my attention and fixed in 6249513c0806a71802b671e690fe4ff62aa79612.
It is included in version 2.6.7 which is slowly rolling out.

jamincollins, 2023-11-16:
I've attempted to install the 2.6.7 release APK, but my phone is telling me it is invalid.

Clon1998, 2023-11-16:
> I've attempted to install the 2.6.7 release APK, but my phone is telling me it is invalid.

Seems like the GitHub action is not properly signing the APK causing problems if you installed the app via the play store previously.

I went ahead and replaced the APK with a properly signed one. Let me know if it is working now.

jamincollins, 2023-11-16:
Different error now, says it conflicts with an existing package...

Uninstalling the Play Store version allowed installing the new version.

Clon1998, 2023-11-16:
> Different error now, says it conflicts with an existing package...
> 
> 
> 
> Uninstalling the Play Store version allowed installing the new version.
> 
> 
> 
> 

Great! I assume it complaints about the different package because the play store version isn't an APK.
Just be aware now that you might loose auto updates via play store because you side loaded the app.

Also could you confirm for me that the cam is working/flipped correctly now?

jamincollins, 2023-11-16:
> Also could you confirm for me that the cam is working/flipped correctly now?

Yes, the camera is working correctly.

closed 2023-11-16 by jamincollins