Clon1998/mobileraker #286: option to export current config

jamincollins, 2023-11-16:
## Feature Request

### Problem Description
I've recently found it necessary to switch between the Play Store version of the application and a direct (side) load.  Doing so required removing the Play Store version.  It seems this removed all configuration (printers, preferences, etc).

### Proposed Solution
Provide an option to export the current configuration to a file(s) on the device.  Along with an option to import the same file(s).

### Alternatives Considered

### Additional Context

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x] Provided a clear description of the problem you're facing.
- [x] Clearly outlined the proposed solution or feature.
- [x] Mentioned any alternative solutions or ideas you've considered.
- [x] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2023-11-16:
That's a good idea.
However, there are not a lot of things stored locally. Only the printer connection data, remote connection settings and app settings.

Webcams and machine Settings (Inverts, Speeds, Move-Steps...) are stored via moonraker.

jamincollins, 2023-11-16:
Yes, but setting up a few printers is rather tedious.

jamincollins, 2023-11-16:
Could also be useful to setup one device (phone), export the settings, and import on another device (tablet).

wikolii, 2024-01-20:
This would be very helpful specially for disabling or hiding macros when setting up multiple printer. Right now you have to manually do that on all printers. thanks!