Clon1998/mobileraker #287: [neopixel] parsing error in klipper.cfg

vstruk, 2023-11-22:
Mobileraker can't parse a valid section in klipper cfg.

### How to reproduce
Add the following lines to your klipper.cfg, restart klipper, try to connect via mobileraker

```
[neopixel top]
pin: z:P2.0
chain_count: 36
```

Observe the error:
``
Parsing of configfile failed:
type 'String' is not a subtype of 'List<dynamic>'
...
``

### Version Information

- IOS
- Mobileraker:  2.6.7
- Klipper: v0.10.0-320-g7e654aed
- Moonraker: v0.8.0-185-gb8ce199

vstruk, 2023-11-23:
Please find the log attached.
[mobileraker_2023-11-22T21-20-52.log](https://github.com/Clon1998/mobileraker/files/13446306/mobileraker_2023-11-22T21-20-52.log)

Clon1998, 2023-11-23:
> Please find the log attached. [mobileraker_2023-11-22T21-20-52.log](https://github.com/Clon1998/mobileraker/files/13446306/mobileraker_2023-11-22T21-20-52.log)

Thanks,
I just pushed changes to resolve the issue in the future. For now, you might want to update Klipper since the next release that will include the fix is scheduled for december.

vstruk, 2023-11-26:
Thank you! 
I'll test it on release.

closed 2023-11-26 by vstruk