Clon1998/mobileraker #288: Thumbnail parsing error

Clon1998, 2023-11-25:
Hello
I have the same problem on Android phone
Here is error dump

```
Exception:
 type 'String' is not a subtype of type 'num' in type cast

#0      _$$GCodeThumbnailImplFromJson (package:common/data/dto/files/gcode_thumbnail.g.dart:11)
#1      new _$GCodeThumbnailImpl.fromJson (package:common/data/dto/files/gcode_thumbnail.freezed.dart:154)
#2      _$GCodeThumbnailFromJson (package:common/data/dto/files/gcode_thumbnail.freezed.dart:18)
#3      new GCodeThumbnail.fromJson (package:common/data/dto/files/gcode_thumbnail.dart:30)
#4      _sortedThumbnails.<anonymous closure> (package:common/data/dto/files/gcode_file.dart:131)
#5      MappedIterable.elementAt (dart:_internal/iterable.dart:375)
#6      ListIterator.moveNext (dart:_internal/iterable.dart:344)
#7      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189)
#8      new _GrowableList.of (dart:core-patch/growable_array.dart:150)
#9      new List.of (dart:core-patch/array_patch.dart:47)
#10     IterableExtension.sortedBy (package:collection/src/iterable_extensions.dart:66)
#11     _sortedThumbnails (package:common/data/dto/files/gcode_file.dart:132)
#12     _$$GCodeFileImplFromJson (package:common/data/dto/files/gcode_file.g.dart:40)
#13     new _$GCodeFileImpl.fromJson (package:common/data/dto/files/gcode_file.freezed.dart:439)
#14     _$GCodeFileFromJson (package:common/data/dto/files/gcode_file.freezed.dart:18)
#15     new GCodeFile.fromJson (package:common/data/dto/files/gcode_file.dart:103)
#16     FileService._parseDirectory.<anonymous closure> (package:common/service/moonraker/file_service.dart:373)
#17     new _GrowableList.generate (dart:core-patch/growable_array.dart:136)
#18     FileService._parseDirectory (package:common/service/moonraker/file_service.dart:369)
#19     FileService.fetchDirectoryInfo (package:common/service/moonraker/file_service.dart:232)
<asynchronous suspension>
#20     FutureHandlerProviderElementMixin.handleFuture.<anonymous closure>.<anonymous closure> (package:riverpod/src/async_notifier/base.dart:348)
<asynchronous suspension>
```

_Originally posted by @Danix6 in https://github.com/Clon1998/mobileraker/issues/189#issuecomment-1826268053_

Clon1998, 2023-11-25:
@Danix6, I went ahead and created a new issue since the original one (#189) was not related to your issue.

Could you please provide me:
- The app log files, latest or all as Zip (App Settings Page -> Scroll to the bottom -> Debug Logs)
- The Version of Mobileraker (Either open the changelog page or at the bottom of the App Settings page)
- The Version of Moonraker

willkaxxx, 2023-11-26:
Hi, @Clon1998 I would also mention bug when thumbnail is not updated if there is already a file with same name.

How to reproduce: 
 1. Upload a file with name tmp.gcode
 2. Upload another gcode, with another thumbnail, but call it as firs one, tmp.gcode
 3. Old image will remain in app
 
 Fluid image:
[image]

App image:
[image]

Maybe it can be fixed in scope of this issue

Regards, Oleksii

Clon1998, 2023-11-26:
> Hi, @Clon1998 I would also mention bug when thumbnail is not updated if there is already a file with same name.
> 
> How to reproduce:
> 
> 1. Upload a file with name tmp.gcode
> 2. Upload another gcode, with another thumbnail, but call it as firs one, tmp.gcode
> 3. Old image will remain in app
> 
> Fluid image: ![image](https://private-user-images.githubusercontent.com/48694879/285655530-8d22e4f4-a916-48fb-9b0b-09aacc4c3e59.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwMTAwNzgsIm5iZiI6MTcwMTAwOTc3OCwicGF0aCI6Ii80ODY5NDg3OS8yODU2NTU1MzAtOGQyMmU0ZjQtYTkxNi00OGZiLTliMGItMDlhYWNjNGMzZTU5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI2VDE0NDI1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTExYjk1MDU1M2NjNWM5MWRiZWFiZTlhOThmYjJlMTM3MjEwY2UyZDU0ZTQ2OWFiNWZlZmYyOTVhZDZmMGE0MTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WcSq5HNZciWtbeb5NFSZ4M0E736tVFEmdxlf_-Yvijg)
> 
> App image: [Screenshot_20231126_155522](https://github.com/Clon1998/mobileraker/assets/48694879/2ea93809-48f3-4712-a9e4-b4f02aa49107)
> 
> Maybe it can be fixed in scope of this issue
> 
> Regards, Oleksii

Sure, upon a quick review of the code, I have found that this behavior is intentional. The app caches all the network images for the duration of the app session. Therefore, unless you force close and reopen the app, the image will not be updated. However, I will look into the possibility of modifying the caching process to also consider the upload time of the file for thumbnail caching.

Clon1998, 2023-11-26:
@willkaxxx 
Fixed in 4c5a0d73217079e3313dcf4b8d0eec9b1052ff12

vcdx71, 2023-12-01:
Thanks, Here's the requested json dump.

{
    "result": {
        "dirs": [
            {
                "modified": 1700874360.855523,
                "size": 4096,
                "permissions": "rw",
                "dirname": ".thumbs"
            }
        ],
        "files": [
            {
                "modified": 1699424397,
                "size": 8991226,
                "permissions": "rw",
                "filename": "Spool_holder_Hyper PLA_1H49m.gcode",
                "uuid": "14c61c8e-ce33-4a7e-9a54-8001e7f87be6",
                "slicer": "Creality",
                "slicer_version": "4.3.5.5641",
                "gcode_start_byte": 77198,
                "gcode_end_byte": 8991180,
                "layer_count": 475,
                "object_height": 95.2,
                "estimated_time": 6542,
                "layer_height": 0.2,
                "first_layer_height": 0.2,
                "first_layer_extr_temp": 220,
                "first_layer_bed_temp": 50,
                "filament_name": "Hyper PLA_1.75",
                "filament_type": "PLA",
                "filament_total": 33094.6,
                "filament_weight_total": 194.596248,
                "thumbnails": [
                    {
                        "width": "96",
                        "height": "96",
                        "size": 5961,
                        "relative_path": ".thumbs/Spool_holder_Hyper PLA_1H49m-96x96.png"
                    }
                ],
                "print_start_time": null,
                "job_id": null
            },
            {
                "modified": 1700948747.7139847,
                "size": 7316454,
                "permissions": "rw",
                "filename": "UltimateWaterproofCase_V6.gcode",
                "uuid": "6e9c9593-12a8-4125-baf8-e854bc46c507",
                "slicer": "Unknown",
                "gcode_start_byte": 60690,
                "gcode_end_byte": 7316439,
                "print_start_time": 1700948988.686949,
                "job_id": "000001"
            },
            {
                "modified": 1699424397,
                "size": 2557664,
                "permissions": "rw",
                "filename": "scraper-K1 Max_0.4_Hyper PLA_28m.gcode",
                "uuid": "acfb2529-38c7-4b0b-97e5-3a38e6d8ea61",
                "slicer": "Creality",
                "slicer_version": "4.3.7.6618",
                "gcode_start_byte": 45756,
                "gcode_end_byte": 2557650,
                "layer_count": 75,
                "object_height": 7.65,
                "estimated_time": 1721,
[... 140 lines cut ...]
                "filament_total": 4150.41,
                "filament_weight_total": 24.4044108,
                "thumbnails": [
                    {
                        "width": "96",
                        "height": "96",
                        "size": 5268,
                        "relative_path": ".thumbs/Power_Cable-Large_Retaining_Ring-Ender-3 Max Neo_0.4_Generic-PLA_1h41m-96x96.png"
                    }
                ],
                "print_start_time": null,
                "job_id": null
            }
        ],
        "disk_usage": {
            "total": 6946127872,
            "used": 903766016,
            "free": 5689516032
        },
        "root_info": {
            "name": "gcodes",
            "permissions": "rw"
        }
    }
}

Clon1998, 2023-12-01:
> > Thanks @vcdx71. What moonraker version is your printer using?
> 
> Thanks, it's v0.8.0-204-gfb15b2a3

Interesting. I am also on 0.8.0-204 and don't have the problem. Therefore, I checked the code of moonraker, and I suspect that the thumbnail's width/height datatype depends on the slicers you used.
Therefore,  do you mind telling me what slicers you used?

vcdx71, 2023-12-01:
> Interesting. I am also on 0.8.0-204 and don't have the problem. Therefore, I checked the code of moonraker, and I suspect that the thumbnail's width/height datatype depends on the slicers you used. [...]

Creality Print.

vcdx71, 2023-12-01:
> > slicer
> 
> Could you share a sliced file with me? Just a relatively small one in case you have one on hand. I want to debug that more in-depth.
[UltimateWaterproofCase_V6.gcode.zip](https://github.com/Clon1998/mobileraker/files/13531845/UltimateWaterproofCase_V6.gcode.zip)

For sure thanks..

Clon1998, 2023-12-01:
> > > slicer
> > 
> > 
> > Could you share a sliced file with me? Just a relatively small one in case you have one on hand. I want to debug that more in-depth.
> > [UltimateWaterproofCase_V6.gcode.zip](https://github.com/Clon1998/mobileraker/files/13531845/UltimateWaterproofCase_V6.gcode.zip)
> 
> For sure thanks..

Thanks.
Sadly the file is one of the only files that does not incldue thumbnail data.
Therfore, could you share `Spool_holder_Hyper PLA_1H49m.gcode` with me?

Clon1998, 2023-12-01:
> > > > > > > slicer
> > > > > > 
> > > > > > 
> > > > > > Could you share a sliced file with me? Just a relatively small one in case you have one on hand. I want to debug that more in-depth.
> > > > > > [UltimateWaterproofCase_V6.gcode.zip](https://github.com/Clon1998/mobileraker/files/13531845/UltimateWaterproofCase_V6.gcode.zip)
> > > > > 
> > > > > 
> > > > > For sure thanks..
> > > > 
> > > > 
> > > > Thanks. Sadly the file is one of the only files that does not incldue thumbnail data. Therfore, could you share `Spool_holder_Hyper PLA_1H49m.gcode` with me?
> > > > [Spool_holder_Hyper PLA_1H49m.gcode.zip](https://github.com/Clon1998/mobileraker/files/13531949/Spool_holder_Hyper.PLA_1H49m.gcode.zip)
> > > 
> > > 
> > > For sure, this one however came on the printer by default during its latest FW update.
> > 
> > 
> > In that case I assume you are on a K1 with firmware managed by creality and not the offical moonraker open source repo?
> 
> It's a K1 Max, the FW is the latest from creality where they give you root access and an install for moonraker, I then updated moonraker after their install was up. Everything in your app works great, including the camera, just can't see the gcode files.

Alright, thanks for all the feedback and gcodes you provided.
I am not 100% what exactly is going on and why these GCode files behave so weird. I tried to parse them manually with the same script that moonraker uses but both of the files you provided did not produce any thumbnail. 
Therefore, I decided to just "fix" this within my app via [6ee8917](https://github.com/Clon1998/mobileraker/commit/6ee891777830e8ca02a533a4ef918e255a0b143c).
The fix will be included in the next release.

vcdx71, 2023-12-01:
> Alright, thanks for all the feedback and gcodes you provided. I am not 100% what exactly is going on and why these GCode files behave so weird. I tried to parse them manually with the same script [...]

Thanks so much!

Arksine, 2023-12-01:
FWIW, the official Moonraker repo does not support extracting metadata from Creality Print Slicer, and will not do so until they resolve the GPL issues around it.  I suspect that the metadata reported here is leftover from running Creality's fork of Moonraker.

closed 2023-12-08 by Clon1998