Clon1998/mobileraker #292: Not saving printer setting when webcams configured in moonraker

jelleh, 2023-11-30:
## Bug Report

### Description
Printer settings will not save when there are webcams configured in moonraker

### Steps to Reproduce
1. Go to 'hamburger menu'
2. Click on 'settings'
3. Change printer displayname
4. Click the save icon
5. See the error in debug log.
6. printer displayname is not changes/saved

Error 400: Cannot overwrite webcam 'camera_top' sourced from Moonraker

### Expected Behavior
When you have configured webcams in moonraker, you can not save printer settings.
When i remove the webcam from moonraker, i can save the settings.
I like to see that this works without removing my webcams in moonraker

### Screenshots
In debug log:
Error 400: Cannot overwrite webcam 'camera_top' sourced from Moonraker

### Version Information
- Device-OS (iOS): 17.1.1
- Mobileraker: 2.6.7-340
- Klipper: v0.12.0-10
- Moonraker: v0.8.0-204

### Debug Logs
In debug log after clicking save when configured [webcam camera_top] in moonraker
Error 400: Cannot overwrite webcam 'camera_top' sourced from Moonraker

### Additional Context
None

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [-] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2023-11-30:
Hey,
Thanks for reporting this issue. Just to confirm, you are talking about webcams that are directly configured via a config file/section in moornaker?

jelleh, 2023-11-30:
Yes that is correct, in my case it is in the file /home/pi/printer_data/moonraker.conf

[webcam camera_side]
location: printer
enabled: True
service: mjpegstreamer
target_fps: 15
target_fps_idle: 5
stream_url: http://ender3.somedomain.net/webcam/?action=stream
snapshot_url: http://ender3.somedomain.net/webcam/?action=snapshot
flip_horizontal: False
flip_vertical: False
rotation: 0
aspect_ratio: 4:3

[webcam camera_top]
location: printer
enabled: True
service: mjpegstreamer
target_fps: 15
target_fps_idle: 5
stream_url: http://ender3.somedomain.net/webcam2/?action=stream
snapshot_url: http://ender3.somedomain.net/webcam2/?action=snapshot
flip_horizontal: False
flip_vertical: False
rotation: 0
aspect_ratio: 16:9

When i comment(#) the webcam sections in the moonraker.conf file, reload moonraker. Than i can save the settings in mobileraker again.

Clon1998, 2023-11-30:
Thanks!
I addressed that issue in [e786db6](https://github.com/Clon1998/mobileraker/commit/e786db692561adef39453f097e94b386458f738c) which will be included in the next release!