Clon1998/mobileraker #298: Instructions on how to use HTTP headers

rumym, 2023-12-16:
## Feature Request

### Problem Description
I have basic auth enabled on my Nginx server and I am trying to configure the http headers in the app to try and parse that information but cannot find instructions on how to configure the username and password via http headers.

### Proposed Solution
Suggest documenting that somewhere, if you help me configure it and wish I can write up the instructions

Clon1998, 2023-12-16:
Hey,
You will need to include the `Authorization` with the Base64 encoding of the username and password.
You can have a look at this [side](https://www.debugbear.com/basic-auth-header-generator) for more info and to generate the header based on the user and password directly.

In the app, you will need to set the header to `Authorization` and the value to `Basic <Base64 combination>`.

[image]

rumym, 2023-12-16:
Thank you! this worked and I learnt something new today :)

Over the break I will try to write up a guide on how to setup nginx proxy manager and secure it with basic auth

closed 2023-12-16 by rumym

Clon1998, 2023-12-16:
> Thank you! this worked and I learnt something new today :)
> 
> Over the break I will try to write up a guide on how to setup nginx proxy manager and secure it with basic auth

Sure, feel free to open a PR to either directly add a Markdown page or link to wherever you want to publish it.