Clon1998/mobileraker #299: App tries to connect to webcam using http instead of https

rumym, 2023-12-16:
## Bug Report

### Description
I am using nginx proxy configured with https and when I setup the printer using https://domain.com and websocket using ws://domain.com the app connects to the printer fine however, within the app the webcam fails as it is trying to connect to http://domain.com/webcam/?action=stream. 

My config in Fluid is /webcam/?action=stream and I have not hardcoded http anywhere in the configs. 

### Expected Behavior
The app should just append /webcam/?action=stream to the configured domain and respect the http/https denotation. 

### Screenshots
[image]

### Version Information

- Device-OS (Android/iOS): Android- Mobileraker: [version]
- Klipper: v1.1.2-6-gd4dee4c8
- Moonraker:v0.8.0-207-g6d8cb762

rumym, 2023-12-16:
Here is the camera config on the printer

[image]

Clon1998, 2023-12-16:
Hey,
Could you share a screenshot of the printer config within the app just to see if its actually configured with HTTPS

Also, just for debugging purposes, please try using an absolute webcam address to see if the nginx config is working.
Tl;Dr; replace the cam URL ` /webcam/?action=stream` with `https://domain.com/webcam/?action=stream`.

rumym, 2023-12-16:
Added above.

If I configure using absolute values it works just fine

Clon1998, 2023-12-16:
> Added above.
> 
> If I configure using absolute values it works just fine

Maybe I am blind but I don't see a config of the printer/machine setting within Mobileraker. I can only see the cam config from fluid and a screenshot of the dashboard.

rumym, 2023-12-16:
Gotcha.. sorry I missed that. Here you go
[image]

rumym, 2023-12-16:
I might have found another bug, if I configure the app to use the local ip:port the webcam just connects to localip and drops the port

rumym, 2023-12-16:
Ugh... this was a problem at my end the relative path of the printer on the app was incorrect and it kept overriding fluid. I fixed that and now it works. The only thing is not sure why the error was showing http instead of https. Either way, closing this bug.

rumym, 2023-12-16:
Looks like the bug I mentioned in https://github.com/Clon1998/mobileraker/issues/299#issuecomment-1858912557 is still an issue

Clon1998, 2023-12-16:
> Gotcha.. sorry I missed that. Here you go <img alt="Screenshot 2023-12-16 at 11 21 04 AM" width="707" src="https://private-user-images.githubusercontent.com/22590128/291011825-62008a38-9d3f-4d08-b1f3-5d4e5f791812.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3NTQ3ODksIm5iZiI6MTcwMjc1NDQ4OSwicGF0aCI6Ii8yMjU5MDEyOC8yOTEwMTE4MjUtNjIwMDhhMzgtOWQzZi00ZDA4LWIxZjMtNWQ0ZTVmNzkxODEyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE2VDE5MjEyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBiM2JlZmZiZmYzNWMyYTQ5MTAxYzZkOTVmZTFhY2RhNTA4N2RmMzE2NjJmYmEzZjg4OWZiNTJhNjU4NGRjMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DKM1dcK2bPDloLcpsJ-cCGDQwg_UxgdK7jBb0lOT1Ys">

I will read trough all of your answers tomorrow. However, just quickly looking at the settings you are currently not using SSL for the websocket connection. Please try to replace `ws` with `wss` and try again.

Clon1998, 2023-12-16:
> I might have found another bug, if I configure the app to use the local ip:port (http://192.168.1.3:4408) the webcam tries connects to localip (http://192.168.1.3/webcam/?action=stream) and drops the port

Thats correct. The port of the cam should be different to the printer port unless you are using a nginx on the pi to directly map the incoming requests to the correct internal web servers running on different ports.
If you want to specifically define a port for the webcam, use an absolute address that includes it.

Clon1998, 2023-12-16:
> Ugh... this was a problem at my end the relative path of the printer on the app was incorrect and it kept overriding fluid. I fixed that and now it works. The only thing is not sure why the error was showing http instead of https. Either way, closing this bug.

Because you used `ws` instead of `wss` and the app transform the ws address into an http/https one for the webcam afaik.

Clon1998, 2023-12-16:
> Looks like the bug I mentioned in [#299 (comment)](https://github.com/Clon1998/mobileraker/issues/299#issuecomment-1858912557) is still an issue. Here is a screenshot of my config and the error
> 
> ![Screenshot_20231216-115748](https://private-user-images.githubusercontent.com/22590128/291013329-3202b5fa-2e6f-4c53-8a5f-9a8308d52697.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3NTcxNzQsIm5iZiI6MTcwMjc1Njg3NCwicGF0aCI6Ii8yMjU5MDEyOC8yOTEwMTMzMjktMzIwMmI1ZmEtMmU2Zi00YzUzLThhNWYtOWE4MzA4ZDUyNjk3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE2VDIwMDExNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY5Y2M0NjhjMzcxMDViZjlmNzUyMWExZjRkMGY3MzMwNTQwOGI2Y2I0NjRkMTI2MjU3ZTkxOWY1MWM3ZWM3ZWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8KwJomQw7N5G57rurpNMelDnhYXUz1VZhOayZMv9jLo) ![Screenshot_20231216-115714](https://private-user-images.githubusercontent.com/22590128/291013332-af358e89-e934-4bb3-a7b9-7804bbb4db1d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3NTcxNzQsIm5iZiI6MTcwMjc1Njg3NCwicGF0aCI6Ii8yMjU5MDEyOC8yOTEwMTMzMzItYWYzNThlODktZTkzNC00YmIzLWE3YjktNzgwNGJiYjRkYjFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE2VDIwMDExNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThiY2ViMWZlZGM2YTJlY2MyNDQzZjE1MzY3MTJjYzc2M2FlZGI5NTRlNWMxNGE5MmY1NjQ4MGE2ZjRhN2Q5MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Meql852v4OeSrA3DtESv6x_YTY76LP40GCGm_rK3w4o)

Since you are running a K1, I assume this is a problem with Crealitie's port mappings (Unless you are running a rooted one?).
Can you try to open the URL that the app is showing on your PC to see if it also shows a 404 page? If that is the case, please try to add the port of the printer (4408) to the URL in the browser and see if it fixes it. If that is the case, please use an absolute address that includes the port for the cam.

rumym, 2023-12-16:
The K1 is rooted. These devices fluid is installed on port 4408 and the creality stuff is installed on port 80. That link does not work because in order for the webcam to work it will need to have 4408 in the URL.

Shouldn't the app just include the port with the relative URL?

rumym, 2023-12-16:
> I will read trough all of your answers tomorrow. However, just quickly looking at the settings you are currently not using SSL for the websocket connection. Please try to replace `ws` with `wss` and [...]

Yes, confirming that this was the issue. Thanks

closed 2023-12-28 by Clon1998