Clon1998/mobileraker #303: Camera configurations broken : not refreshing any settings when changed or adding in Fluidd or Mainsail

Ltek, 2023-12-25:
## Bug Report

### Description
I have 3 cams configured... 1 directly in Moonraker and 2 in Fluid. 
MobileRaker is not updating its cams at all... it will not added a new one if I add it in Fluidd. If I change any existing cam setting or the cam name, MobileRaker still has the old camera info/names no matter what I do (kill app, restart Firmware/Klipper, etc)

### Steps to Reproduce
1. use Mainsail or Fluidd to add a new cam and/or change any existing cam settings
2. kill MobileRaker and re-open.
3. Go to Cam sections and you will see nothing in the cam section changed. Still has all the old info, no new printer, no changed settings

### Version Information
- Device-OS (Android/iOS): [OS]
- Mobileraker: 2.6.8 - 2023-12-08
- Klipper: 09faed31
- Moonraker:  v0.8.0-231-gcc5dea71

### Debug Logs
zip attached
[mobileraker_debug-logs.zip](https://github.com/Clon1998/mobileraker/files/13767086/mobileraker_debug-logs.zip)

Clon1998, 2023-12-25:
> ## Bug Report
> ### Description
> I have 3 cams configured... 1 directly in Moonraker and 2 in Fluid. MobileRaker is not updating its cams at all... it will not added a new one if I add it in Fluidd. If [...]

Hey,
What version of Fluidd and Mainsail are you using?
I am unable to reproduce this issue with your description. Changing the title of a webcam via Mainsail triggers an instant refresh in the Mobileraker on my setup.

Ltek, 2023-12-30:
> Hey, What version of Fluidd and Mainsail are you using? I am unable to reproduce this issue with your description. Changing the title of a webcam via Mainsail triggers an instant refresh in the Mobileraker on my setup.

MobileRaker debug logs attached, tested again today.  

Today I started with 4 cams (one was Moonraker's) then 
[MobileRaker_DebugLogs.zip](https://github.com/Clon1998/mobileraker/files/13799022/MobileRaker_DebugLogs.zip)
deleted 2 and renamed 1... MobileRaker did not sync and still has all the old 4. 

moonraker v0.8.0-241-gca595163
mainsail v2.9.0
fluidd v1.27.1

[MobileRaker_DebugLogs.zip](https://github.com/Clon1998/mobileraker/files/13799023/MobileRaker_DebugLogs.zip)

Clon1998, 2023-12-31:
Thanks,
I just realized that Fluidd is still using the old webcam approach. That means that any change to a webcam and fluid webcams, in general, are not synced among different UIs. Only Mainsail and Mobileraker currently make use of the webcam API that enables syncing between webcam settings.

Can you confirm that you tried mainsail while changing webcams?

Ltek, 2024-01-01:
I didnt know that. I prefer Fluidd for most things it seems odd they are behind the curve on web cam API use.

I did confirm... 
Create cam in Mainsail -> shows in Fluidd 
Create cam in Fluidd -> does not show in Mainsail

Clon1998, 2024-01-01:
Okay. 
I remember talking to both devs because I proposed a change to the webcam API. Mainsail also switched but I think Fluidd did not yet.

Anyway, i will close this issue as complete.

closed 2024-01-01 by Clon1998