Clon1998/mobileraker #304: Add support for go2rtc

Xelinor, 2023-12-26:
## Feature Request

### Problem Description
I'm unable to view my camera feeds in Mobileraker that use WebRTC streams provided via go2rtc

### Proposed Solution
Add support for the go2rtc WebRTC camera feeds

### Additional Context
Mainsail commits that enabled this functionality:
https://github.com/mainsail-crew/mainsail/pull/1651

go2rtc documentation:
https://github.com/AlexxIT/go2rtc

Clon1998, 2023-12-28:
Hey,
I am struggling with setting up go2rtc.
I was able to add a source stream and can also watch that stream via MJPEG. However, If I select WebRtc via the admin UI or try using Mainsail I don't get any video feed.

Xelinor, 2023-12-28:
Your URL to view a stream should look something like this: http://local.ip:port/stream.html?src=configured_name

So for mine it's http://192.168.68.101:1984/stream.html?src=build_plate

And then (at least in Mainsail) you would select WebRTC(go2rtc)

I'm using go2rtc to use an RTSP security camera (eufy) as the WebRTC source, if that helps at all? And it's running through home assistant locally, so home assistant receives the RTSP stream and handles the conversion to WebRTC.

Hopefully that helps? Or is there an aspect of go2rtc that is tripping you up maybe?

Clon1998, 2023-12-29:
Thanks,
I figured it out. The problem was that I needed to transcode my jpeg input to H265

Xelinor, 2023-12-29:
My dude you are the best, thank you :)

commit 74116d15 (2023-12-29): Fixes #304 - Added Support for Go2Rtc