Clon1998/mobileraker #306: Mobileraker uses remote connection instead of local connection

Horgul123, 2023-12-28:
## Bug Report

### Description
Mobileraker doesn't use local connection 

### Steps to Reproduce
Ioen printer in mobileraker

### Expected Behavior
User local connection of possible 

### Screenshots
See below 

### Version Information

- Device-OS (Android/iOS):  Android 14
- Mobileraker: 2.6.8-357
- Klipper:  unknown (creality K1) 
- Moonraker: 0.8.0-231
[image]
[image]
[image]

Clon1998, 2023-12-28:
Hey,
are you sure you are using the correct moonraker port?
It seems like the app is unable to connect to moonraker via the default ports.

Horgul123, 2023-12-28:
What is the default port?
I've added port 4409 now and it seems to work. Thanks for the hint

Clon1998, 2023-12-28:
> What is the default port? I've added port 4409 now and it seems to work. Thanks for the hint

The default port is `7125` it should be defined within the `moonraker.conf`:
```
# moonraker.conf

[server]
host: 0.0.0.0
#  The host address in which to bind the HTTP server.  Default is to bind
#  to all ipv4 interfaces.  If set to "all" the server will bind to all
#  ipv4 an ipv6 interfaces.
port: 7125
#   The port the HTTP server will listen on.  Default is 7125
ssl_port: 7130
#   The port to listen on for SSL (HTTPS) connections.  Note that the HTTPS
#   server will only be started of the certificate and key options outlined
#   below are provided.  The default is 7130.
```

closed 2023-12-28 by Clon1998