Clon1998/mobileraker #308: Error if Powerstate is off

dMarabou, 2023-12-29:
## Bug Report

Since approx two weeks ago I can't Power on my printer via Mobileraker. The powerbutton is missing if the Printer is powered off and the option "bound_services: Klipper" is set

### Steps to Reproduce
Printer is powercontrolled via Tasmota-Socket. Klipper Service is bound to Powerstate
````
#moonraker.conf

[power printer]
type: tasmota
address: 192.168.5.43
bound_services: klipper #<----

````
### Expected Behavior
The Powerbutton is visible as before. 

### Screenshots
[image]

[image]

### Version Information

- Device-OS (Android/iOS): iOS17
- Mobileraker: 2.6.8-357
- Klipper: v0.12.0-55-gb5025580
- Moonraker: v0.8.0-240-g5585884d

Clon1998, 2023-12-29:
Hey,
Did this behavior occur after updating mobileraker or after changing something in the `[power]` config (Setting `bound_services: klipper `)?

Clon1998, 2023-12-29:
So I had a quick look and to me it seems like this is caused by moonraker.
I tried different `[power]` types (gpio, tasmota, klipper_device....) but for nearly all of them I get this error. Also, Fluidd and Mainsail stopped working (Given that all of these configured devices are dummies and do not control any actual power device).

I will investigate this further and get back to you.

Clon1998, 2023-12-29:
Related to https://github.com/Arksine/moonraker/issues/777

dMarabou, 2023-12-30:
I didn't change the config.

I'm not 100% sure but I think the error occurred after an specific update of Klipper, moonraker, etc..
Since then there were like 3 or 4 more updates

Clon1998, 2023-12-30:
> I didn't change the config.
> 
> I'm not 100% sure but I think the error occurred after an specific update of Klipper, moonraker, etc.. Since then there were like 3 or 4 more updates

Alright. Does Fluidd/Mainsail behave normally?

dMarabou, 2023-12-30:
Yes Mainsail works as usual and as soon as the printer is powered on Mobileraker works also flawless. 
I already reinstalled the app but it didn't change the behavior its behavior.

Clon1998, 2023-12-30:
> Yes Mainsail works as usual and as soon as the printer is powered on Mobileraker works also flawless. 
> I already reinstalled the app but it didn't change its behavior.  

Alright,
could you upload the log files of mobileraker while the app is in that weird state. They are located at the very bottom of the app settings page.

dMarabou, 2023-12-30:
> Alright,
> could you upload the log files of mobileraker while the app is in that weird state. They are located at the very bottom of the app settings page.

Of course. Here is the newest file:
```

12:25:39.920 (+0:00:00.003353)
💡 AppLifecycleState changed from null to AppLifecycleState.resumed
12:25:39.984 (+0:00:00.067405)
💡 Completed FirebaseRemote init
12:25:40.008 (+0:00:00.090712)
💡 Machine in box is printi (d7dce3d9-0289-46b6-8b39-570adafbb4d5)#330722490
12:25:40.008 (+0:00:00.090769)
💡 Completed Hive init
12:25:40.050 (+0:00:00.132430)
💡 Completed PaymentService init
12:25:40.050 (+0:00:00.132660)
💡 Selecting machine printi
12:25:40.050 (+0:00:00.132689)
💡 Received fetchAll
12:25:40.056 (+0:00:00.138597)
💡 Theme selected: 5, available theme len: 6
12:25:40.056 (+0:00:00.138634)
💡 Theme selected: 5, available theme len: 6
12:25:40.057 (+0:00:00.140261)
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {Supporter: EntitlementInfo(identifier: Supporter, isActive: true, willRenew: false, latestPurchaseDate: 2023-09-11T16:17:08Z, originalPurchaseDate: 2023-09-11T16:17:08Z, productIdentifier: mobileraker_3299_lifetime, isSandbox: false, ownershipType: OwnershipType.purchased, store: Store.appStore, periodType: PeriodType.normal, expirationDate: null, unsubscribeDetectedAt: null, billingIssueDetectedAt: null), supporter_subscription: EntitlementInfo(identifier: supporter_subscription, isActive: false, willRenew: false, latestPurchaseDate: 2023-07-03T13:59:16Z, originalPurchaseDate: 2023-07-03T13:59:20Z, productIdentifier: mobileraker_500_1m, isSandbox: false, ownershipType: OwnershipType.purchased, store: Store.appStore, periodType: PeriodType.normal, expirationDate: 2023-08-03T13:59:16Z, unsubscribeDetectedAt: 2023-07-04T00:15:46Z, billingIssueDetectedAt: null)}, active: {Supporter: EntitlementInfo(identifier: Supporter, isActive: true, willRenew: false, latestPurchaseDate: 2023-09-11T16:17:08Z, originalPurchaseDate: 2023-09-11T16:17:08Z, productIdentifier: mobileraker_3299_lifetime, isSandbox: false, ownershipType: OwnershipType.purchased, store: Store.appStore, periodType: PeriodType.normal, expirationDate: null, unsubscribeDetectedAt: null, billingIssueDetectedAt: null)}), allPurchaseDates: {mobileraker_500_1m: 2023-07-03T13:59:16Z, mobileraker_3299_lifetime: 2023-09-11T16:17:08Z}, activeSubscriptions: [], allPurchasedProductIdentifiers: [mobileraker_3299_lifetime, mobileraker_500_1m], nonSubscriptionTransactions: [StoreTransaction.create(transactionIdentifier: aa9a193619, revenueCatIdentifier: aa9a193619, productIdentifier: mobileraker_3299_lifetime, purchaseDate: 2023-09-11T16:17:08Z)], firstSeen: 2023-06-19T19:43:12Z, originalAppUserId: $RCAnonymousID:2698915d79c14b6398f1590e9fb8f831, allExpirationDates: {mobileraker_3299_lifetime: null, mobileraker_500_1m: 2023-08-03T13:59:16Z}, requestDate: 2023-12-29T23:48:17Z, latestExpirationDate: 2023-08-03T13:59:16Z, originalPurchaseDate: 2023-07-03T13:59:20Z, originalApplicationVersion: null, managementURL: null)
12:25:40.057 (+0:00:00.140336)
💡 RCat ID: $RCAnonymousID:2698915d79c14b6398f1590e9fb8f831
12:25:40.057 (+0:00:00.140377)
💡 Received isSupporter true
12:25:40.057 (+0:00:00.140394)
💡 Max allowed machines for non Supporters is -1
12:25:40.057 (+0:00:00.140411)
💡 Updated allMachinesProvider: printi (d7dce3d9-0289-46b6-8b39-570adafbb4d5)
12:25:40.058 (+0:00:00.140434)
💡 Completed initialRoute init
12:25:40.058 (+0:00:00.140449)
💡 Started initializeAvailableMachines
12:25:40.058 (+0:00:00.140460)
💡 Received all machines
12:25:40.058 (+0:00:00.140499)
💡 machineProvider creation STARTED d7dce3d9-0289-46b6-8b39-570adafbb4d5
12:25:40.058 (+0:00:00.140546)
💡 machineProvider creation DONE d7dce3d9-0289-46b6-8b39-570adafbb4d5 - returns null: false
12:25:40.058 (+0:00:00.140576)
💡 initialized all machineProviders
12:25:40.058 (+0:00:00.140586)
💡 Completed initializeAvailableMachines
12:25:40.064 (+0:00:00.146573)
💡 Notifications are permitted: true
12:25:40.064 (+0:00:00.146628)
💡 Successfully initialized ports!
12:25:40.064 (+0:00:00.146689)
💡 Updated hiddenMachinesProvider: 
12:25:40.075 (+0:00:00.157871)
💡 Successfully initialized AwesomeNotifications and created channels and groups!
12:25:40.075 (+0:00:00.157924)
💡 Initializing notification listeners
12:25:40.075 (+0:00:00.157969)
💡 Initializing remote messaging
12:25:40.079 (+0:00:00.162409)
💡 Restored 1 LiveActivities from storage: {d7dce3d9-0289-46b6-8b39-570adafbb4d5: 2F579694-BD53-4639-8E7D-6525CF69AA04}
12:25:40.080 (+0:00:00.162455)
💡 Started to listen for LiveActivity updates
12:25:40.080 (+0:00:00.162515)
💡 Completed LiveActivityService init
12:25:40.080 (+0:00:00.162527)
💡 Initializing machineRepoListener
12:25:40.080 (+0:00:00.162536)
💡 Completed NotificationService init
[... 81 lines cut ...]
⛔ isRefreshing SKIPPED
12:25:49.287 (+0:00:09.369689)
💡 Invalidating klipper service provider, to retry klippy fetching
12:25:49.299 (+0:00:09.381715)
💡 >>>Fetching Server.Info
12:25:49.300 (+0:00:09.382821)
💡 >>>Fetching Printer.Info
12:25:49.301 (+0:00:09.383645)
💡 Printer Service received klippyState: KlipperState.disconnected
12:25:49.351 (+0:00:09.433491)
💡 <<<Received Server.Info
12:25:49.351 (+0:00:09.434181)
⚠️ Jrpc Error while refreshing KlippyObject: Klippy Host not connected
12:25:49.351 (+0:00:09.434374)
💡 Printer Service received klippyState: KlipperState.disconnected
12:25:49.352 (+0:00:09.434580)
💡 Printer Service received klippyState: KlipperState.disconnected
12:25:49.365 (+0:00:09.448345)
⛔ isRefreshing SKIPPED
12:25:51.989 (+0:00:12.072248)
💡 Getting fcm.client
12:25:54.896 (+0:00:14.979267)
💡 Show Dialog request for DialogType.logging

```

Clon1998, 2023-12-31:
Thanks,
I think I found the problematic code section. I will get back to you once I resolved the issue.

dMarabou, 2023-12-31:
> Thanks,
> I think I found the problematic code section. I will get back to you once I resolved the issue.

Danke für deinen Einsatz und die schöne App! 
Wünsche dir einen Guten Rutsch ins neue Jahr.

Clon1998, 2024-01-10:
> Also running into this issue, hoping to see it fixed soon. Love the work :)

Well, it is fixed 🦾
I am currently preparing the next build/release.

commit 40b3a60e (2023-12-31): Fixed #308 -  Ensured that printer.info is only called after the server reported klippy is ready