Clon1998/mobileraker #311: Camera URL seems to ignore FLUIDD port

funkyluke, 2024-01-04:
## Bug Report

### Description
The Camera Stream url is not correctly generated from the connection string and webcam config

### Steps to Reproduce
1. Setup printer so that fluidd runs on e.g port 4408
3. connect mobileraker to the fluidd instance by using the [printer-ip]:4408 address
4. camera stream endpoint is automatically received from fluidd
5. camera view shows 404 error because the generated url is [printer-ip]/webcam/?action=stream, not [printer-ip]:4408/webcam/?action=stream
6. The same happens when using moonrakers port,

### Expected Behavior
The correct URL should be generated and the camera stream should be showing

### Version Information

- Device-OS (Android/iOS): [Android]
- Mobileraker: 2.6.8-357
- Klipper: whatever is stock on K1 Max currently
- Moonraker: 0.8.0-243

funkyluke, 2024-01-04:
This seems to be related to #212 as my fluidd conf also uses a relative webcam path that is incorrectly appended

Clon1998, 2024-01-04:
Hey,
webcams defined via Fluidd are currently not using the new Webcam API of Moonraker and, therefore, do not sync with Mobileraker.