Clon1998/mobileraker #315: unable to see colors on IOS notification timer

BlackNet, 2024-01-15:
When a print job is running, and it starts the notification timer, the text is impossible to see.  I have found no area to change the colors of that.  Tried changing system to dark mode and makes no difference.

BlackNet, 2024-01-15:
[image]

BlackNet, 2024-01-15:
I am on 17.3. I tried changing modes but no change in the coloring.

Clon1998, 2024-01-15:
> I am on 17.3. I tried changing modes but no change in the coloring.

Thanks for letting me know.
I believe Apple has finally resolved the issue with the colorScheme/mode detection for live activities in versions 17.1 and 17.2. In the previous version, 16.4, I was unable to retrieve the colorScheme, and the background of the activity would not update (Stay on light mode). In version 17.0, I would always get the light color schema, so I modified the activity to provide a static color schema. Currently, I am downloading the new version, XCode 17.3, to test how the activity behaves. I assume that the color schema is stuck on light mode again. However, the change 2902a21 I made earlier should resolve this issue in any case. All colors are now static and ignore any of the colorScheme/modes provided by the user.

closed 2024-01-19 by Clon1998

BlackNet, 2024-01-23:
Just an update FYI. New version does work and it's readable.  Looks like it does not change from light to dark but that's ok.  [image]

Clon1998, 2024-01-23:
> Just an update FYI. New version does work and it's readable.  Looks like it does not change from light to dark but that's ok.  [image]

Thanks for the feedback!
I removed any color mode adaption because apple broke several different components in different iOS versions.

commit 2902a215 (2024-01-15): Fix #315- Fixed live activity color on Ios 16.4
commit 4511b288 (2024-01-15): Fix #315- More fixes for the live activity color