Clon1998/mobileraker #320: Webcam won't load on remote connection

m-klecka, 2024-01-19:
## Bug Report

### Description
The Webcam won't load when on remote connection. I'm using wireguard for that. If I access mainsail from Firefox on Android using the wireguard ip, it works just fine, but when accessing through the same wireguard ip from mobileraker, the Webcam won't load and just gives an error

### Expected Behavior
This should work normally

### Screenshots
[image]
[image]

### Version Information

- Device-OS (Android/iOS): android
- Mobileraker: 2.6.9, 0.4.0-41-g7297fbca
- Klipper: 0.12.0-65-g96ab9069 
- Moonraker: 0.8.0-247-g3008a13e

### Debug Logs
[mobileraker_2024-01-19T08-21-07.log](https://github.com/Clon1998/mobileraker/files/13985774/mobileraker_2024-01-19T08-21-07.log)

### Additional Context

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

m-klecka, 2024-01-19:
> Thanks for reporting the issue. The app's log sadly does not state any issue on the request and offers the schema of the camera-streamer's webrtc protocol. I also was unable to reproduce the issue with my VPN setup, cam and printers. Therefore, I'd like to have a look at the logs of `crowsnest` in debug mode.
> 
> To do this:
> 
>     1. Set the log level in the `crowsnest.conf` to debug :
> 
> 
> ```
> [crowsnest]
> ...
> log_level: debug   # <---- UPDATED to debug
> ...
> ```
> 
>     2. Restart crowsnest
> 
>     3. Try to connect via VPN on the browser + via mobileraker
> 
>     4. Upload the logs

here it is:)
[crowsnest.log](https://github.com/Clon1998/mobileraker/files/13994160/crowsnest.log)

Clon1998, 2024-01-19:
Thanks,
what is the operation mode for the cam? The log states both MJPEG (Adaptiv) and WebRTC?

m-klecka, 2024-01-19:
It's webrtc. Mjpeg in the log is probably when I first started messing with crowsnest

Clon1998, 2024-01-19:
> It's webrtc. Mjpeg in the log is probably when I first started messing with crowsnest

Alright. But it works in LAN mode?

m-klecka, 2024-01-19:
> > It's webrtc. Mjpeg in the log is probably when I first started messing with crowsnest
> 
> Alright. But it works in LAN mode? 

Yes LAN mode is fine. Only remote connection has problems.

Clon1998, 2024-01-19:
> > > It's webrtc. Mjpeg in the log is probably when I first started messing with crowsnest
> > 
> > 
> > Alright. But it works in LAN mode?
> 
> Yes LAN mode is fine. Only remote connection has problems.

I will need to dive deeper into this issue and compare my implementation with the one of Mainsail.
I currently cant wrap my head around the reason why it works in local mode but does not work via VPN (Since the browser's WebRtc is working).