Clon1998/mobileraker #321: App Crashes and then is reset

DelusionalAI, 2024-01-19:
## Bug Report

### Description
Rarely when clicking on a live notification to open the app, the crashes with the below error screen. If you close and then reopen the app, it launches with all app settings reset, so I need to re add the printers and so on.

### Expected Behavior
Just expected the app to open to the printer whose notification I click on. 

### Screenshots
[image]

### Version Information

- Device-OS (Android/iOS): iOS
- Mobileraker: 2.6.9
- Klipper: 0.12.0-65-g96ab9069
- Moonraker: 0.8.0-262-gc4f1d251
- Mobileraker-companion: v0.4.0-41-g7297fbca

### Debug Logs
[mobileraker_2024-01-19T07-40-35.log](https://github.com/Clon1998/mobileraker/files/13989846/mobileraker_2024-01-19T07-40-35.log)
[mobileraker_2024-01-19T07-40-26.log](https://github.com/Clon1998/mobileraker/files/13989847/mobileraker_2024-01-19T07-40-26.log)
[mobileraker_2024-01-19T00-41-41.log](https://github.com/Clon1998/mobileraker/files/13989848/mobileraker_2024-01-19T00-41-41.log)
[companion-logs.txt](https://github.com/Clon1998/mobileraker/files/13989865/companion-logs.txt)

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2024-01-19:
Hey,
I am sorry to hear about that behavior. I will have a look at that ASAP.

Clon1998, 2024-01-19:
I looked into the issue and couldn't find the root cause of the problem with the live_activity feature. However, I did come across a few issues that could cause the app to fail to start up properly. This could lead to problems when trying to access local storage. By default, if the app can't load the local storage, it resets it entirely to ensure the app is able to start. 
However, I have reworked this exact code to give the user the option to either reset the storage or restart the app.

commit a046b74d (2024-01-19): Fix #321 - Removed Auto-Reset and let the user decide about this action
closed 2024-01-19 by Clon1998